OP-TEE/build #18: 'make run' for QEMU config hangs after starting the first terminal window

pm215, 2015-10-26:
If I do a 'make run' (or 'make run-only') for the QEMU config, following the instructions in https://github.com/OP-TEE/optee_os, then this hangs after starting the first terminal window.

This seems to be because the makefile process does (more or less)
   gnome_terminal
   gnome_terminal
   qemu

and since the gnome_terminal execution isn't backgrounded then we never continue past starting the first terminal window. Adding an '&' to the definiton of the launch-terminal function fixes this for me:

```
diff --git a/qemu.mk b/qemu.mk
index d278895..3754554 100644
--- a/qemu.mk
+++ b/qemu.mk
@@ -193,7 +193,7 @@ endef
 
 define launch-terminal
        @nc -z  127.0.0.1 $(1) || \
-       gnome-terminal -e "$(BASH) -c '$(SOC_TERM_PATH)/soc_term $(1); exec /bin/bash -i'" --title=$(2)
+       gnome-terminal -e "$(BASH) -c '$(SOC_TERM_PATH)/soc_term $(1); exec /bin/bash -i'" --title=$(2) &
 endef
 
 .PHONY: run
```

jforissier, 2015-10-26:
Hi Peter, thanks for reporting this.
Which distribution/gnome-terminal are you running? I have no problem with Ubuntu 15.04. Now, adding a '&' certainly doesn't hurt, so I guess we should do that.

pm215, 2015-10-26:
This is Ubuntu trusty.
```
$ gnome-terminal --version
GNOME Terminal 3.6.2
```

pm215, 2015-10-26:
Ah, it turns out that if you already have *one* gnome-terminal running, then running 'gnome-terminal' from the command line will start a new window (presumably by talking to the other process) and then return to the command line. But if this 'gnome-terminal' is the first one, then it does not background itself or return to the command line until all the windows have closed.

I guess that you use gnome-terminal as your standard terminal, so it works for you. I use xterm, so the gnome-terminal started by this makefile is typically the only one running.

ototo, 2015-10-27:
It might make sense to replace gnome-terminal in all scripts with xterm as xterm seems to be installed by default on majority of distros (haven't seen one that would not have it yet), while gnome-terminal is only installed by default when distro is gnome based or gnome DE has been installed.

jbech-linaro, 2015-10-27:
I've never seen this problem myself, but indeed, I use gnome-terminal as my standard terminal.

@ototo , I'm fine with updating to use `xterm` instead, I guess I was a bit selfish when I decided to use gnome-terminal instead of something more "generic" like xterm. Would you mind submit a patch for this?

Also, it might be worth trying to spawn the users standard terminal instead of trying to force a certain terminal. What about using: 
```
/usr/bin/x-terminal-emulator -> /etc/alternatives/x-terminal-emulator
/etc/alternatives/x-terminal-emulator -> /usr/bin/gnome-terminal.wrapper
```

Or is the `update-alternatives` still only a Ubuntu based feature?

jforissier, 2015-10-27:
@ototo IMO xterm is a bit old-fashioned :wink: but you may have a point. FWIW, the following works fine on my laptop:
```
diff --git a/qemu.mk b/qemu.mk
index d278895..0593c64 100644
--- a/qemu.mk
+++ b/qemu.mk
@@ -193,7 +193,7 @@ endef
 
 define launch-terminal
        @nc -z  127.0.0.1 $(1) || \
-       gnome-terminal -e "$(BASH) -c '$(SOC_TERM_PATH)/soc_term $(1); exec /bin/bash -i'" --title=$(2)
+       xterm -title $(2) -e $(BASH) -c "$(SOC_TERM_PATH)/soc_term $(1)" &
 endef
 
 .PHONY: run
```

pm215, 2015-10-27:
Yes, x-terminal-emulator is an Ubuntu/Debian-ism. If you want to be fancy you could borrow the (public domain) wrapper script i3-sensible-terminal from i3wm (https://github.com/i3/i3/blob/next/i3-sensible-terminal), which basically does:
```
for terminal in $TERMINAL x-terminal-emulator urxvt rxvt terminator Eterm aterm xterm gnome-terminal roxterm xfce4-terminal termite lxterminal mate-terminal terminology; do
    if command -v $terminal > /dev/null 2>&1; then
        exec $terminal "$@"
    fi
done
```
But this is probably veering into over-engineering and just using xterm would do fine :-)

ototo, 2015-10-27:
And one more bit: to make xterm looking better I'm using the following ~/XTerm file:
```
XTerm*Background: black
XTerm*Foreground: green
XTerm*faceName: Cousine
XTerm*faceSize: 14
```

jenswi-linaro, 2015-10-27:
What? Everyone knows that xterm has white background and black foreground. ;-)

closed 2015-10-28 by cedric-chaumont-st