OP-TEE/build #28: RFC: Bug in *_CLEAN_COMMON_FLAGS?

vchong, 2015-11-23:
Hi,

In common.mk:
```LINUX_CLEAN_COMMON_FLAGS ?= $(LINUX_COMMON_FLAGS)```

In hikey.mk:
```LINUX_COMMON_FLAGS += ARCH=arm64 Image modules dtbs```
```LINUX_CLEAN_COMMON_FLAGS += ARCH=arm64```

```
> make linux-clean
rm -f /home/devel/optee/repo/build/../linux/.config
make -C /home/devel/optee/repo/build/../linux
LOCALVERSION= CROSS_COMPILE="/usr/bin/ccache
/home/devel/optee/repo/build/../toolchains/aarch64/bin/aarch64-linux-gnu-"
ARCH=arm64 Image modules dtbs ARCH=arm64 clean
```

So we're re-making everything before cleaning?

For other platforms, it's not as bad since we run ```make``` without specifying any targets so we only repeat the ```ARCH=arm64``` flag, i.e.

```
> make linux-clean
rm -f /home/devel/optee/repo/build/../linux/.config
make -C /home/devel/optee/repo/build/../linux
LOCALVERSION= CROSS_COMPILE="/usr/bin/ccache
/home/devel/optee/repo/build/../toolchains/aarch64/bin/aarch64-linux-gnu-"
ARCH=arm64 ARCH=arm64 clean
```

Same situation for ```OPTEE_OS_CLEAN_COMMON_FLAGS``` and ```OPTEE_LINUXDRIVER_CLEAN_COMMON_FLAGS```.

How should we fix this? Quick and easy way is to just redefine ```LINUX_CLEAN_COMMON_FLAGS``` in hikey.mk, i.e.
```LINUX_CLEAN_COMMON_FLAGS = LOCALVERSION= CROSS_COMPILE=$(CROSS_COMPILE_NS_KERNEL) ARCH=arm64```?
Or ```LINUX_CLEAN_COMMON_FLAGS := LOCALVERSION= CROSS_COMPILE=$(CROSS_COMPILE_NS_KERNEL) ARCH=arm64```?
Or remove the dependency of *_CLEAN_COMMON_FLAGS on *_COMMON_FLAGS?
Or something else?

Thanks!

jbech-linaro, 2015-11-23:
Indeed, seems like a bug. The "clean" target should only contain clean / mrproper etc for the ARCH in use, no other dependencies should be stated. Should be quite easy to do this in the "common"-file instead of hacking individual files directly.

jbech-linaro, 2015-11-25:
This would be sufficient, the other stuff could be kept as it is in qemu.mk, hikey.mk etc... since we already have the correct ARCH flag there.

```
diff --git a/common.mk b/common.mk
index ae98c70..ece5ff5 100644
--- a/common.mk
+++ b/common.mk
@@ -73,7 +73,7 @@ linux-common: linux-defconfig
 linux-defconfig-clean-common:
        @rm -f $(LINUX_PATH)/.config
 
-LINUX_CLEAN_COMMON_FLAGS ?= $(LINUX_COMMON_FLAGS)
+LINUX_CLEAN_COMMON_FLAGS ?= CROSS_COMPILE=$(CROSS_COMPILE_NS_KERNEL)
        
 linux-clean-common: linux-defconfig-clean
        $(MAKE) -C $(LINUX_PATH) $(LINUX_CLEAN_COMMON_FLAGS) clean
```

qemu gives ...
```
make -C /home/jbech/devel/optee/build/../linux CROSS_COMPILE="/usr/bin/ccache /home/jbech/devel/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" ARCH=arm clean
```

hikey gives ...
```
make -C /home/jbech/devel/optee/build/../linux CROSS_COMPILE="/usr/bin/ccache /home/jbech/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" ARCH=arm64 clean
```

etc ...

closed 2015-12-04 by pascal-brand-st