OP-TEE/build #61: Juno.mk: BusyBox target

Miraje, 2016-04-22:
When I was checking the `juno.mk` file I saw there was these lines for BusyBox:

```
BUSYBOX_COMMON_TARGET = fvp
BUSYBOX_CLEAN_COMMON_TARGET = fvp clean
```
As juno is an Versatile Express board I was wondering if I should change to:

```
BUSYBOX_COMMON_TARGET = vexpress
BUSYBOX_CLEAN_COMMON_TARGET = vexpress clean
```
I had this doubt because when I saw the content of `generate-cpio-rootfs.sh` script the were some flags that I didn't know if could generate some problems or not:

`CFLAGS=${CFLAGS-"-Wno-strict-aliasing -Wno-unused-result -marm -mabi=aapcs-linux -mthumb -mthumb-interwork -mcpu=cortex-a15"} `

Note: I am using juno-r2 model and have already OP-TEE working correctly.

If someone could help me with this issue I would be very thankful.

Miraje, 2016-04-25:
Should I leave as it is?

jbech-linaro, 2016-04-25:
Hi @Miraje , Thanks for reaching out. I think we can leave it as it is. It's just a couple of flags for creating the root fs and the same rootfs can be shared between Juno and the pure FVP build.

Miraje, 2016-04-25:
@jbech-linaro Thank you very much for your response. I will leave as it is then =)

closed 2016-04-25 by Miraje