OP-TEE/build #64: Flash HiKey Debian

lordqwerty, 2016-05-17:
Hello,

I am having some issues flashing the debian version for HiKey. As per the instructions of running `make flash` I have closed PINS 1-2, 5-6 leaving 3-4 open. I have attached photos of my board connected and a screenshot of my terminal. 

As per the instructions at [HiKey Wiki](http://wiki.lemaker.org/HiKey(LeMaker_version):Quick_Start) I have also added the rules. 

Can anyone spot the error? 
Happy to set this up and update README with pictures etc. 

[image]
[image]
[image]

vchong, 2016-05-17:
It would help to have a serial cable connected to the board and per the instruction:
Wait until you see the (UART) message
"Android Fastboot mode - version x.x Press any key to quit."
on the serial console before pressing any key on your 'build' console to continue flashing.

Is the board fresh out of the box? Do you see any device listed when running the command `fastboot devices`? If not, then maybe try `make recovery` and follow the instructions printed instead of `make flash`.

jbech-linaro, 2016-05-17:
Hi Jorden,

Your connections are all fine, so this is a software issue.

Are you seeing the `Android Fastboot mode - version x.x Press any key to quit.` message? If not, then you probably need to flash it in recovery mode the first time. So if you don't see that message, disconnect power, put the jumper back on PIN 3-4 and then run `make recovery` instead and follow the instructions coming from make.

Not strictly related, but if you can flash, then this might lead to another issue ... I.e., is this the 4GB LeMaker HiKey or is it 8GB? We only had access to the 4GB version when writing this. So the ptable (partition table) is for 4GB HiKey's:
https://github.com/OP-TEE/build/blob/master/hikey.mk#L411

If you have the 8GB, then you should download this partition table instead
http://builds.96boards.org/releases/reference-platform/debian/hikey/16.03/bootloader/ptable-linux-8g.img

And then you must update the Makefile accordingly.
Here https://github.com/OP-TEE/build/blob/master/hikey.mk#L365
and here https://github.com/OP-TEE/build/blob/master/hikey.mk#L411

Feel free to send a pull request / patch if you can come up with something working for both the 4GB and 8GB.

lordqwerty, 2016-05-18:
Currently flashing the board now, will report how it goes. 

In terms of the the 8gb and 4gb flash it seems that the HiKey Boards currently sold now the [1gb RAM version](http://www.lenovator.com/product/86.html#params) and the [2gb RAM version](http://www.lenovator.com/product/90.html#params) only have 8gb eMMC flash. Would it therefore be better to switch the Makefile to use 8gb and have a 4gb flag to be set if needed? Depending on the decision I am happy to update the Makefile as appropriate. 

Also would it be worth adding a small section on the udev rules needed in the project README? I am happy to do this.

jbech-linaro, 2016-05-18:
Sounds good and I'm OK with defaulting to the 8GB flash version as long as it's easy to still flash the 4GB version by some kind of flag as you already indicated.

Feel free to create a patch for both that and the README udev changes. It'll be reviewed in detail when you've created the pull request.

Thanks!

jbech-linaro, 2016-06-29:
@lordqwerty is this still something you are committed to do? I.e, provide a patch for it?

lordqwerty, 2016-06-29:
@jbech-linaro sorry I currently do not have the time to do this, no flexibility in work at the moment.

jbech-linaro, 2016-06-29:
Ok, I'll keep it open for a while until I've decided whether I should close it or provide a fix on my own.

gitfineon, 2016-12-08:
Android Scripts are using a flag called `TARGET_USERDATAIMAGE_4GB=true` to create the filesystem image, but atm that doesn't work properly with the 4GB HiKey Board from CircuitCo.
https://source.android.com/source/devices.html#running-android-hikey

jbech-linaro, 2016-12-08:
@gitfineon what were the issues you were facing when trying AOSP on the 4GB Board? Probably not relvant for OP-TEE issues, but maybe we should let the 96Boards people know about it and eventually John Stultz.

gitfineon, 2016-12-09:
@jbech-linaro it was not possible to flash the image as expected including the 4g option with the `./flash-all.sh /dev/ttyUSB1 4g` script. It stopped with a "imagesize too big" message ... i modified the script to flash all parts invidiually like in older guides and with a reboot to UEFI fastboot inbetween. Maybe I made a mistake there, but it was possible to flash all parts including `userimage.img`. Everything worked fine until AOSP tries to write to userdata ... `ext4 write error`

Rebuilding userdate image with or without `TARGET_USERDATAIMAGE_4GB=true` was recognized but I did not had the time to verify the root cause. Make it completely clean and build all images again lasts more the 3 hours on my machine. Flashing the Binaries from [96boards](http://www.96boards.org/documentation/ConsumerEdition/HiKey/Downloads/AOSP.md/) worked and I reused the 4g ptable, but I think that was not the problem.

I did not want to fill a bug without verifying this behavior, maybe you can try to build the official AOSP stuff for CircuitCo 4GB version?

closed 2017-01-23 by jforissier