OP-TEE/build #85: RPi3 wrong default uboot env

igoropaniuk, 2016-12-15:
In the default [uboot env](https://github.com/OP-TEE/build/blob/master/rpi3/firmware/uboot.env), there are two similar vars for storing address, where fdt have to be loaded: fdt_addr_r and fdt_ad**d**dr_r (seems it's a spelling mistake). 
In case  if someone tries to setup TFTP/NFS boot (described in this [manual](https://github.com/OP-TEE/optee_os/blob/master/documentation/rpi3.md)) and follow step by step (where **fdt_addr_r** value is used) he won't receive a workable solution.

jforissier, 2016-12-15:
What should be fixed in your opinion? `uboot.env` or `rpi3.md`?

jbech-linaro, 2016-12-15:
So, there are several ways to skin a cat .. we could.
1. Update the docs
2. Fix the uboot.env file (the binary)
3. Remove the pre-generated uboot.env file and instead generate it from a [text file](https://github.com/jbech-linaro/build/commit/c5c1c8e18020e4f869a01770238e24c2b3bb2db6).

In my opinion I'd prefer alt 3, otherwise alt 2 and last alt 1. Updating docs with something that is incorrect doesn't sound like a good thing (even though it "solves" the problem). Alt 2 ... well, that's a quick thing to do. I can do it later today and then we have a working solution. But in the long run I prefer to generate it from a clean u-boot-env.txt file that will build the uboot.env file when building the RPi3 setup, i.e, we're talking about making changes to OP-TEE/build/rpi3.mk.

@jforissier what do you say, should I start out with alt2 and then we won't have the issue any longer? And then when someone finds half a day he or she could building upon the patch I started working with.

igoropaniuk, 2016-12-15:
@jbech-linaro 
I think alt 3 is the best option here. Anyway it's better to have a convenient way to edit/generate uboot.env "offline" on a dev pc and then easily deploy it, because, as for me, editing uboot env via UART looks like hell :)

jforissier, 2016-12-15:
@jbech-linaro thanks for clarifying. I think we all agree. Let's do as you propose, (2) then (3). And let's make sure no animals are harmed in the process ;)

jbech-linaro, 2016-12-15:
@igoropaniuk volunteered to make a proper fix right away (3) and I'll do my best to not harm the animals :)

closed 2016-12-21 by jbech-optee-adm

wallento, 2016-12-21:
May I ask what variables made it into the env and which not? The environment is like a third of the original size, in particular stuff like the board_*, ethernet and serial are missing.

I was about to prepare a related patch that randomizes the serial and ethernet MAC and I am not sure if the removed variables are redundant.

wallento, 2016-12-21:
Please also see #90

jbech-linaro, 2016-12-21:
@wallento , basically what I did when I created the uboot.env.txt (which is what Igor used in this PR) was remove everything that didn't seem to be used anywhere. At the same time when doing this cleanup I also replaced some hardcoded numbers with variables instead.

IIRC the ethernet variable etc will still show up, i.e. it seems like that is a variable that is created automatically. Having that said, it's a couple of month since I did the changes and haven't looked at it since then (I've just re-used the same uboot.env), so I could have forgotten something. But don't be afraid of make improvement. I'm pretty sure this could be re-written with more logic to be able to supported different targets in a better way.

wallento, 2016-12-21:
Thanks @jbech-linaro. I am unfortunately also out of my comfort zone here. We used the randomization because we had multiple rpi3 with identical ethernet MAC addresses on our network. I will dig deeper into how the MAC is derived with the current HEAD.