OP-TEE/build #103: Enable network (eth0) in qemu image?

msa2, 2017-01-12:
When qemu boots, I see

```
Freeing unused kernel memory: 1024K (80900000 - 80a00000)
preparing networking...
ip: SIOCGIFFLAGS: No such device
running rc.d services...
mounting tmpfs at /dev ...done
mounting pts at /dev/pts ...done
creating initial mdev device nodes ...done
setting system time from /dev/rtc0
```

..and one booted
```
root@Vexpress:/ ls /sys/class/net/
lo
root@Vexpress:/ ifconfig eth0 192.168.1.158 up
ifconfig: SIOCSIFADDR: No such device
root@Vexpress:/ soc_term: read fd EOF
```
I've searched, and tried multiple "-netdev ..." variations on qemu start, but none seems to help. Anyone? What I'm missing in configuration (seems to be some basic linux config issue?) (I even enable E1000 and PCI, but it changed nothing).

jenswi-linaro, 2017-01-12:
Did you compile the driver into the kernel or as a module? In the latter case I guess you haven't included the module in the rootfs.

msa2, 2017-01-12:
Ahh, but which driver? I just guessed some and enabled in linux .config
```
CONFIG_PCI=y
CONFIG_E1000=y
```
and in boot I see ...
```
e1000: Intel(R) PRO/1000 Network Driver - version 7.3.21-k8-NAPI
e1000: Copyright (c) 1999-2006 Intel Corporation.
```
...but, still resulting the "No such device" listed in first message.

I'm just looking for some configuration recipe that someone has used (including any additional qemu parameters, if needed). In current try I added line
`	-netdev user,id=vmnic -device e1000,netdev=vmnic \
`
... and also tried
`		-netdev user,id=vmnic -device virtio-net,netdev=vmnic \
`
but nothing seems to change.

msa2, 2017-01-12:
Ok, I got the eth0 interface qith qemu command
`	-netdev user,id=vmnic -device virtio-net-device,netdev=vmnic \
`
... but probably needs other configuration to actually communicate...
... google gives so many different solutions, all slightly not working or something :-(

igoropaniuk, 2017-01-12:
Hi @msa2,
when you configured SLiRP user networking with:
`-netdev user,id=vmnic -device virtio-net-device,netdev=vmnic`
did your check if virtual eth0 received ip address?

msa2, 2017-01-13:
Configuring `e1000 `doesn't seem to have any role at this point (it does not bring up any 'eth0' device unless -netdev is used), thus I removed it and tried with plain configuration and adding the line into qemu start (qemu.mk)

`-netdev user,id=vmnic -device virtio-net-device,netdev=vmnic \ `

The tail end of the normal world console log looks as follows:
```
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Advanced Linux Sound Architecture Driver Initialized.
clocksource: Switched to clocksource arch_sys_counter
NET: Registered protocol family 2
TCP established hash table entries: 16384 (order: 4, 65536 bytes)
TCP bind hash table entries: 16384 (order: 5, 131072 bytes)
TCP: Hash tables configured (established 16384 bind 16384)
UDP hash table entries: 1024 (order: 3, 32768 bytes)
UDP-Lite hash table entries: 1024 (order: 3, 32768 bytes)
NET: Registered protocol family 1
RPC: Registered named UNIX socket transport module.
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
Unpacking initramfs...
Freeing initrd memory: 9008K (8837a000 - 88c46000)
futex hash table entries: 256 (order: 2, 16384 bytes)
workingset: timestamp_bits=30 max_order=19 bucket_order=0
squashfs: version 4.0 (2009/01/31) Phillip Lougher
9p: Installing v9fs 9p2000 file system support
io scheduler noop registered (default)
tpm tpm0: tmptee_init complete
4000000.flash: Found 2 x16 devices at 0x0 in 32-bit bank. Manufacturer ID 0x000000 Chip ID 0x000000
Intel/Sharp Extended Query Table at 0x0031
Using buffer write method
libphy: Fixed MDIO Bus: probed
usbcore: registered new interface driver usb-storage
mousedev: PS/2 mouse device common for all mice
rtc-pl031 9010000.pl031: rtc core: registered pl031 as rtc0
ledtrig-cpu: registered to indicate activity on CPUs
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
optee firmware:optee: probing for conduit method from DT.
optee firmware:optee: initialized driver
oprofile: no performance counters
oprofile: using timer interrupt.
NET: Registered protocol family 17
9pnet: Installing 9P2000 support
Registering SWP/SWPB emulation handler
rtc-pl031 9010000.pl031: setting system clock to 2017-01-13 07:19:52 UTC (1484291992)
ALSA device list:
  No soundcards found.
Freeing unused kernel memory: 1024K (80800000 - 80900000)
preparing networking...
udhcpc (v1.25.0.git) started
Sending discover...
Sending select for 10.0.2.15...
Lease of 10.0.2.15 obtained, lease time 86400
running rc.d services...
mounting tmpfs at /dev ...done
mounting pts at /dev/pts ...done
creating initial mdev device nodes ...done
setting system time from /dev/rtc0
Starting telnetd...
uptime idle: 3.21 0.49

Please press Enter to activate this console. 
root@Vexpress:/ ifconfig -a
eth0      Link encap:Ethernet  HWaddr 52:54:00:12:34:56  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1180 (1.1 KiB)  TX bytes:342 (342.0 B)

lo        Link encap:Local Loopback  
          LOOPBACK  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```
So, no address on eth0 interface, log shows mysterious referencence to 10.0.2.15, but that does not seem to be anywhere. Loopback interface is not up, but if you bring it up, ping to loopback address works.

```
root@Vexpress:/ ping 10.0.2.15
PING 10.0.2.15 (10.0.2.15): 56 data bytes
ping: sendto: Network is unreachable
...
root@Vexpress:/ ping 10.0.2.15
PING 10.0.2.15 (10.0.2.15): 56 data bytes
ping: sendto: Network is unreachable
root@Vexpress:/ telnet 91.197.85.151 80
telnet: can't connect to remote host (91.197.85.151): Network is unreachable
root@Vexpress:/ telnet 10.0.2.15
telnet: can't connect to remote host (10.0.2.15): Network is unreachable
```
I assumed that at least plain TCP connections should have worked in this mode? If so, there is something missing in my configuration, and the magical thing is what I'm looking for...

jbech-linaro, 2017-01-13:
I'm afraid I have no solution for you @msa2 , but indeed it would be nice to get this working since it's pretty useful to have network connection from the QEMU environment.

msa2, 2017-01-13:
Some qemu networking docs seems imply that dhcp is used. The root file system does not contain any dchp client program (dhclient needed?)

jbech-linaro, 2017-01-13:
That might very well be the case. I just tried on another device (real hardware), but otherwise same setup, i.e., same kernel, same root fs (busybox) and I get no IP-address. The only other devices I've used network with are HiKey and Raspberry Pi3, but in both cases there I've used another root fs (Debian/Ubuntu based) instead of the busybox initramfs based root fs.

jforissier, 2017-01-13:
Busybox comes with `udhcpc`. I'm using busybox on HiKey and I used to set up networking with:
```
ip link set eth0 up
udhcpc -i eth0 -s /etc/udhcp/simple.script
```
The file `/etc/udhcp/simple.script` comes from `busybox/examples/udhcp/simple.script`.

igoropaniuk, 2017-01-13:
@msa2
You receive this message because of set `CONFIG_IP_PNP_DHCP=y` in the default Kconfig (kernel built-in DHCP client). IP value is OK for this configuration(it's default subnet, which virtual DHCP server on your host is using). What you need to do is to call `udhcpc`, which Jerome mentioned
```
Sending discover...
Sending select for 10.0.2.15...
Lease of 10.0.2.15 obtained, lease time 86400
```

jbech-linaro, 2017-01-13:
@jforissier , that did it on the device I'm currently using (DRA7), thanks!

msa2, 2017-01-13:
Well, tricky to get the simple script (would need to add it to filelist). However, I just used `cat > simple.script` and pasted the content to terminal window. Then...
```
root@Vexpress:/root udhcpc -i eth0 -s /etc/udhcp/simple.script
udhcpc (v1.25.0.git) started
Sending discover...
Sending select for 10.0.2.15...
Lease of 10.0.2.15 obtained, lease time 86400
root@Vexpress:/root ifconfig
eth0      Link encap:Ethernet  HWaddr 52:54:00:12:34:56  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3540 (3.4 KiB)  TX bytes:1710 (1.6 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```
So, the recipe is not yet working for 'qemu' repo...

jbech-linaro, 2017-01-13:
> Well, tricky to get the simple script (would need to add it to filelist). However, I just used cat > simple.script and pasted the content to terminal window. 

Yeps, someone should add it to https://github.com/OP-TEE/build/blob/master/common.mk#L288 or probably even better https://github.com/linaro-swg/gen_rootfs/blob/master/filelist.txt. It might be that I'm "the someone" if no-one else do it before me :)

jforissier, 2017-01-13:
@msa2 did you try adding debug statements to `simple.script`? What happens if you run `ifconfig eth0 10.0.2.15` manually? Does the interface get the ip?

jforissier, 2017-01-13:
#msa2 BTW, you did not forget to `chmod +x simple.script`, did you? ;)

msa2, 2017-01-13:
Yes, it was executable.

HOWEVER, it did work when I actually moved the script into directory /etc/udhcp/ !!!
There must be some check for that, but it would have saved some time, if udhcp had given some error message...

Anyway, all seems ok, except the minor detail in QEMU docs (some places) that claim this is enabled by default without -netdev. This just gives...
```
root@Vexpress:/ udhcpc -i eth0 -s /etc/udhcp/simple.script
udhcpc: SIOCGIFINDEX: No such device
```
...but, with it does work

`		-netdev user,id=vmnic -device virtio-net-device,netdev=vmnic \
`
So, someone just needs to write simple instructions (and qemu possibly include this script file in root file system).

igoropaniuk, 2017-01-13:
I just followed the same steps, and faced the same issue with `/etc/udhcp/simple.script` 
I just tried to setup network manually:
```
$ ifconfig eth0 10.0.2.15 netmask 255.255.255.0
$ route add default gw 10.0.2.2
$ echo "nameserver 8.8.8.8" > /etc/resolv.conf
$ telnet <somewebserv> 80
```
This solution works fine

igoropaniuk, 2017-01-13:
@msa2 

here is it :)
https://github.com/OP-TEE/build/pull/104

msa2, 2017-01-13:
Issue can be closed then. But hopefully, somewhere documented so that the next person running into this, does not need to spend so much time to find the solution.

jforissier, 2017-01-18:
Closing this issue since https://github.com/OP-TEE/build/pull/104 has been merged (TL;DR: build with `QEMU_USERNET_ENABLE=y` and networking should just work).

closed 2017-01-18 by jforissier