OP-TEE/build #114: "repo sync" gives git working space in state "* (no branch)"

msa2, 2017-01-27:
Being a bit inexperienced git user, I get always tripped over with this? Is there some reason for this? Now I try to cope by doing "git branch master", "git checkout master", but still manage to get things messed up very often.

I would like, that when I dowload "rpi.xml", it would give the build/linux/etc. gits in branch selected state, and it would be nice if plain "git pull" would work, when I need the updates.

Or, do you have a "recipe" for working with these gits, when you have local mods for them?

igoropaniuk, 2017-01-27:
Hi @msa2,

Actually, what I do after `repo init`/`repo sync` is just perform set up of local master branch to track remote branch master from linaro-swg/optee for repositories I'm going to work with. After this you can use simple `git pull`:
```
$ git checkout -b master optee/master
$ git pull
```

jbech-linaro, 2017-01-27:
That is and has been the behavior of `repo` since the beginning from what I remember. The reason for that is that repo under the hood just checks out the commits (instead of branches).

> Or, do you have a "recipe" for working with these gits, when you have local mods for them?

For updating my entire forest in one go, I do this (I tend to do that on daily basis since it resolves eventual inter-dependencies between gits)
```bash
# -d here, will detach from your current branch in a git and force it
# back to what is stated in manifest
# -j3 is just to run 3 jobs in parallel
$ repo sync -j3 -d
```
In some rare cases I don't use `-d` and then it will keep using my own branches that I've checked out and not the ones stated in the manifest (xml) file. But it will also try to rebase you current branch, which might not be what you want, so ... word of warning and again, that is why I most often just `-d`.

If do work in a git, then I just simply check it out as a new branch
```bash
$ git checkout -b my_branch
# or if I strictly want to use a certain remote/branch as base for my local branch
$ git checkout -b mybranch optee/master
# or ... if I want to use the same name as the remote and at the same time track it.
$ git checkout -t optee/master
```

From there you can simply use `$ git pull` if you want to. Personally I most of update in two steps, 1) update the remote(s) and then 2) manually rebase.

```bash
$ git remote update
```

Then I rebase manually if needed
```bash
# Here I'm on a local branch
$ git rebase optee/master
```

If you really would like to have a branch in all gits after initializing repo, then you can do:
```bash
$ repo start <mybranchname> --all
```
And "swish" you'll have and be on the `<mybranchname>` in all gits.

In fact a lot of stuff can be done directly with repo. Personally besides checking out the entire forest and syncing it using repo, I tend to do everything else directly in each git with git commands. That seems to work for me, it's not very often I get into problems with either repo nor git.

msa2, 2017-01-27:
Thank you. I think I will at now use the simple `git checkout -b master optee/master` approach.

The `repo sync ...` has a problem that if I'm working on something, I don't want to be distracted in fixing some merge conflicts in other parts of the system.

jbech-optee-adm, 2017-02-21:
We're closing this issue since the question has been answered. If you however feel that you have additional questions or still thinks this is an issue, please feel free to re-open the issue again.

closed 2017-02-21 by jbech-optee-adm