OP-TEE/build #119: drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c (Invalid argument)

nikoskoutr, 2017-02-02:
Hello, I am trying to build OP-TEE following the isntructions provided for raspberry pi 3.
Whenever i repo sync i get the following:

Fetching project linux.git
Fetching projects: 100% (1/1), done.  
error: unable to create file drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c (Invalid argument)
error: unable to create file drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.h (Invalid argument)
Checking out files: 100% (54166/54166), done.
Traceback (most recent call last):
  File "/media/Gniar/devel/optee-ref/.repo/repo/main.py", line 531, in <module>
    _Main(sys.argv[1:])
  File "/media/Gniar/devel/optee-ref/.repo/repo/main.py", line 507, in _Main
    result = repo._Run(argv) or 0
  File "/media/Gniar/devel/optee-ref/.repo/repo/main.py", line 180, in _Run
    result = cmd.Execute(copts, cargs)
  File "/media/Gniar/devel/optee-ref/.repo/repo/subcmds/sync.py", line 821, in Execute
    project.Sync_LocalHalf(syncbuf, force_sync=opt.force_sync)
  File "/media/Gniar/devel/optee-ref/.repo/repo/project.py", line 1327, in Sync_LocalHalf
    self._InitWorkTree(force_sync=force_sync)
  File "/media/Gniar/devel/optee-ref/.repo/repo/project.py", line 2486, in _InitWorkTree
    raise GitError("cannot initialize work tree")
error.GitError: cannot initialize work tree

Also, in the linux directory, there is no git initialized, in contrast with the other directories.

jbech-linaro, 2017-02-02:
Hi @nikoskoutr ,

I have never seen that issue (I did a full setup from scratch locally last week) and our daily test build, which does clone **and** build everything (including kernel) has no errors, it's all green: https://travis-ci.org/OP-TEE/build

So, I'm suspecting something wrong locally on you computer. could it be that you haven't updated the "repo" binary for a while? Sounds like a long shot, but otherwise I have no good idea.

What if you try to clone it manually just to test. I.e.
```bash
$ git clone https://github.com/linaro-swg/linux.git
$ git checkout -t origin/electron752-rpi3-optee
```
Does that work?

vchong, 2017-02-02:
Did you change the source url for the linux repo? If so, try deleting below and do `repo sync` or `repo sync linux` again.
```
.repo/projects/linux.git
.repo/project-objects/linux.git
```

jforissier, 2017-02-02:
@nikoskoutr are you trying to build on Windows maybe? (see http://stackoverflow.com/questions/3689137/error-git-checkout-index-unable-to-create-file).

nikoskoutr, 2017-02-02:
I tried a manual git clone and i got the same error about the aux.c and aux.h files and it ended with an error.
Due to a power failure, i didn't have a chance to copy the results on the terminal, so now i am cloning again in order to provide the exact result.
I did not change the source url for the linux repo, i will try your sugestion though, after I have finished cloning.
I am building it on elementary OS, kernel version: 4.4.0-59.

I suspect something is wrong with my git? I will also try re-installing it, although the installation is fresh.
Thank you all for the quick and helpful responses!

jforissier, 2017-02-02:
@nikoskoutr maybe your version of Git is weird. You may try rebuilding it from source.

nikoskoutr, 2017-02-02:
I have tried other versions of git, but there was no difference. I tried to "touch aux.c", which created the file succesfully in my home directory,
but when i try to run the same command "touch aux.c" inside the directory that it is supposed to be i get the error:
"touch: setting times of 'aux.c': No such file or directory"

By the way the result from the git clone is:
```Cloning into 'linux'...
remote: Counting objects: 5123661, done.
remote: Compressing objects: 100% (5018/5018), done.
remote: Total 5123661 (delta 4047), reused 456 (delta 456), pack-reused 5118184
Receiving objects: 100% (5123661/5123661), 1.66 GiB | 1.41 MiB/s, done.
Resolving deltas: 100% (4231413/4231413), done.
Checking connectivity... done.
error: unable to create file drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c (Invalid argument)
error: unable to create file drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.h (Invalid argument)
Checking out files: 100% (56252/56252), done.
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry the checkout with 'git checkout -f HEAD'
```

vchong, 2017-02-02:
What is /media/Gniar? As @jforissier had mentioned, are you maybe building on a virtual linux machine hosted on windows and /media/Gniar is a shared drive from/on the host?

igoropaniuk, 2017-02-02:
Hi @nikoskoutr 
In case if NTFS/FAT32 partition is mounted in /media/Gniar, here is an answer:
https://github.com/raspberrypi/linux/issues/1254
http://web.archive.org/web/20120414111738/http://www.blindedbytech.com/2006/11/16/forbidden-file-and-folder-names-on-windows

BR,
Igor

nikoskoutr, 2017-02-02:
@vchong Thanks a lot! The external drive Gniar was previously used for windows storage, so it is ntfs.
As i am writing this i am moving the repo directory to a compatible ext4 file system and i will run it again.

Thank you all for your responses, and i am sorry for wasting your time on something that was my mistake.

Best regards
Nick

closed 2017-02-02 by nikoskoutr