OP-TEE/build #121: OP-TEE with Android on RPI3

kdy1029, 2017-02-07:
Hello,
I have a question about running Android with OP-TEE on Raspberry Pi 3. I heard that OP-TEE supports both 32 bit and 64 bit, but it seems that OP-TEE for RPI3 only supports 64 bit. Running 32 bit OS would make me simple because of drivers, and so on. Is there any Android code like Hikey? Could you give me any advice?

Thanks,
D

jenswi-linaro, 2017-02-08:
When you say 32-bit OS I assume it's OP-TEE in secure world you're talking about (there's an OS, Linux, in normal world also). It should be fairly easy to make OP-TEE run in 32-bit mode, aka Aarch32, on RPI3. It's just that someone has to take the time to fix it.

Regarding Android and RPI3, maybe @vchong has something to add there.

vchong, 2017-02-09:
There's no port for OP-TEE on Android RPI3 atm. To do the port can be quite a bit of work. At a very high level, you'll have to figure out what's added to https://android.googlesource.com/device/linaro/hikey to integrate op-tee into aosp. You can use https://github.com/kuscsik/device-linaro-hikey-1/commits/aosp_optee as a reference. Then port those additions to your aosp source code for rpi3, e.g. https://github.com/android-rpi/device_brcm_rpi3 or wherever you get yours from, and see if you can get it to build and run without errors. Probably some effort will also be required to overcome selinux enforcement so that the trusted applications can run properly. Finally, please contribute your patches back after you get it to work.

kdy1029, 2017-02-10:
Thank you for replies. I will try to do that and upload it if I'm done.

closed 2017-02-10 by kdy1029

Tigerly, 2017-11-12:
@kdy1029 hello, just wondering whether we can have op-tee with android on raspberry pi 3 now?
Thanks in advance.

jbech-linaro, 2017-11-13:
Hi @kdy1029 , not what I'm aware of unfortunately.

kdy1029, 2017-11-13:
@Tigerly Sorry, we changed to udoo for some reasons.