OP-TEE/build #122: Status of 'electron752-rpi3-4.8-optee' branch?

msa2, 2017-02-20:
I wanted to use a newer kernel and noticed that rpi3-4.8-optee branch. I checked out, and it seems to build, but something does not seem be quite right

```
[   43.519351] console [ttyS0] enabled
[   43.582090] of_cfs_init
[   43.624720] of_cfs_init: OK
[   43.745019] EXT4-fs (mmcblk0p2): couldn't mount as ext3 due to feature incompatibilities
[   43.886243] EXT4-fs (mmcblk0p2): couldn't mount as ext2 due to feature incompatibilities
[   44.303650] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[   44.436369] VFS: Mounted root (ext4 filesystem) on device 179:2.
[   44.588474] devtmpfs: mounted
[   44.654420] Freeing unused kernel memory: 832K (ffffffc000820000 - ffffffc0008f0000)
```
...and tftp/nfs boot crashes
```
[   43.906105] console [ttyS0] enabled
[   43.968838] of_cfs_init
[   44.011440] of_cfs_init: OK
[  155.149657] VFS: Unable to mount root fs via NFS, trying floppy.
[  155.249703] List of all partitions:
[  155.307025] 0100            4096 ram0  (driver?)
[  155.382591] 0101            4096 ram1  (driver?)
[  155.458176] 0102            4096 ram2  (driver?)
[  155.533660] 0103            4096 ram3  (driver?)
[  155.609060] 0104            4096 ram4  (driver?)
[  155.684529] 0105            4096 ram5  (driver?)
[  155.760380] 0106            4096 ram6  (driver?)
[  155.835840] 0107            4096 ram7  (driver?)
[  155.911273] 0108            4096 ram8  (driver?)
[  155.986943] 0109            4096 ram9  (driver?)
[  156.062522] 010a            4096 ram10  (driver?)
[  156.139407] 010b            4096 ram11  (driver?)
[  156.216278] 010c            4096 ram12  (driver?)
[  156.293173] 010d            4096 ram13  (driver?)
[  156.370141] 010e            4096 ram14  (driver?)
[  156.446986] 010f            4096 ram15  (driver?)
[  156.523978] b300         7761920 mmcblk0  driver: mmcblk
[  156.611015]   b301           32768 mmcblk0p1 000f3789-01
[  156.697788]   b302         7728128 mmcblk0p2 000f3789-02
[  156.784435] No filesystem could mount root, tried:  nfs
[  156.869985] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(2,0)
[  157.004190] CPU: 1 PID: 1 Comm: swapper/0 Not tainted 4.8.15-v8 #17
[  157.105863] Hardware name: Raspberry Pi 3 Model B (DT)
[  157.189194] Call trace:
[  157.229033] [<ffffff80080899b0>] dump_backtrace+0x0/0x1d8
[  157.316697] [<ffffff8008089bac>] show_stack+0x24/0x30
[  157.398709] [<ffffff80083d992c>] dump_stack+0xa4/0xe0
[  157.480715] [<ffffff80081833cc>] panic+0x104/0x244
[  157.558515] [<ffffff8008821208>] mount_block_root+0x20c/0x258
[  157.651790] [<ffffff8008821490>] mount_root+0x11c/0x134
[  157.736604] [<ffffff8008821618>] prepare_namespace+0x170/0x1b8
[  157.831288] [<ffffff8008820dbc>] kernel_init_freeable+0x244/0x268
[  157.930250] [<ffffff800867fff0>] kernel_init+0x18/0x108
[  158.015058] [<ffffff8008083680>] ret_from_fork+0x10/0x50
[  158.101210] SMP: stopping secondary CPUs
[  158.164915] Kernel Offset: disabled
[  158.221435] Memory Limit: none
[  158.271024] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(2,0)
```
Any ideas what to fix to get this working? For NFS I'm trying to use fixed ip in the boot params (this worked just fine with the older electron branch).

jbech-linaro, 2017-02-20:
@igoropaniuk , you never got anywhere with this branch? You opened an issue with RPi3 kernel developers, but there is not fix yet? (FIQ/IRQ issue?)

igoropaniuk, 2017-02-20:
hi @msa2,
I've already spent quite a lot of time on trying to fix this, but still without any success. There is an issue in enumeration of usb root host by dwc2, that's why networking doesn't work (smsc95xx USB 2.0 Ethernet)

More details here https://github.com/raspberrypi/linux/issues/1779

BTW, we've already discussed usage of 32bit linux kernel by default in our rpi3 setup (as ARM64 isn't officially supported for RaspberryPi) as a possible way to avoid facing such issues (and that are not somehow directly related to OP-TEE). 

@jbech-linaro, right, no any updates since our last discussion. I think it will be better if I delete this branch for now to avoid any confusion in future.

jbech-linaro, 2017-02-21:
> I think it will be better if I delete this branch for now to avoid any confusion in future.

I agree, let's remove it. @jforissier , would you mind? I.e, delete the `rpi3-4.8-optee` branch in our kernel git?

jforissier, 2017-02-21:
I see a branch called electron752-rpi3-4.8-optee, is that the one? https://github.com/linaro-swg/linux/tree/electron752-rpi3-4.8-optee

jbech-linaro, 2017-02-21:
Yes, that is the one. Now it feels like I'm a bit lazy here. I could have just deleted it myself, that would save you some time :)

jforissier, 2017-02-21:
@jbech-linaro np :) the branch is gone now.

Electron752, 2017-02-22:
Hi, I saw a cross reference to this topic.  It's too bad you guys deleted the branch!

Anyway, I did some testing today and network boot is working just fine with the dwc_otg usb driver on the pi3 in arm64.  The only trick I had to do was force nfs version 3 in the command line file.   The default seems to be version 2 and it doesn't appear to be working correctly.   I don't think this is RPI specific.

Anyway, here is the cmdline.txt that I used:
coherent_pool=4M console=tty1 root=/dev/nfs ro nfsroot=192.168.1.100:/srv/pi9,vers=3 ip=192.168.1.35

vchong, 2017-02-22:
Just fyi if the maintainers are interested in bringing back the branch and nobody else has a backup, deleted branches can be revived by contacting support@github.com.

igoropaniuk, 2017-02-22:
Hi @Electron752,
Anyway, I've forked this branch before you've backported dwc_otg to 4.8 kernel

BTW, have you ever faced usb root host enumeration issues with backported dwc_otg on 4.8 kernel I've described in https://github.com/raspberrypi/linux/issues/1779?

@msa2 
Could you please post here the whole boot output from rpi3 dmesg or try to check if there is any mention of smsc95xx. Something like this?
```
[   155.149657] smsc95xx v1.0.4
[   155.149657] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:7c:97:f6
```

Electron752, 2017-02-22:
@igoropaniuk 

No, I never saw issues quite like that.  Usually, it either works or it doesn't.  Not the random behavior you described.   

I can only think of two things that could be causing that:

1. You are missing the RPI power management driver.  I think the downstream source changed to this  back around October or so.  If you are missing it, the usb won't be powered correctly but you may sometimes see it work just because the firmware powered it for awhile to use the USB for booting.
2. You simply have a board with some loose connections and need to buy a new RPI.

When I forgot to have the vers=3 in the cmdline.txt, I would indeed detect the smsc95xx just fine.  I would just fail after trying to mount the root filesystem several times.  I'm thinking ver 2 doesn't work with 64 bit period and it isn't RPI specific.

jbech-linaro, 2018-08-24:
I'm closing the ticket, either because it has already been answered or that it is no longer relevant or it could be lack of response from the author. Having that said, feel free to re-open the ticket if you have more to add to the ticket.

Also, we're about to step up the 4.14 kernel pretty soon, please see:
https://github.com/OP-TEE/build/pull/269
https://github.com/OP-TEE/manifest/pull/119
https://github.com/OP-TEE/optee_os/pull/2394

closed 2018-08-24 by jbech-linaro