OP-TEE/build #129: Any place to download a stable version?

hypersimple, 2017-03-10:
Any place to download a stable version for the whole system?

When I check out from this repo to build, lots of files under "edk2/BaseTools/Source/C/VfrCompile/Pccts/" use the wrong header path. It is painful to fix them one by one.

vchong, 2017-03-10:
Try checking out the 2.3.0 tag maybe? But we're not aware of any problems atm. What kind of errors are you getting? Can you list steps to reproduce?

hypersimple, 2017-03-10:
How to checkout the 2.3.0 tag entirely? Could you show the command? Thanks a lot!

There are a number of errors. Maybe because the build process mixes with the files on my own machine. For example:

err.c:(.text+0x69c): multiple definition of `zzsyn'

As it uses "/usr/include/err.h" on my 64-bit Ubuntu.

Also, some ".h" files originally under "op-tee/edk2/BaseTools/Source/C/VfrCompile/Pccts/antlr" seem to be moved to "../h/", and "../support/set/" (for set.h).

vchong, 2017-03-10:
That shouldn't happen. Sounds like something wrong with your build environment. Are you following the steps here: https://github.com/OP-TEE/build#7-get-and-build-the-solution?

hypersimple, 2017-03-10:
Yes. Exactly follow the instructions since they are not complex.
I previously use something like "arm-linux-gnueabihf-gcc" to build ARM-related stuff. Do not know if this will affect the building process.

vchong, 2017-03-10:
Patch submitted: https://github.com/OP-TEE/build/pull/130

jbech-optee-adm, 2017-10-21:
We are closing this issue/question, why?
 - We believe the question has been answered.
 - Quite often when we have answered the questions, the issuer forget to close the issue which results in many open issues for us which makes it hard to keep track of the ones actually waiting for an answer.
 - In some cases we ask follow-up questions, but do not hear back from the one creating the issue.

If you however feel that you have additional questions or still thinks this is an issue, then please feel free to re-open the issue again. When closing issues we will only show this entire message once.

// OP-TEE admins

closed 2017-10-21 by jbech-optee-adm