OP-TEE/build #143: Is it possible to build the RPI3 OP-TEE version as 32 bit?

dnovick, 2017-05-05:
So, I'm using the RPI3 for a crypto demo with ARM TZ temporarily, while I get some issues resolved with a couple of Cortex A-9 platforms that I want to use long term. In addition to using the RPI3 for a demo, I want to use the implementation as a stable comparison point to assist in debugging the other platforms (yeah, I know there are differences). 

Both of my other platforms are v7 arch...and support only 32 bit builds. If possible, I'd like to build the RPI3 port as 32 bit as well. It will make comparison much easier. Is this possible? I took a stab at changing the 4 overrides at the top of the top level makefile from 64 to 32, but that just called the build to barf. 

So, if it is supported, what steps must I follow in order to build a 32 bit version of both secure and non-secure SW? Is this documented anywhere?

Thanks,
Dave

dnovick, 2017-05-05:
So, I went ahead and cleaned...and then built again with the 4 overrides set to 32. This time it looked like it was going to build (strange)..but it blew up when trying to assemble linux kernel/bounds.s with the following error:

arm-linux-gnueabihf-gcc: error: unrecognized command line option '-mgeneral-regs-only'.

This looks like a makefile issue where the 32 bit gcc doesn't understand that flag, but the 64 bit one does.

I haven't spent any time yet finding exactly where in the make hierarchy this issue is rooted...I figured I'd just post details first, in case 1) 32 bit builds aren't really supported and/or 2) there is something I can or should do to successfully build 32 bit binaries.

Thanks,
Dave

jforissier, 2017-05-05:
Hi @dnovick,

It's quite possible Linux does not support 32-bit for RPI3 (I don't know, but I do know it's the case for HiKey for instance). Even so, you may run 32-bit secure world binaries (OP-TEE OS and TAs) with a 64-bit Linux kernel, if you adjust `build/rpi3.mk` as needed.

dnovick, 2017-05-06:
OK...Thanks for the response.

closed 2017-05-06 by dnovick