OP-TEE/build #144: HiKey: build fails with the latest changes in the grub master

igoropaniuk, 2017-05-10:
With the latest changes in the `master` branch in [git://git.savannah.gnu.org](git://git.savannah.gnu.org), which we are using by default in [hikey.xml](https://github.com/OP-TEE/manifest/blob/master/hikey.xml],) HiKey build fails. Latest stable revision (tag `2.02`) is built successfully.

Error message:
```
./grub-mkimage: info: locating grub_file_get_device_name at 0x72e0 (0x400).
./grub-mkimage: info: locating grub_efi_print_device_path at 0x28cc (0x400).
./grub-mkimage: info: locating grub_partition_iterate at 0x9d50 (0x400).
./grub-mkimage: info: locating grub_boot_time_head at 0x19418 (0xd5a8).
./grub-mkimage: info: locating grub_vsnprintf at 0x93e0 (0x400).
./grub-mkimage: error: undefined symbol grub_arch_sync_dma_caches.
Makefile:274: recipe for target 'grub' failed
```

I suppose this [commit](http://git.savannah.gnu.org/cgit/grub.git/commit/?id=265292f2b0da0eb414c409871ba0f94a99ec33c1) causes this issue, but not sure (I didn't run git bisect)

igoropaniuk, 2017-05-10:
I suggest using `2.02 `tag instead until it's fixed

jbech-linaro, 2017-05-10:
> I suggest using 2.02tag instead until it's fixed

Agree, would you mind to send an update to the xml-files (not-stable)?

igoropaniuk, 2017-05-11:
@jbech-linaro sure, PR will be created in a couple of minutes

closed 2017-05-12 by jforissier