OP-TEE/build #146: OP-TEE build failure for HiKey board

ykbs0903, 2017-05-21:
Hi, 
I have encountered an error during OP-TEE build (for HiKey debian)

After "#make all run" command in "_7.5 Build the solution_" section (https://github.com/OP-TEE/build),
i got the message below.

**< build log >**
...
...
...
make[3]: Entering directory '/root/devel/optee/optee_test/ta/socket'
  CC      /root/devel/optee/build/../optee_test/out/ta/socket/ta_entry.o
  CC      /root/devel/optee/build/../optee_test/out/ta/socket/user_ta_header.o
  CPP     /root/devel/optee/build/../optee_test/out/ta/socket/ta.lds
  LD      /root/devel/optee/build/../optee_test/out/ta/socket/873bcd08-c2c3-11e6-a937-d0bf9c45c61c.elf
  OBJDUMP /root/devel/optee/build/../optee_test/out/ta/socket/873bcd08-c2c3-11e6-a937-d0bf9c45c61c.dmp
  OBJCOPY /root/devel/optee/build/../optee_test/out/ta/socket/873bcd08-c2c3-11e6-a937-d0bf9c45c61c.stripped.elf
  SIGN    /root/devel/optee/build/../optee_test/out/ta/socket/873bcd08-c2c3-11e6-a937-d0bf9c45c61c.ta
make[3]: Leaving directory '/root/devel/optee/optee_test/ta/socket'
make[2]: Leaving directory '/root/devel/optee/optee_test/ta'
make[1]: Leaving directory '/root/devel/optee/optee_test'
make -C /root/devel/optee/build/../hello_world HOST_CROSS_COMPILE="/usr/bin/ccache /root/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" TA_CROSS_COMPILE="/usr/bin/ccache /root/devel/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" TA_DEV_KIT_DIR=/root/devel/optee/build/../optee_os/out/arm/export-ta_arm32 TEEC_EXPORT=/root/devel/optee/build/../optee_client/out/export
make[1]: Entering directory '/root/devel/optee/hello_world'
make -C host CROSS_COMPILE="/usr/bin/ccache /root/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-"
make[2]: Entering directory '/root/devel/optee/hello_world/host'
/usr/bin/ccache /root/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-gcc -Wall -I../ta/include -I/root/devel/optee/build/../optee_client/out/export/include -I./include   -c -o main.o main.c
/usr/bin/ccache /root/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-gcc -lteec -L/root/devel/optee/build/../optee_client/out/export/lib -o hello_world main.o
make[2]: Leaving directory '/root/devel/optee/hello_world/host'
make -C ta CROSS_COMPILE="/usr/bin/ccache /root/devel/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-"
make[2]: Entering directory '/root/devel/optee/hello_world/ta'
  CC      hello_world_ta.o
  CC      user_ta_header.o
  CPP     ta.lds
  LD      8aaaf200-2450-11e4-abe2-0002a5d5c51b.elf
  OBJDUMP 8aaaf200-2450-11e4-abe2-0002a5d5c51b.dmp
  OBJCOPY 8aaaf200-2450-11e4-abe2-0002a5d5c51b.stripped.elf
  SIGN    8aaaf200-2450-11e4-abe2-0002a5d5c51b.ta
make[2]: Leaving directory '/root/devel/optee/hello_world/ta'
make[1]: Leaving directory '/root/devel/optee/hello_world'
dpkg-deb: building package 'op-tee' in 'optee_2.3-0.deb'.
make: *** No rule to make target 'run'.  Stop.
root@kbs0903-VirtualBox:~/devel/optee/build# 

Does anyone know how to fix this problem?
Thanks.

vchong, 2017-05-21:
I think section 7.5 shouldn't only say `make all`, not `make all run`.

`run` is not a valid target for hikey debian. While we fix the doc, you can safely ignore the error and proceed to https://github.com/OP-TEE/build/blob/master/docs/hikey.md#51-building-the-op-tee-debian-based-setup.

jforissier, 2017-05-21:
That's right, `run` is for QEMU only.

ykbs0903, 2017-05-21:
Thank you for answer :)

closed 2017-05-21 by ykbs0903