OP-TEE/build #153: "make toolchains" downloads 64bit tools even for 32bit Ubuntu build machine

P1119r1m, 2017-06-08:
Next steps were successfully executed:

> ## 7.4 Get the toolchains
> In OP-TEE we're using different toolchains for different targets (depends on
> ARMv7-A ARMv8-A 64/32bit solutions). In any case start by downloading the
> toolchains by:
> ```bash
> $ cd build
> $ make toolchains
> ```

but when after that we're trying to build the solution:

> ## 7.5 Build the solution
> We've configured our repo manifests, so that repo will always automatically
> symlink the `Makefile` to the correct device specific makefile, that means that
> you simply start the build by running:
> 
> ```bash
> $ make
> ``` 

process fails, because downloaded tools (like compiler) are for 64bit build platform.

P.S. It fails with following message:

> "/bin/bash: /home/user/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-gcc: cannot execute binary file".

Additional info:

> $ readelf -h /home/user/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-gcc
> ELF Header:
>   Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00 
>   Class:                             ELF64
>   Data:                              2's complement, little endian
>   Version:                           1 (current)
>   OS/ABI:                            UNIX - System V
>   ABI Version:                       0
>   Type:                              EXEC (Executable file)
>   Machine:                           Advanced Micro Devices X86-64
>   Version:                           0x1
>   Entry point address:               0x40385e
>   Start of program headers:          64 (bytes into file)
>   Start of section headers:          926032 (bytes into file)
>   Flags:                             0x0
>   Size of this header:               64 (bytes)
>   Size of program headers:           56 (bytes)
>   Number of program headers:         10
>   Size of section headers:           64 (bytes)
>   Number of section headers:         30
>   Section header string table index: 29

> $ uname -m
> i686

jbech-linaro, 2017-06-08:
From what I know we have only been building the code using 64-bit OS. I guess it would be possible to download toolchains built for 32-bit environments and patch our current build system. Having that said it's nothing that we have on our radar for the moment. Do you think you could have a look at it and see whether you could come up with a patch for us?

P1119r1m, 2017-06-08:
Thank you! I got your point! I will think about that in advance!

vchong, 2017-06-08:
@P1119r1m Try the 32bit tools below. They might just work.

toolchain.mk
SRC_AARCH64_GCC
http://releases.linaro.org/components/toolchain/binaries/6.3-2017.05/aarch64-linux-gnu/gcc-linaro-6.3.1-2017.05-i686_aarch64-linux-gnu.tar.xz

SRC_AARCH32_GCC
http://releases.linaro.org/components/toolchain/binaries/6.3-2017.05/arm-linux-gnueabihf/gcc-linaro-6.3.1-2017.05-i686_arm-linux-gnueabihf.tar.xz

LEGACY_SRC_AARCH64_GCC
http://releases.linaro.org/components/toolchain/binaries/4.9-2017.01/aarch64-linux-gnu/gcc-linaro-4.9.4-2017.01-i686_aarch64-linux-gnu.tar.xz

vchong, 2017-06-08:
If the above doesn't work, maybe try older versions below as well.

AARCH32_PATH 			?= $(TOOLCHAIN_ROOT)/aarch32
AARCH32_CROSS_COMPILE 		?= $(AARCH32_PATH)/bin/arm-linux-gnueabihf-
AARCH32_GCC_VERSION 		?= gcc-linaro-arm-linux-gnueabihf-4.9-2014.08_linux
SRC_AARCH32_GCC 		?= http://releases.linaro.org/archive/14.08/components/toolchain/binaries/${AARCH32_GCC_VERSION}.tar.xz

AARCH64_PATH 			?= $(TOOLCHAIN_ROOT)/aarch64
AARCH64_CROSS_COMPILE 		?= $(AARCH64_PATH)/bin/aarch64-linux-gnu-
AARCH64_GCC_VERSION 		?= gcc-linaro-aarch64-linux-gnu-4.9-2014.08_linux
SRC_AARCH64_GCC 		?= http://releases.linaro.org/archive/14.08/components/toolchain/binaries/${AARCH64_GCC_VERSION}.tar.xz

LEGACY_AARCH64_PATH             ?= $(TOOLCHAIN_ROOT)/aarch64-legacy
LEGACY_AARCH64_CROSS_COMPILE    ?= $(LEGACY_AARCH64_PATH)/bin/aarch64-linux-gnu-
LEGACY_AARCH64_GCC_VERSION      ?= gcc-linaro-aarch64-linux-gnu-4.9-2014.08_linux
LEGACY_SRC_AARCH64_GCC          ?= http://releases.linaro.org/archive/14.08/components/toolchain/binaries/${LEGACY_AARCH64_GCC_VERSION}.tar.xz

P1119r1m, 2017-06-09:
@vchong Thanks for your advice, but probably the easiest way is to use 64bit OS. Moreover, recently, while resolving problems during installation of some auxiliary software packages, my VitrualBox OS had crashed :(

jbech-optee-adm, 2017-10-21:
We are closing this issue/question, why?
 - We believe the question has been answered.
 - Quite often when we have answered the questions, the issuer forget to close the issue which results in many open issues for us which makes it hard to keep track of the ones actually waiting for an answer.
 - In some cases we ask follow-up questions, but do not hear back from the one creating the issue.

If you however feel that you have additional questions or still thinks this is an issue, then please feel free to re-open the issue again. When closing issues we will only show this entire message once.

// OP-TEE admins

closed 2017-10-21 by jbech-optee-adm