OP-TEE/build #157: Documentation for OP-TEE architecture was not found

P1119r1m, 2017-07-03:
Dear community. I'm sorry that I am posting question here, but I haven't found better place nor on the [main page](https://www.op-tee.org), nor on the github official page.

The question is: "Is there any documentation about how different OP-TEE parts interconnected?".

I am about the documentation which describes the way that, for example, uboot starts Linux, then it starts OP-TEE-OS and so on. Also it is interesting to know why should we have supplicant and how it works in system.

Thanks a lot!

jbech-linaro, 2017-07-04:
> The question is: "Is there any documentation about how different OP-TEE parts interconnected?".

Yes, but it's scattered in different places.
- https://youtu.be/kk3_DUMJrTI
- http://connect.linaro.org/resource/bkk16/bkk16-110/
- https://www.youtube.com/watch?v=Fksx4-bpHRY
- https://github.com/jbech-linaro/comcast_cryptoapi_ta_example <- Deprecated, but the image wraps it up.

I hope this helps!
// Joakim

P1119r1m, 2017-07-04:
@jbech-linaro Thank you! It is really better than nothing.

closed 2017-10-21 by jbech-optee-adm