OP-TEE/build #163: Correct output for hikey debian dpkg install

acarb95, 2017-07-11:
I've built and flashed OP-TEE 2.5.0 on a Linaro Hikey 2GB board. I've copied over the .deb files (optee_2.5.0-0.deb and linux-image-4.9.0-optee-rpb_4.9.0-optee-rpb-8_amd64.deb) and then I run `dpkg --force-all -i optee_2.5.0-0.deb` and `dpkg --force-all -i linux-image-4.9.0-optee-rpb_4.9.0-optee-rpb-8_amd64.deb`.

This is the output I get for optee:
```
Selecting previously unselected package op-tee.
(Reading database ... 62583 files and directories currently installed.)
Preparing to unpack optee_2.5.0-0.deb ...
Unpacking op-tee (2.5.0-0) ...
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: trying to overwrite '/usr/lib/aarch64-linux-gnu/libteec.so.1.0', which is also in package1
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: trying to overwrite '/usr/lib/aarch64-linux-gnu/libteec.so.1', which is also in package l1
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: trying to overwrite '/usr/bin/tee-supplicant', which is also in package tee-supplicant 2.1
Setting up op-tee (2.5.0-0) ...
```

This is the output I get for the linux-image:
```
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: package architecture (amd64) does not match system (arm64)
Selecting previously unselected package linux-image-4.9.0-optee-rpb.
(Reading database ... 62601 files and directories currently installed.)
Preparing to unpack linux-image-4.9.0-optee-rpb_4.9.0-optee-rpb-8_amd64.deb ...
Unpacking linux-image-4.9.0-optee-rpb (4.9.0-optee-rpb-8) ...
Setting up linux-image-4.9.0-optee-rpb (4.9.0-optee-rpb-8) ...
update-initramfs: Generating /boot/initrd.img-4.9.0-optee-rpb
Copying hi6220-hikey.dtb as /boot/hi6220-hikey.dtb-4.9.0-optee-rpb
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-4.9.0-optee-rpb
Found initrd image: /boot/initrd.img-4.9.0-optee-rpb
Found devicetree image: /boot/hi6220-hikey.dtb-4.9.0-optee-rpb
Found linux image: /boot/vmlinuz-4.4.0-135-arm64
Found initrd image: /boot/initrd.img-4.4.0-135-arm64
Found devicetree image: /boot/hi6220-hikey.dtb-4.4.0-135-arm64
[  588.215649] FAT-fs (mmcblk0p1): bogus number of reserved sectors
[  588.232546] EXT4-fs (mmcblk0p1): VFS: Can't find ext4 filesystem
[  588.244055] EXT4-fs (mmcblk0p1): VFS: Can't find ext4 filesystem
[  588.255728] EXT4-fs (mmcblk0p1): VFS: Can't find ext4 filesystem
[  588.279353] FAT-fs (mmcblk0p1): bogus number of reserved sectors
[  588.294811] ntfs: (device mmcblk0p1): read_ntfs_boot_sector(): Primary boot sector is invalid.
[  588.303405] ntfs: (device mmcblk0p1): read_ntfs_boot_sector(): Mount option errors=recover not used. .
[  588.315627] ntfs: (device mmcblk0p1): ntfs_fill_super(): Not an NTFS volume.
[  588.413527] FAT-fs (mmcblk0p2): bogus number of reserved sectors
[  588.430380] EXT4-fs (mmcblk0p2): VFS: Can't find ext4 filesystem
[  588.441835] EXT4-fs (mmcblk0p2): VFS: Can't find ext4 filesystem
[  588.453328] EXT4-fs (mmcblk0p2): VFS: Can't find ext4 filesystem
[  588.477019] FAT-fs (mmcblk0p2): bogus number of reserved sectors
[  588.488393] ntfs: (device mmcblk0p2): read_ntfs_boot_sector(): Primary boot sector is invalid.
[  588.497005] ntfs: (device mmcblk0p2): read_ntfs_boot_sector(): Mount option errors=recover not used. .
[  588.509200] ntfs: (device mmcblk0p2): ntfs_fill_super(): Not an NTFS volume.
[  588.605170] FAT-fs (mmcblk0p3): bogus number of reserved sectors
[  588.622174] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
[  588.633627] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
[  588.645091] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
[  588.668689] FAT-fs (mmcblk0p3): bogus number of reserved sectors
[  588.680054] ntfs: (device mmcblk0p3): read_ntfs_boot_sector(): Primary boot sector is invalid.
[  588.688580] ntfs: (device mmcblk0p3): read_ntfs_boot_sector(): Mount option errors=recover not used. .
[  588.700901] ntfs: (device mmcblk0p3): ntfs_fill_super(): Not an NTFS volume.
[  588.801464] FAT-fs (mmcblk0p4): bogus number of reserved sectors
[  588.818755] EXT4-fs (mmcblk0p4): VFS: Can't find ext4 filesystem
[  588.830474] EXT4-fs (mmcblk0p4): VFS: Can't find ext4 filesystem
[  588.842116] EXT4-fs (mmcblk0p4): VFS: Can't find ext4 filesystem
[  588.866028] FAT-fs (mmcblk0p4): bogus number of reserved sectors
[  588.967981] FAT-fs (mmcblk0p5): bogus number of reserved sectors
[  588.985052] EXT4-fs (mmcblk0p5): VFS: Can't find ext4 filesystem
[  588.996689] EXT4-fs (mmcblk0p5): VFS: Can't find ext4 filesystem
[  589.008284] EXT4-fs (mmcblk0p5): VFS: Can't find ext4 filesystem
[  589.032219] FAT-fs (mmcblk0p5): bogus number of reserved sectors
[  589.231501] FAT-fs (mmcblk0p7): bogus number of reserved sectors
[  589.248411] EXT4-fs (mmcblk0p7): VFS: Can't find ext4 filesystem
[  589.259853] EXT4-fs (mmcblk0p7): VFS: Can't find ext4 filesystem
[  589.271464] EXT4-fs (mmcblk0p7): VFS: Can't find ext4 filesystem
[  589.294928] FAT-fs (mmcblk0p7): bogus number of reserved sectors
[  589.436210] FAT-fs (mmcblk0p8): bogus number of reserved sectors
[  589.453274] EXT4-fs (mmcblk0p8): VFS: Can't find ext4 filesystem
[  589.464771] EXT4-fs (mmcblk0p8): VFS: Can't find ext4 filesystem
[  589.476201] EXT4-fs (mmcblk0p8): VFS: Can't find ext4 filesystem
[  589.499814] FAT-fs (mmcblk0p8): bogus number of reserved sectors
done
```

This doesn't seem like the correct output to me. Where could I be going wrong in the build process?

vchong, 2017-07-11:
Is everything working ok after you reboot? WARNING: Please flash the wifi fw .deb as well before rebooting.

acarb95, 2017-07-11:
I flashed the wifi fw .deb by using the samke dpkg command. Upon reboot I get a huge call trace and stack dump. 

```
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.1(release):fb1158a
NOTICE:  BL1: Built : 17:53:06, Jul 10 2017
NOTICE:  syspll frequency:1190494208Hz
NOTICE:  succeed to init lpddr3 rank0 dram phy
INFO:    lpddr3_freq_init, set ddrc 533mhz
INFO:    init ddr3 rank0
INFO:    ddr3 rank1 init pass
INFO:    lpddr3_freq_init, set ddrc 800mhz
INFO:    init ddr3 rank0
INFO:    ddr3 rank1 init pass
INFO:    Samsung DDR
NOTICE:  BL1: Booting BL2
NOTICE:  acpu_dvfs_set_freq: set acpu freq success!NOTICE:  BL2: v1.1(release):fb1158a
NOTICE:  BL2: Built : 17:53:06, Jul 10 2017
NOTICE:  BL1: Booting BL3-1
NOTICE:  BL3-1: v1.1(release):fb1158a
NOTICE:  BL3-1: Built : 17:53:07, Jul 10 2017
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: CFG_SHMEM_START type NSEC_SHM 0x3ee00000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: CFG_TA_RAM_START type TA_RAM 0x3f200000 size 0x00e00000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: VCORE_UNPG_RW_PA type TEE_RAM_RW 0x3f03f000 size 0x000c1000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: VCORE_UNPG_RX_PA type TEE_RAM_RX 0x3f000000 size 0x0003f000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: SPI_BASE type IO_NSEC 0xf7000000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: GPIO6_BASE type IO_NSEC 0xf7000000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:537: Physical mem map overlaps 0xf7000000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: PMX1_BASE type IO_NSEC 0xf7000000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:537: Physical mem map overlaps 0xf7000000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: PMX0_BASE type IO_NSEC 0xf7000000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:537: Physical mem map overlaps 0xf7000000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: PERI_BASE type IO_NSEC 0xf7000000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:537: Physical mem map overlaps 0xf7000000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: PMUSSI_BASE type IO_NSEC 0xf8000000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:524: CONSOLE_UART_BASE type IO_NSEC 0xf7000000 size 0x00200000
DEBUG:   [0x0] TEE-CORE:add_phys_mem:537: Physical mem map overlaps 0xf7000000
DEBUG:   [0x0] TEE-CORE:verify_special_mem_areas:462: No NSEC DDR memory area defined
DEBUG:   [0x0] TEE-CORE:add_va_space:563: type RES_VASPACE size 0x01400000
DEBUG:   [0x0] TEE-CORE:add_va_space:563: type SHM_VASPACE size 0x01400000
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type IO_NSEC      va 0x3b400000..0x3b5fffff pa 0xf7000000..)
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type NSEC_SHM     va 0x3b600000..0x3b7fffff pa 0x3ee00000..)
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type TA_RAM       va 0x3b800000..0x3c5fffff pa 0x3f200000..)
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type RES_VASPACE  va 0x3c600000..0x3d9fffff pa 0x00000000..)
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type IO_NSEC      va 0x3da00000..0x3dbfffff pa 0xf8000000..)
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type SHM_VASPACE  va 0x3dc00000..0x3effffff pa 0x00000000..)
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type TEE_RAM_RX   va 0x3f000000..0x3f03efff pa 0x3f000000..)
DEBUG:   [0x0] TEE-CORE:dump_mmap_table:690: type TEE_RAM_RW   va 0x3f03f000..0x3f0fffff pa 0x3f03f000..)
DEBUG:   [0x0] TEE-CORE:init_xlation_table:414: xlat used: 1/5
DEBUG:   [0x0] TEE-CORE:init_xlation_table:414: xlat used: 2/5
INFO:    TEE-CORE: 
DEBUG:   [0x0] TEE-CORE:init_canaries:169: #Stack canaries for stack_tmp[0] with top at 0x3f081938
DEBUG:   [0x0] TEE-CORE:init_canaries:169: watch *0x3f08193c
DEBUG:   [0x0] TEE-CORE:init_canaries:169: #Stack canaries for stack_tmp[1] with top at 0x3f082178
DEBUG:   [0x0] TEE-CORE:init_canaries:169: watch *0x3f08217c
  Booting `Debian GNU/Linux'

Loading Linux 4.9.0-optee-rpb ...
Loading initial ramdisk ...
Loading devicetree ...
EFI stub: Booting Linux Kernel...
EFI stub: EFI_RNG_PROTOCOL unavailable, no randomness supplied
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map...
DEBUG:   TEE-CORE:generic_boot_cpu_on_handler:718: cpu 1: a0 0x0
DEBUG:   TEE-CORE:init_secondary_helper:703: Secondary CPU Switching to normal world boot
DEBUG:   TEE-CORE:generic_boot_cpu_on_handler:718: cpu 2: a0 0x0
DEBUG:   TEE-CORE:init_secondary_helper:703: Secondary CPU Switching to normal world boot
DEBUG:   TEE-CORE:generic_boot_cpu_on_handler:718: cpu 3: a0 0x0
DEBUG:   TEE-CORE:init_secondary_helper:703: Secondary CPU Switching to normal world boot
[... 406 lines cut ...]
[   31.849077]  (detected by 4, t=5252 jiffies, g=-18, c=-19, q=5369)
[   31.855270] Task dump for CPU 6:
[   31.858500] systemd-udevd   R  running task        0   255      1 0x00000002
[   31.865565] Call trace:
[   31.868020] [<ffff0000080860cc>] __switch_to+0x94/0xa8
[   31.873166] [<ffff00000810a4bc>] task_change_group_fair+0xb4/0xf0
[   31.879268] [<ffff0000080f45f8>] sched_change_group+0x50/0xd8
[   31.885020] [<ffff0000080f55dc>] cpu_cgroup_fork+0x4c/0x88
[   31.890511] [<0000000000000001>] 0x1
[   31.894088] Task dump for CPU 7:
[   31.897318] kworker/7:1     x    0   156      0 0x00000000
[   31.902824] Call trace:
[   31.905274] [<ffff0000080860cc>] __switch_to+0x94/0xa8
[   31.910420] [<ffff000008771f20>] __schedule+0x3e8/0x718
[   31.915653] rcu_sched kthread starved for 5195 jiffies! g18446744073709551598 c18446744073709551597 f1
[   31.927839] rcu_sched       S    0     7      2 0x00000000
[   31.933334] Call trace:
[   31.935783] [<ffff0000080860cc>] __switch_to+0x94/0xa8
[   31.940925] [<ffff000008771d50>] __schedule+0x218/0x718
[   31.946155] [<ffff000008772288>] schedule+0x38/0x98
[   31.951039] [<ffff000008775640>] schedule_timeout+0x1b0/0x3a0
[   31.956793] [<ffff00000813baec>] rcu_gp_kthread+0x48c/0x810
[   31.962372] [<ffff0000080eafac>] kthread+0xec/0x100
[   31.967255] [<ffff000008083680>] ret_from_fork+0x10/0x50
```

vchong, 2017-07-11:
Was the board working before you started installing the deb file?

Did you run `make flash` or `make recovery` first before installing the deb files?

If you ran `make flash`, can you please verify that `debian_system.img` was flashed successfully? If not successful, please try again using `make recovery`.

acarb95, 2017-07-11:
I ran `make recovery` before installing the deb files. The board works fine from what I can tell before I installed the deb files and rebooted.

vchong, 2017-07-11:
One last stab at this. Are you using a 4gb or 8gb board?

acarb95, 2017-07-11:
8 gb board

vchong, 2017-07-11:
Everything seems ok. :( If you've anything else connected to the board, e.g. hdmi monitor, mouse, keyboard, try removing them and test again? If Debian is not a hard requirement for you, perhaps you can consider trying the regular hikey build as well?

acarb95, 2017-07-12:
That worked! Why would the keyboard, mouse, and/or HDMI affect it?

vchong, 2017-07-12:
Not sure. :( Might be a compatibility issue. Try starting again with just the monitor plugged in. Then, after boot up is ok, try reconnecting the keyboard and mouse one by one to see who's the culprit, and then perhaps try a new brand/model. Also, if you're using an usb hub, try without it or a different hub.

vchong, 2017-07-12:
Do you maybe have both the microusb cable (the one next to the regular usb port) connected at the same time as the mouse/keyboard? If so, maybe also try booting up regularly with just the mouse/keyboard and without the microusb cable.

acarb95, 2017-07-12:
I'm using a wireless keyboard, so my best guess would be it's the keyboard. I'll still start one by one to confirm which component is causing it. 

I did still have the micro usb cable inserted with the keyboard/mouse when I booted. I'll try booting without that as well. 

I'll post my findings with the details of the problem thing (if it's the mouse, keyboard, or monitor) so others can find it, then I'll close the issue.

Thanks for your fast responses!

acarb95, 2017-08-14:
Yes, sorry, I forgot to post. I am able to reboot with both my wireless keyboard and mouse plugged in. The culprit seems to be the monitor. I also tried with just the monitor plugged in and I still get a kernel oops and stack trace dump.

I am using an amazon basic HDMI cord and a Samsung SyncMaster B2430 monitor. This setup worked for me when I was using optee 1.1. Do you have a list of monitors that work with the Hikey board and optee 2.5.0? A monitor is required for the project I'm working on.

vchong, 2017-08-15:
@acarb95 There is https://docs.google.com/spreadsheets/d/1aYNUlGx5J8RmfWrtML4lFJ-7tV8u-3AH_3IaiOlWfoE. Note that it's not comprehensive though. It might have something to do with the kernel being used as well. If you think something should work but is not (with the kernel), please submit a bug to http://bugs.96boards.org, but remove anything op-tee related. Thanks!

jbech-linaro, 2017-08-16:
Right, I can't use HDMI on my monitor at home with HiKey (Asus VG248QE), but on the monitor at work (Philips) it works fine. I think that was one of the first "bugs" I asked about when I got my HiKey 2 years ago. I think they told me that it is a hardware issue in HiKey and there is nothing to do about it. Some monitors work others don't :(

acarb95, 2017-08-16:
This particular monitor works with this Hikey board, I've used it before. It's only after I upgraded to OP-TEE 2.5.0 that it stopped working. I'll see if I can figure out if it's a kernel issue or not, gather as much information as possible, and file a bug. 

As for this issue, I think it's safe to close as my original question is answered. 

Thanks for the help!

closed 2017-10-21 by jbech-optee-adm