OP-TEE/build #166: Error "unrecognized command line option" occurs in generating root file system

yieux, 2017-07-18:
The issue occurs in the step  7.6 "Flash the device". 
`$ make flash`
When busybox is used to generate file system. I get some error from aarch64-linux-gnu-gcc for building  applets.o. 

```
The prerequisites of '__build' are being made.
Live child 0x18d4950 (applets/applets.o) PID 25942 
aarch64-linux-gnu-gcc: error: unrecognized argument in option '-mabi=aapcs-linux'
aarch64-linux-gnu-gcc: note: valid arguments to '-mabi=' are: ilp32 lp64
aarch64-linux-gnu-gcc: error: unrecognized command line option '-marm'; did you mean '-fasm'?
aarch64-linux-gnu-gcc: error: unrecognized command line option '-mthumb'
aarch64-linux-gnu-gcc: error: unrecognized command line option '-mthumb-interwork'
Reaping losing child 0x18d4950 PID 25942 
/home/yieux/Project/optee/busybox/scripts/Makefile.build:197: recipe for target 'applets/applets.o' failed
make[3]: *** [applets/applets.o] Error 1
Removing child 0x18d4950 PID 25942 from chain.
Reaping losing child 0x9f6770 PID 25934 
/home/yieux/Project/optee/busybox/Makefile:372: recipe for target 'applets_dir' failed
make[2]: *** [applets_dir] Error 2
Removing child 0x9f6770 PID 25934 from chain.
Reaping losing child 0x10bf6f0 PID 25373 
Makefile:112: recipe for target '_all' failed
make[1]: *** [_all] Error 2
Removing child 0x10bf6f0 PID 25373 from chain.
make[1]: Leaving directory '/home/yieux/Project/optee/busybox'
Reaping losing child 0x21f7850 PID 25351 
common.mk:134: recipe for target 'busybox-common' failed
make: *** [busybox-common] Error 2
```

1)The line 79 of generate-cpio-rootfs.sh
```
case $1 in
    "vexpress")
        CFLAGS=${CFLAGS-"-Wno-strict-aliasing -Wno-unused-result -marm -mabi=aapcs-linux -mthumb -mthumb-interwork -mcpu=cortex-a15"}
````
It looks like that the options such as mthumb is supported by only arm(aarch32) not arm64(aarch64). 

, how can I do this job for different processor, or what I did wrong？

If someone could help me with this issue I would be very thankful.

vchong, 2017-07-18:
Which platform are you building for? It's not supposed to go to `case` `vexpress` in `generate-cpio-rootfs.sh` for arm64(aarch64). Did you make any changes to the makefiles?

yieux, 2017-07-19:
Since target flash is not included in /build/Makefile, i add one line to qemu.mk
`-include mediatek.mk`
Then I get the above errors for  `make flash ` or  `make busybox`. 

As @vchong  said, I remove the additional line and execute `make busybox`. 

----------------------------------------------------------------------------------------------------------------------------------
For  system type, I select
```
ARM system type
> 1. Allow multiple platforms to be selected (ARCH_MULTIPLATFORM) (NEW)
```
And only the options:
`ARMv7 based platforms (Cortex-A, PJ4, Scorpion, Krait) (ARCH_MULTI_V7) --------- suggest to accept`  
`ARM Ltd. Versatile Express family (ARCH_VEXPRESS)  ------------------------------------ automatically accept`  
is selected as  CPU Core family.  

Now, busybox looks normal. 
----------------------------------------------------------------------------------------------------------
After that, I execute 
`make flash -f mediatek.mk `
and select platform  `ARMv8 software model (Versatile Express) (ARCH_VEXPRESS)`

However, I get the same errors with previous contents.

vchong, 2017-07-19:
You either use `qemu.mk` only, in which case there's no need to `make flash`, or use `mediatek.mk` only, and can then `make flash` to the board. Why are you trying to mix 2 different makefiles for 2 different platforms at the same time? You specify the one platform you want to use during step 7.3, i.e.

`repo init -u https://github.com/OP-TEE/manifest.git -m default.xml` -> for qemu
or
`repo init -u https://github.com/OP-TEE/manifest.git -m mt8173-evb.xml` -> for mediatek

yieux, 2017-07-19:
OK, I get it. 
Thanks for your help.~

jbech-optee-adm, 2017-10-21:
We are closing this issue/question, why?
 - We believe the question has been answered.
 - Quite often when we have answered the questions, the issuer forget to close the issue which results in many open issues for us which makes it hard to keep track of the ones actually waiting for an answer.
 - In some cases we ask follow-up questions, but do not hear back from the one creating the issue.

If you however feel that you have additional questions or still thinks this is an issue, then please feel free to re-open the issue again. When closing issues we will only show this entire message once.

// OP-TEE admins

closed 2017-10-21 by jbech-optee-adm