OP-TEE/build #197: Error while building optee - {arm-linux-gnueabihf-gcc failed: Exec format error .}

Souravpunoriyar, 2017-09-24:
I am trying to run optee using qemu.
I have followed exactly the same steps as mentioned in Readme.md.
But i am getting the following errors while building .
Could you please give me hint to build it properly ?

make -C /home/KELA/Desktop/projects/optee/build/../linux LOCALVERSION= CROSS_COMPILE="/usr/bin/ccache /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-" ARCH=arm
make[1]: Entering directory '/home/KELA/Desktop/projects/optee/linux'
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
scripts/kconfig/conf  --silentoldconfig Kconfig
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
  CHK     include/config/kernel.release
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
  CHK     include/generated/uapi/linux/version.h
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
  CHK     include/generated/utsrelease.h
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
  GEN     include/generated/mach-types.h
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
  CC      kernel/bounds.s
ccache: error: execv of /home/KELA/Desktop/projects/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf-gcc failed: Exec format error
Kbuild:45: recipe for target 'kernel/bounds.s' failed
make[2]: *** [kernel/bounds.s] Error 1
Makefile:1033: recipe for target 'prepare0' failed
make[1]: *** [prepare0] Error 2
make[1]: Leaving directory '/home/KELA/Desktop/projects/optee/linux'
common.mk:159: recipe for target 'linux-common' failed
make: *** [linux-common] Error 2

jforissier, 2017-09-24:
If you're using a 32-bit linux distribution it won't work. Use a x64 distribution or adjust the cross compiler manually.

Souravpunoriyar, 2017-09-24:
Great.. thanks a lot.
I Adjusted the cross compiler.
Its all working well.

closed 2017-09-24 by Souravpunoriyar