OP-TEE/build #207: Build for Qemu

generosocarbone, 2017-10-24:
I'm trying to build for qemu, but I have this problem on the repo command
`repo sync
Fetching project gen_rootfs.git
Fetching projects:   7% (1/13)  Fetching project optee_client.git
Fetching projects:  15% (2/13)  Fetching project optee_benchmark.git
Fetching projects:  23% (3/13)  Fetching project optee_os.git
remote: Counting objects: 7, done.        
remote: Total 7 (delta 5), reused 7 (delta 5), pack-reused 0        
From https://github.com/OP-TEE/optee_os
   b6449075..e4a1f581  master     -> optee/master
Fetching projects:  30% (4/13)  Fetching project soc_term.git
Fetching projects:  38% (5/13)  Fetching project hello_world.git
Fetching projects:  46% (6/13)  Fetching project build.git
remote: Counting objects: 3, done.        
remote: Compressing objects: 100% (1/1), done.        
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0        
From https://github.com/OP-TEE/build
   034c3cd..908c823  master     -> optee/master
Fetching projects:  53% (7/13)  Fetching project optee_test.git
Fetching projects:  61% (8/13)  Fetching project busybox.git
fatal: unable to connect to busybox.net:
busybox.net[0: 140.211.167.122]: errno=Connection timed out

fatal: unable to connect to busybox.net:
busybox.net[0: 140.211.167.122]: errno=Connection timed out

error: Cannot fetch busybox.git

error: Exited sync due to fetch errors`

jbech-linaro, 2017-10-24:
Strange, when I'm running it I get:
```bash
From git://busybox.net/busybox3290), 1.07 MiB | 1.00 MiB/s     
   6f97b30a9..0402cb32d  master     -> busybox/master
```

Having that said, I'm not using the stable, I'm using bleeding edge, but ... checking the [line in the manifest](https://github.com/OP-TEE/manifest/blob/master/default_stable.xml#L12) says that it is checking out `git://busybox.net/busybox` commit `dbf5a6da6a4295ce26edd1ce34fde567d19afa02`

And doing that manually seems to be working:
```bash
$ git clone git://busybox.net/busybox
Cloning into 'busybox'...
remote: Counting objects: 102391, done.
remote: Compressing objects: 100% (25848/25848), done.
remote: Total 102391 (delta 81975), reused 94697 (delta 75919)
Receiving objects: 100% (102391/102391), 19.86 MiB | 603.00 KiB/s, done.
Resolving deltas: 100% (81975/81975), done.
[jbech@manjaro tmp]
$ cd busybox/
[jbech@manjaro busybox]
$ git checkout dbf5a6da6a4295ce26edd1ce34fde567d19afa02
Note: checking out 'dbf5a6da6a4295ce26edd1ce34fde567d19afa02'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at dbf5a6da6... blkdiscard: new applet
[jbech@manjaro busybox]
$ 
```
I.e, seems to be working here. Firewall issue? It uses the git-protocol and not http(s).

generosocarbone, 2017-10-31:
I had to change a couple of networks, none of them with firewall and/or proxy, but it works now.
Thanks

closed 2017-10-31 by generosocarbone