OP-TEE/build #213: Instructions for running OPTEE on a board

mkravchik, 2017-11-22:
Hi,

I'm a newbie so excuse me if it is too basic. I've succeeded running my code on QEMU and now want to run it on a real Xilinx 7000 board. Can anyone provide some details on the steps of this process?

Thanks a lot!

vchong, 2017-11-22:
Seems like Xilinx 7000 is only supported in https://github.com/OP-TEE/optee_os, but not here in https://github.com/OP-TEE/build. You can maybe try to contact the maintainers (Yan Yan and Feng Yu) for more info: https://github.com/OP-TEE/optee_os/blob/master/MAINTAINERS.md.

mkravchik, 2017-11-22:
Right, I asked them and they told me how to build OPTEE. But I miss the understanding of the rest of the system? Like what Linux and bootloaders should I use and how to build them? How do I package all the stuff together and use it on the board?

vchong, 2017-11-23:
Right, Linaro doesn't currently have the 7000 board so you'll have to get those other info from them as well. It looks like you've posted your question on https://github.com/OP-TEE/optee_os/issues/1964 so can this be closed? Thanks!

mkravchik, 2017-11-23:
Yes, I asked it there. i wold close the issue, but I'd really want to understand the general guidelines:
1. Where should I be taking the bootloaders and Linux in general case?
2. How should I package all of this together?

jbech-linaro, 2017-11-23:
For all Linaro maintained platforms this is the place, i.e. our build.git. The "build.git" will pick everything from:
* https://github.com/OP-TEE
* https://github.com/linaro-swg
* And a few other upstream repos (busybox, EDK2, QEMU, Grub etc).

For platforms that we aren't maintaining by ourselves, it is hard for us to say, since in some cases they have trees internally for their devices. So I second what @vchong said, it's the Xilinx guys (Sören) that would know exactly.

alwyas, 2019-09-04:
Hi,
@mkravchik, you mentioned they told you how to build, did you finally package them together and use it on the board? 
How did you create the linux image with optee integrated?

vegetablejedi, 2024-11-04:
Hello, @mkravchik

> Right, I asked them and they told me how to build OPTEE. But I miss the understanding of the rest of the system? Like what Linux and bootloaders should I use and how to build them? How do I package all the stuff together and use it on the board?

Could you please share how to configure OP-TEE on Zynq 7K as suggested by the maintainers?