OP-TEE/build #215: Make flash not available (RPI3)

darakian, 2017-11-27:
Has this been removed? What is the correct way to make an sd card image for a raspberry pi 3?
```
make: *** No rule to make target 'flash'.  Stop.
```

vchong, 2017-11-27:
Was `make flash` ever available for the rpi3?

Try `make img-help` and follow the instructions printed. Per `rpi3.mk`:

```
# Creating images etc, could wipe out a drive on the system, therefore we don't
# want to automate that in script or make target. Instead we just simply provide
# the steps here.
```

darakian, 2017-11-27:
Got it. My bad.

vchong, 2017-11-27:
No problem. Closing issue but feel free to post new one if required.

closed 2017-11-27 by vchong

darakian, 2017-12-06:
Is there an issue with making the sd card image on a different device than where the image was built? My ubuntu system is a vm on a fedora system.

jbech-linaro, 2017-12-06:
> Is there an issue with making the sd card image on a different device than where the image was built? My ubuntu system is a vm on a fedora system.

I'm not 100% sure I follow you here, but I'd say no, there is no issue moving things around. As long as you can mount the card inside the VM you are good to go. Where you have built them doesn't matter, it is still the same cross compiler being used.

vchong, 2017-12-06:
If you're using the instructions from `make img-help` verbatim then yes. You can probably transfer/copy the required files to the other device and substitute the variable (`GEN_ROOTFS_PATH`) with whatever it is on the other device.

darakian, 2017-12-06:
The reason for this is that Virt manager doesn't want to let me mount my sd card reader to the guest. It seems like the path of least resistance to try to get the files out and make the image outside. I'll give it a spin and report back. Thanks.

jbech-linaro, 2017-12-06:
I see, well if you can mount a folder accessible from you Fedora dist containing the Ubuntu-built files, then  you should just be able to put the files on the SD-card. I.e, if you do that from Fedora or the Ubuntu VM doesn't matter. A side note here, doing the "move file to SD-card" for the RPi3/OP-TEE setup becomes tedious pretty quickly. After doing that a handful of times you will probably start to mumble and swear :smile: . So I think it's well worth spending ~1h trying to configure the tftp/nfs setup. Since that saves so much time in the long run and actually makes RPi3 a quite nice platform to use after that.

> Alternately are there build instructions for fedora? Specifically what packages are required?

I think most people in our team are running Ubuntu. Me personally also runs OP-TEE on Arch Linux and that works fine and wasn't too hard to get working. Basically what I did was starting a build and when I noticed errors I searched for the particular package corresponding to the Ubuntu package and tried to build again and so on. Besides the packages needed, please also make sure to put Python2 on the `$PATH` before Python3. Some of the projects we are using doesn't work with Python3 unfortunately. This can be done globally on the dist, but you can also make temporary changes. I didn't want to mess with my system, so I just created a folder where I symlinked python to python2 then I just export/source that line before starting my build. I guess there tons of ways how this could be done and this is one of them.

```bash
$ pwd && ll
/home/jbech/devel/init
total 8
drwxr-xr-x  2 jbech jbech 4096 Aug  1  2016 .
drwxr-xr-x 45 jbech jbech 4096 Dec  5 23:49 ..
lrwxrwxrwx  1 jbech jbech   16 Aug  1  2016 python -> /usr/bin/python2
```

```bash
$ cat ~/bin/init_python2 
export PATH=$HOME/devel/init:$PATH
```

```bash
# Before building OP-TEE
source ~/bin/init_python2
```