OP-TEE/build #224: aarch64-linux-gnu-gcc cannot execute binary file: Exec format error

Fizbanoide, 2018-01-17:
Hi everybody,

I've been trying to get the OP-TEE working on a Raspberry 3 following every instructions from the prerequisites, but i keep getting this error at the step 7.5 Build the solution. I tried different OS, 32 and 64 bits, but i keep getting this error, even if all the step before went perfectly. Here is the full log : 

> make -C /home/pi/devel/optee/build/../optee_os O=out/arm CFG_ARM64_core=y CROSS_COMPILE=" /home/pi/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" CROSS_COMPILE_core=" /home/pi/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" CROSS_COMPILE_ta_arm64=/home/pi/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu- CROSS_COMPILE_ta_arm32=/home/pi/devel/optee/build/../toolchains/aarch32/bin/arm-linux-gnueabihf- CFG_TEE_CORE_LOG_LEVEL=3 DEBUG=0 CFG_TEE_BENCHMARK=n PLATFORM=rpi3
make[1]: Entering directory '/home/pi/devel/optee/optee_os'
  CHK     out/arm/conf.mk
  CHK     out/arm/include/generated/conf.h
/bin/bash: /home/pi/devel/optee/build/../toolchains/aarch64/bin/aarch64-linux-gnu-gcc: cannot execute binary file: Exec format error
mk/compile.mk:228: recipe for target 'out/arm/core/include/generated/.asm-defines.s' failed
make[1]: *** [out/arm/core/include/generated/.asm-defines.s] Error 126
make[1]: Leaving directory '/home/pi/devel/optee/optee_os'
common.mk:306: recipe for target 'optee-os-common' failed
make: *** [optee-os-common] Error 2

Any idea what i might be doing wrong ?

jforissier, 2018-01-17:
`aarch64-linux-gnu-gcc` is a Linux x86_64 binary, so it will run only on a Linux machine running a 64-bit distribution for Intel-compatible CPUs.

Fizbanoide, 2018-01-17:
Hi jforissier,

I don't really get it, this means they say it should work on Raspberry but because the processor is actually an ARM it won't work on it ?

jbech-linaro, 2018-01-17:
@Fizbanoide , what does your **build machine** say when you are running:
```bash
$ uname -m
```
And what does `file` say?
```bash
$ cd <...>/rpi3/toolchains/aarch64/bin
$ file aarch64-linux-gnu-gcc
```

jforissier, 2018-01-17:
It is assumed that everything is built using cross-compilation. With some relatively minor changes to the build files, it should be possible to build natively, but many target platforms have limited resources such as slow CPUs, limited/slow persistent storage, no networking capabilities etc. so native build is often not an option.

igoropaniuk, 2018-01-17:
@jbech-linaro, based on home directory "/home/pi" in provided logs, seems that @Fizbanoide is trying to  build everything on RPi3.

Fizbanoide, 2018-01-17:
Thanks for your answers.

@jbech-linaro Here are the results : 

`uname -m
aarch64`

`aarch64-linux-gnu-gcc: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=dfec1a1c34b791582e69a236ebf2f7b95df3fea0, stripped`

@igoropaniuk  Exactly, i'm running all this on a RPi3

jbech-linaro, 2018-01-17:
> aarch64-linux-gnu-gcc: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=dfec1a1c34b791582e69a236ebf2f7b95df3fea0, stripped

So, this proves what @jforissier said, i.e., you are trying to run a x86 binary on Arm, that doesn't work.

> @igoropaniuk Exactly, i'm running all this on a RPi3

I see, with all dependencies and an Arm toolchain instead of x86 toolchain it should be possible to build, eventually one would need to tweak our makefiles in build.git. But I think this is asking for lots of problem, so I would probably just build using a cross compiler on a regular PC.

Fizbanoide, 2018-01-18:
@jbech-linaro Thanks, i'll try this then, let you know if I could make it work ! Last question, does this mean that i need to change the OS on my RPi3 ?

jbech-linaro, 2018-01-18:
@Fizbanoide , what I'm doing when developing TEE stuff for my RPi3 is simply to use an ordinary (x86) computer running Linux (I've been using Ubuntu and Arch Linux). I compile everything there, then I simply just deploy all the files to RPi3.

There are two ways to do this. Either you copy all built files to the SD-card (which is rather cumbersome) and that is described when you run `$ make img-help` from build.git. A better alternative is to take the time to setup tftp and NFS and then just let U-Boot pick up the newly built files, that is described [here](https://github.com/OP-TEE/build/blob/master/docs/rpi3.md#5-nfs-boot).

With the NFS setup I've been using both the small busybox based root fs that are being built when build RPi3 for OP-TEE as described in our [docs](https://github.com/OP-TEE/build/blob/master/docs/rpi3.md) and I've also used a debian based root fs. I know some people like @seqlabs have successfully been using OP-TEE with default Raspbian for RPi3.

Fizbanoide, 2018-01-18:
@jbech-linaro Perfect, thanks for all your advices ! I'll get into it and try to make it work !

Fizbanoide, 2018-01-23:
It's getting better so far, i'm trying to set up the U-Boot's envrionment variables to make it work, still having TFTP error: 'FIle not found', even when following the READme file. I'll let you know if i find a solution !

closed 2018-01-26 by Fizbanoide

berserkr, 2018-04-27:
The instructions are not clear. I am also trying to build on the pi itself. As I understand, the pi is 32bit (v7, v8s will be 64 bit). When I run make on the pi, the same thing happens where it fails. Based on the comments above, does this mean that we MUST build on a separate machine, then flash the pi?

jbech-linaro, 2018-04-27:
@berserkr, two things. Pi itself is an Armv8 device, but stock software from Pi organisation is 32-bit builds. Then for the setup we are running, we are using a fork with runs 64-bit kernel, but this isn't an official kernel by the Pi organisation.

I would just say, just forget about compiling OP-TEE stuff on the Pi itself, just do it using cross compilers on a regular PC instead (x86 for example). This is no must, but it will probably save you some head ache like I mentioned in another comment higher up.

berserkr, 2018-04-27:
@jbech-linaro I am doing now that man, thank you. I will need to set up NFS Boot, hopefully I can get it up and running.

jbech-linaro, 2018-05-02:
Hi,

> ... 
Still, the instructions ask us to navigate the the rpi3/boot dir and do the following:

Paths have changed, but docs (which should be updated) are still referring to the old way of doing it. Having that said, the [rpi3.mk here|https://github.com/OP-TEE/build/blob/master/rpi3.mk#L27-L31] tells where the files are located now.

> After I ran make, which completed fine, I started the nfs steps. Should I have done anything else? I did not do "make flash" as that seems to be needed only for native compilation?

Right, for an tftp/nfs setup you don't have to run `make flash`. Having that said, you need to update root-fs in one or another way. How that should be done isn't really written in this guide. For me personally I've been using a script created on the fly by some [additional make file rules](https://github.com/jbech-linaro/build/blob/rpi_nfs/rpi3.mk#L274-L287). This is a bit hacky and as said a personal thing, so I don't want to submit this as a patch. But for me personally this gets the job done. With that I only have to rebuild and reboot and then all files have been updated.

DeepakArya83, 2018-08-02:
Hello Fizbanoide,
I am facing the same issue while executing make command as defined in step 7.5. How this problem is resolved at ur end. Can u please share the steps to solve this issue. 

Thanks in advance.