OP-TEE/build #225: Trying to debug TA application with Raspberry 3B

Fizbanoide, 2018-01-26:
Hi everybody, I have a Raspberry 3b working well with OP-TEE via U-Boot and NFS,  and xtest is working fine on it. I tried also the Hello World example (https://github.com/linaro-swg/optee_examples/tree/master/hello_world),  which also works fine. The only thing is I'm trying to get the DMSG on a  terminal, on my computer for example. Do I need a UART cable ? Or maybe  can I do it via the NFS or TFTP server ? I don't really understand the 6 section (https://github.com/OP-TEE/build/blob/master/docs/rpi3.md#5-nfs-boot)

igoropaniuk, 2018-01-26:
Hi @Fizbanoide,

right, you need to use UART (you can buy the chipest USB-TTL adapter for a couple bucks, it should work) to be able to receive SW output (from OPTEE OS core and TAs)

jbech-linaro, 2018-01-26:
Hi @Fizbanoide ,

Glad to hear that you got that working. Regarding USB, what I have is a normal [USB to UART cable](https://www.google.se/search?q=usb+tty+cable). With that connected to the pins according to section 6.1, you should be able to get the debug prints by just open `/dev/ttyUSB0` (or so) on PC connected to the Pi.

Fizbanoide, 2018-01-26:
Thanks @igoropaniuk and @jbech-linaro , I wasn't sure about it ! Any chance otherwise I can write in a log file that I can access from the Rpi3 ?

igoropaniuk, 2018-01-26:
@Fizbanoide ,

I'm not sure if it's a good advice from my side, but if an setup with UART doesn't currently work for you, you can try to follow this discussion https://github.com/OP-TEE/optee_os/issues/954 and as temporary solution apply/rework patches from RFC pull requests https://github.com/OP-TEE/optee_os/pull/810 https://github.com/OP-TEE/optee_client/pull/51 by @jforissier (unfortunately they weren't merged). 

These patches introduce a possibility to forward debug output to linux syslog (I guess that's what you asked for)

Fizbanoide, 2018-01-26:
@igoropaniuk I'll check those, thanks !

Fizbanoide, 2018-01-29:
Hi guys,

Just received the UART cable, and it's working perfectly with Busybox. Thanks again for your help @igoropaniuk @jbech-linaro !

closed 2018-01-29 by Fizbanoide