OP-TEE/build #227: Is there any UUID for the cpu ?

Fizbanoide, 2018-02-09:
Hello guys,

I've been asking myself and couldn't find so far any answer. Is there any UUID written on the CPU for example to diferentiate two RPi ? 
I've already tried "cat /proc/cpuinfo", but there is no Serial number on this file.

jforissier, 2018-02-12:
Hi @Fizbanoide ,

Where to find a unique ID is quite hardware-specific, there is no general answer. Many SoCs have one-time programmable memory areas, which can be used to store such IDs. IDs may also be stored in flash memory (for instance, UEFI variables).

Fizbanoide, 2018-02-12:
Hi @jforissier ,

I'll check those, thanks. I was also wondering what is the application UUID important for ? To link the host and the ta application ? Or can I use it maybe to identify an application in another application ?

jforissier, 2018-02-12:
>  I was also wondering what is the application UUID important for ? To link the host and the ta application ?

Yes.

> Or can I use it maybe to identify an application in another application ?

Yes, why not.

Fizbanoide, 2018-02-13:
Perfect, thanks !

closed 2018-02-13 by Fizbanoide