OP-TEE/build #230: edk2-clean-common target fails in qemu_v8 build

goatshriek, 2018-02-18:
I'm having problems running the clean target in a build using the qemu_v8 manifest. Make stops after trying to run the edk2-clean-common target with the message "No rule to make target 'cleanall'. Stop." Digging into it appears that the referenced Makefile in the edk2 repo (ArmPlatformPkg/Scripts/Makefile) only has a clean target but the edk2 clean target in common.mk uses cleanall. Some other makefiles (such as fvp and all three for hikey) are using the build script in edk2 directly in their edk2-call function, so changing common.mk to use 'clean' instead would change their behavior.

I've changed qemu_v8.mk file with a special edk2-clean rule instead of edk2-clean-common that uses clean instead of cleanall and created a pull request if that's an acceptable solution. Not very elegant, but it got the job done for me.

goatshriek, 2018-02-22:
Waiting for pull request #231 to fix this issue.

jforissier, 2018-02-22:
@goatshriek does #231 work for you?

goatshriek, 2018-02-24:
Yes, it works just fine! I've switched to using that locally with no problems so far. I stumbled across another issue while I was testing it; running clean twice fails due to the strace-clean target removing the Makefile on it's way out. I will submit a separate patch to rectify that issue though - it's just a matter of adding a test for the file the way hikey makefiles already do.

jforissier, 2018-02-24:
OK, good.

closed 2018-03-02 by goatshriek