OP-TEE/build #232: No message from TA with Raspberry PI

rosinskm, 2018-02-19:
Hi guys,
I am trying to debug my own TA, but I doesn't see the DMGS, EMSG, IMSG and FMSG from TA. I have a Raspberry PI 3 working well with OP-TEE. Xtest and all the example run well, but I never see the message from TA. I try to use a UART to TTL to see the information from the TA(like this example https://github.com/OP-TEE/build/issues/225 ), it doesn't work. Could you help me to resolve it ?

Fizbanoide, 2018-02-20:
Hi @rosinskm ,

Do you have any error message ? Or have any trouble with busybox ?

rosinskm, 2018-02-20:
I have no error message, and I don't really understand why I need to use busybox to receive messages from my TA.

Fizbanoide, 2018-02-21:
Sorry my mistake, i meant with microcom, you don't receive anything on it ? Are the host part of the examples working ?

rosinskm, 2018-02-21:
On my screen and on microcom(I use putty), I receive the same thing. In both terminal, the host part working well an I receive all the "print" from the examples. For example, I would like to have on the raspberry screen the host print and with microcom the TA message/log/print(for the debug).

jbech-linaro, 2018-02-22:
This might not be an answer, but just recently I had an issue when remotely connecting (SSH) to a machine where I compiled and run xtest/OP-TEE (using QEMU). I got blank terminals back from the server. It turned out that if I disabled the default terminal (gnome-terminal) and instead used xterm, things where working as it should. I do not think this is your issue, but it could be worth trying it from xterm for example.

You have noticed that we just recently changed the output from all the debug messages? I.e., it is a shorter and more compressed format now, like this:
```bash
D/TC:0 core_mmu_alloc_l2:261 L2 table used: 1/4
D/TC:0 core_mmu_alloc_l2:261 L2 table used: 2/4
I/TC:  
D/TC:0 ...
```
That is probably not the problem either, but I thought it could be worth mention that, so it is not a simple thing such as that, that tricks you.

Do you have any log file to share, that captures everything from the boot to it is up and running xtest?

rosinskm, 2018-02-22:
It is not the problem.
Xtest run well as you can see.

[image]

The TA part never sends me Info Log. 
For example, when I execute the opte_example_hello_world, I got this message : 

[image]

And I would like to have message from TA, like the IMSG : 

[image]

I already put CFG_TEE_TA_LOG_LEVEL =4 to see all the message from TA, still have nothing.

During the boot of my raspberry, I can see the following line : 

[image]

rosinskm, 2018-02-27:
I have finally found my problem. Some functions were missing in /optee_os/core/arch/plat_rpi3/main. c.

[image]

Only consoleInit() was here in the begin, you have to pick up the other function from another platform to have the trace from your TA (I have used plat_stm).

rosinskm, 2018-02-27:
I have finally found my problem. Some functions were missing in /optee_os/core/arch/plat_rpi3/main. c.

[image]

Only consoleInit() was here at the begin, you have to pick up the other function from another platform to have the trace from your TA (I have used plat_stm).

closed 2018-02-27 by rosinskm

jforissier, 2018-02-27:
Hi @rosinskm ,

Which version of OP-TEE are you using? The latest code uses the generic console routines in https://github.com/OP-TEE/optee_os/blob/master/core/kernel/console.c with the 8250 driver in https://github.com/OP-TEE/optee_os/blob/master/core/drivers/serial8250_uart.c.
See https://github.com/OP-TEE/optee_os/blob/master/core/arch/arm/plat-rpi3/main.c#L68-L73.

rosinskm, 2018-02-27:
Yes, I see it but it was not working on my PI. So I must add the function in /optee_os/core/arch/plat_rpi3/main.c.

jforissier, 2018-02-27:
> So I must add the function in /optee_os/core/arch/plat_rpi3/main.c.

Your call. I would have said "I must understand why and propose a fix".

jbech-linaro, 2018-02-27:
Hi, so today I've setup my RPi3 (using tftp and nfs) from scratch. I see all prints, both from normal and secure world, so I have no clue why this isn't working for you.

```bash
...
## Transferring control to ARM TF (at address 8400000) (dtb at 3ab2d000)...
NOTICE:  BL3-1: v1.1(debug):1da4e155
NOTICE:  BL3-1: Built : 10:00:46, Feb 27 2018
...
I/TC:  OP-TEE version: 3.0.0-33-ge2998dec #1 Tue Feb 27 09:00:32 UTC 2018 aarch64
D/TC:0 tee_ta_register_ta_store:609 Registering TA store: 'REE' (priority 10)
D/TC:0 tee_ta_register_ta_store:609 Registering TA store: 'Secure Storage TA' (priority 9)
D/TC:0 mobj_mapped_shm_init:559 Shared memory address range: 8800000, a800000
I/TC:  Initialized
D/TC:0 init_primary_helper:831 Primary CPU switching to normal world boot
INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address = 0x80000
INFO:    BL3-1: Next image spsr = 0x3c9
D/TC:  generic_boot_cpu_on_handler:869 cpu 1: a0 0x0
D/TC:  init_secondary_helper:854 Secondary CPU Switching to normal world boot
D/TC:  generic_boot_cpu_on_handler:869 cpu 2: a0 0x0
D/TC:  init_secondary_helper:854 Secondary CPU Switching to normal world boot
D/TC:  generic_boot_cpu_on_handler:869 cpu 3: a0 0x0
D/TC:  init_secondary_helper:854 Secondary CPU Switching to normal world boot
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.6.3-01139-g7d976fc52ae3 (jbech@debby) (gcc version 6.2.1 20161016 (Linaro GCC 6.2-2016.11) ) #1 SMP Tue Feb 27 10:09:21 CET 2018
...
root@linaro-developer:~# xtest
...
<here are tons of both NW and SW debug prints>
...
+-----------------------------------------------------
15787 subtests of which 0 failed
66 test cases of which 0 failed
0 test case was skipped
TEE test application done!
root@linaro-developer:~# 
```

rosinskm, 2018-02-28:
Hi, Think you for helping me. In my company, we are two with the same problem. I will try to fix it later. For the moment, I will develop my TA with the tips I have explained before. 
If I success to fix the bug, I'll upload it.