OP-TEE/build #236: Broken system_img link hikey_debian.mk

acarb95, 2018-03-07:
The system_img url no longer works for hikey_debian. I get this error:
```
--2018-03-06 17:12:02--  http://builds.96boards.org/releases/reference-platform/debian/hikey/16.06/hikey-rootfs-debian-jessie-alip-20160629-120.emmc.img.gz
Resolving builds.96boards.org (builds.96boards.org)... 54.196.141.104
Connecting to builds.96boards.org (builds.96boards.org)|54.196.141.104|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://releases.linaro.org/96boards//reference-platform/debian/hikey/16.06/hikey-rootfs-debian-jessie-alip-20160629-120.emmc.img.gz [following]
--2018-03-06 17:12:02--  https://releases.linaro.org/96boards//reference-platform/debian/hikey/16.06/hikey-rootfs-debian-jessie-alip-20160629-120.emmc.img.gz
Resolving releases.linaro.org (releases.linaro.org)... 52.77.230.32
Connecting to releases.linaro.org (releases.linaro.org)|52.77.230.32|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-03-06 17:12:04 ERROR 404: Not Found.
```

https://github.com/OP-TEE/build/blob/b60156e816a459173a82dd04aaf5d0757863bbc6/hikey_debian.mk#L24

What should this point to now? The latest on the releases.linaro.org (builds.96boards.org redirects there), is 15.11.

vchong, 2018-03-07:
Hi Amanda, sorry about that. builds.96boards.org has transitioned to releases.linaro.org, but I couldn't find the original 16.06 rootfs either, maybe due to an existing `reference-platform` dir in r.l.o, the one from b.96b.o didn't/couldn't be moved over. In any case, it's been a bit hard to keep the debian build updated due to lack of utilization. Since you're probably the most active user we know, at least publicly, and to a certain degree invested in this as well, do you mind helping us try the one from http://releases.linaro.org/debian/images/developer-arm64/latest? Use `wget http://releases.linaro.org/debian/images/developer-arm64/latest/linaro-stretch-developer-*.tar.gz` so you can put it in a script and not have to worry about changing dates-buildnum in the filename. If it doesn't work, maybe try falling back to the 15.11 one as well. Thanks!

P/S: Btw, remember reading your pretty detailed analysis of the kernel hdmi issue some time ago. Any progress on that front?

vchong, 2018-03-14:
@acarb95 Any luck with above?

This is just to inform you that support for hikey_debian will be dropped [1] soon. You can still checkout the 3.0.0 release for use but note that it'll no longer be maintained.

[1] https://github.com/OP-TEE/manifest/pull/96

acarb95, 2018-03-15:
@vchong Thanks for the heads up! I haven't tried those images yet, I'll let you know if they work once I try them.

acarb95, 2018-03-15:
I have made progress on the HDMI issue as well, I've gotten the monitor to display, but it logs out every time xterm is opened and cannot find any networks to connect to. So I still have more debugging to do.