OP-TEE/build #241: Problem syncing repo qemu_v8_stable

cezane, 2018-03-15:
I am trying to configure the environment to work with qemu_v8_stable version, but without success. Is there some problem with the repository?

```
repo init -u https://github.com/OP-TEE/manifest.git -m qemu_v8_stable.xml
Get https://gerrit.googlesource.com/git-repo/clone.bundle
Get https://gerrit.googlesource.com/git-repo
remote: Finding sources: 100% (11/11)
remote: Total 11 (delta 0), reused 11 (delta 0)
Unpacking objects: 100% (11/11), done.
From https://gerrit.googlesource.com/git-repo
   685320b..1f36570  master     -> origin/master
Get https://github.com/OP-TEE/manifest.git
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
```

Also, as some problem occurs with repo init, same occurs when running repo sync:

```
repo sync
Fetching project code
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Server does not provide clone.bundle; ignoring.
remote: Counting objects: 93   
```

Why am I getting this "404 not found"?

jbech-linaro, 2018-03-15:
@cezane , that's a weird message being seen because there is no support for "clone bundles" on the server (GitHub) just as @jforissier said. You can safely just ignore it, things are still working as it should. But, if it hurts your eyes, then add `--no-clone-bundle` to the `repo init ...` line and you will not see it anymore.

(apparently @jforissier is a `repo` specialist afterall! :+1:)

cezane, 2018-03-15:
As I never used repo, I just opened this issue because I tried to syncing the repository before and I got some errors. Unfortunately, I did not save the error(s) message(s). So, I thought this problem could be because of this "clone.bundle/404 not found"... As you said it is not, I am trying syncing again and I will say if the earlier problem occurs again when the process finishes.

closed 2018-05-17 by cezane