OP-TEE/build #257: Openocd rpi3 jlink debugger run error

saurabhsamanta, 2018-05-02:
Trying to use openocd with jlink debugger with rpi3 (https://www.op-tee.org/docs/rpi3/). While running I am getting following error

embedded:startup.tcl:21: Error: target requires -dap parameter instead of -chain-position!
in procedure 'script' 
at file "embedded:startup.tcl", line 60

jbech-linaro, 2018-05-02:
I think I'm the only one in the Linaro / OP-TEE team who has been using the JTAG on RPi3. Having that said, it's been more than a year ago since I last tried it, so I don't dare to say much about the status. I basically would need to try it out first. But, a wild guess is that things have changed in newer OpenOCD releases, it might be as simple as an update [here](https://github.com/OP-TEE/build/blob/master/rpi3/debugger/pi3.cfg#L23) that is needed. I.e., change the line to maybe use `-dap` instead of `-chain-position`.

saurabhsamanta, 2018-05-03:
I tried changing the parameter to -dap. Which gives following error
embedded:startup.tcl:21: Error: DAP name invalid!

jbech-linaro, 2018-05-03:
Ok, well, I have no time to try this right now. I might give it a try when we are getting closer to the next OP-TEE release (3.2.0) where we need to test our RPi3 setup. But the target date for that is in beginning of July, until then I'm afraid you are on your own if no-one else chime in here.

jeevan1995, 2018-05-07:
Hello,
I am trying to compile a simple C++ program printing a small output
Is C++ program is supported in the current build toolchain?
what should be added in the build chain in order to enable for C++ compilation?

jbech-linaro, 2018-05-07:
Hi @jeevan1995 , are you talking about C++ running in a TA (not supported) or something running in Linux as a user space application (supported, just use g++ for Arm).

jbech-linaro, 2019-02-04:
@saurabhsamanta , it took a couple of months, but there is a PR with a fix for this now, see #342 .