OP-TEE/build #271: rpi3_stable.xml not available

jdeus, 2018-06-27:
rpi3_stable.xml is stated as available, but it's not:

`repo init -u https://github.com/OP-TEE/manifest.git -m rpi3_stable.xml`

`fatal: manifest 'rpi3_stable.xml' not available
fatal: manifest rpi3_stable.xml not found`

jbech-linaro, 2018-06-27:
I'll take the blame for this one. We removed the stable manifests starting from the 3.1.0 release. It is still possible to get hold of old stable releases, but that require one to checkout a certain commit manually. I have still on my to-do to update our README's with information how to get stable versions as it is now as well as getting older stable releases. I'll mark this as a bug a and assign it to myself.

In the mean time here is the stable branch corresponding to 3.1.0 to use with manifest (repo init ...):
https://github.com/OP-TEE/manifest/tree/3.1.0

jbech-linaro, 2018-08-27:
I will send a patch for this later today.

PR #297 (2018-08-27): Cleanup regarding manifests and deprecated builds.
- Clarify how to get older releases.
- Remove references to deprecated builds like HiKey Debian (not the old stable manifest) and MTK8173.
- We're not using Travis to build matrix any more, so the icon telling the status has been removed.

This should fix #271,  #278 and partly #296.
commit d4a38f19 (2018-08-27): docs: explain how to get various stable releases

Fixes #271

Acked-by: Jerome Forissier <jerome.forissier@linaro.org>
closed 2018-08-27 by jforissier