OP-TEE/build #276: Not booting

jdeus, 2018-07-03:
All right, I'm following the following tutorial https://blog.crysys.hu/2018/06/op-tee-default-build-and-installation-on-the-raspberry-pi/ and I'm stuck when booting the board: it's stuck on the rainbow screen and I got nothing on the UART (correctly wired as per your instructions). I'm using a working 3.3V dongle.

Here are the steps I followed (make img-help), as root: 
`
  fdisk /dev/sdx   # where sdx is the name of your sd-card
   > p             # prints partition table
   > d             # repeat until all partitions are deleted
   > n             # create a new partition
   > p             # create primary
   > 1             # make it the first partition
   > <enter>       # use the default sector
   > +32M          # create a boot partition with 32MB of space
   > n             # create rootfs partition
   > p
   > 2
   > <enter>
   > <enter>       # fill the remaining disk, adjust size to fit your needs
   > t             # change partition type
   > 1             # select first partition
   > e             # use type 'e' (FAT16)
   > a             # make partition bootable
   > 1             # select first partition
   > p             # double check everything looks right
   > w             # write partition table to disk.

	mkfs.vfat -F16 -n BOOT /dev/sdb1
	mkdir -p /media/boot
	mount /dev/sdb1 /media/boot
	cd /media
	gunzip -cd /root/devel/optee/build/../out-br/images/rootfs.cpio.gz | cpio -idmv "boot/*"
	umount boot

	mkfs.ext4 -L rootfs /dev/sdb2
	mkdir -p /media/rootfs
	mount /dev/sdb2 /media/rootfs
	cd rootfs
	gunzip -cd /root/devel/optee/build/../out-br/images/rootfs.cpio.gz | cpio -idmv
	rm -rf /media/rootfs/boot/*
	cd .. && umount rootfs
`

Things I checked:

`
fdisk /dev/sdb
p

Disk /dev/sdb: 14.7 GiB, 15816720384 bytes, 30892032 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xbbd9b29d

Device     Boot Start      End  Sectors  Size Id Type
/dev/sdb1  *     2048    67583    65536   32M  e W95 FAT16 (LBA)
/dev/sdb2       67584 30892031 30824448 14.7G 83 Linux

mount /dev/sdb1 /media/boot
ls -lh /media/boot
total 23M
-rwxr-xr-x 1 root root  14K Jul  1 16:29 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root  50K Jul  1 15:30 bootcode.bin
-rwxr-xr-x 1 root root  181 Jul  1 15:30 config.txt
-rwxr-xr-x 1 root root  19K Jul  1 15:30 COPYING.linux
-rwxr-xr-x 1 root root 2.5K Jul  1 15:30 fixup_cd.dat
-rwxr-xr-x 1 root root 6.5K Jul  1 15:30 fixup.dat
-rwxr-xr-x 1 root root 9.6K Jul  1 15:30 fixup_db.dat
-rwxr-xr-x 1 root root 9.6K Jul  1 15:30 fixup_x.dat
-rwxr-xr-x 1 root root 9.8M Jul  1 17:08 Image
-rwxr-xr-x 1 root root 1.5K Jul  1 15:30 LICENCE.broadcom
-rwxr-xr-x 1 root root 410K Jul  3 10:46 optee.bin
-rwxr-xr-x 1 root root 627K Jul  1 15:30 start_cd.elf
-rwxr-xr-x 1 root root 4.8M Jul  1 15:30 start_db.elf
-rwxr-xr-x 1 root root 2.8M Jul  1 15:30 start.elf
-rwxr-xr-x 1 root root 3.8M Jul  1 15:30 start_x.elf
-rwxr-xr-x 1 root root  16K Jul  3 10:47 uboot.env
-rwxr-xr-x 1 root root 367K Jul  3 10:47 u-boot-rpi.bin

mount /dev/sdb2 /media/rootfs
ls -lh /media/rootfs
total 84K
drwxr-xr-x 2 root root 4.0K Jul  3 10:48 bin
drwxr-xr-x 2 root root 4.0K Jul  3 10:48 boot
drwxr-xr-x 4 root root 4.0K Jul  3 10:48 dev
drwxr-xr-x 6 root root 4.0K Jul  3 10:48 etc
-rwxr-xr-x 1 root root  178 Jul  3 10:48 init
drwxr-xr-x 5 root root 4.0K Jul  3 10:48 lib
lrwxrwxrwx 1 root root    3 Jul  3 11:11 lib64 -> lib
lrwxrwxrwx 1 root root   11 Jul  3 11:11 linuxrc -> bin/busybox
drwx------ 2 root root  16K Jul  3 11:11 lost+found
drwxr-xr-x 2 root root 4.0K Jul  1 15:30 media
drwxr-xr-x 2 root root 4.0K Jul  1 15:30 mnt
drwxr-xr-x 2 root root 4.0K Jul  1 15:30 opt
drwxr-xr-x 2 root root 4.0K Jul  1 15:30 proc
drwx------ 2 root root 4.0K Jul  1 15:30 root
drwxr-xr-x 2 root root 4.0K Jul  1 15:30 run
drwxr-xr-x 2 root root 4.0K Jul  1 17:16 sbin
drwxr-xr-x 2 root root 4.0K Jul  1 15:30 sys
drwxrwxrwt 2 root root 4.0K Jul  1 15:30 tmp
drwxr-xr-x 6 root root 4.0K Jul  3 10:48 usr
drwxr-xr-x 4 root root 4.0K Jul  3 10:48 var

ls -lh /media/rootfs/boot/
total 0
`

make stdout https://pastebin.com/ru7Cq7kK
make stderr https://pastebin.com/dJSJu2rq

Building on Debian

Any help greatly appreciated!

jbech-linaro, 2018-07-03:
Hi @jdeus ,

I cannot vouch for the instructions they provide on the URL you provided (even though it looks more or less like as a copy/paste of our official instructions), but if you follow our official instructions ([here](https://github.com/OP-TEE/build#6-manifests) and [here](https://github.com/OP-TEE/build/blob/master/docs/rpi3.md)), then I guarantee that it will work, since two of us have [independently tested this OK for the 3.2.0-rc1 tag](https://github.com/OP-TEE/optee_os/pull/2404#issuecomment-399892169) released 1.5 week ago.

// Regards Joakim

jdeus, 2018-07-03:
Hello,

Turns out the instructions are exactly the same as provided on your repo. Still can't get it to work unfortunately, I've tried reformatting the SD card multiple times and followed the instructions given on your repo, still no luck

jbech-linaro, 2018-07-03:
@jdeus , if I find some time I will try it once more tomorrow. You can never say never, but it feel a bit unlikely that it has stopped working since just a week ago. We haven't merged that many patches since last week. I'll make a reply here if I find time to try it out.

jdeus, 2018-07-03:
That would be great. 

I looked a bit around and found that _uboot.env_ in the boot partition seems messed up:
https://ufile.io/z14iw

Here's the entire _rootfs.cpio.gz_ archive: https://ufile.io/04k70

What's going on?

jbech-linaro, 2018-07-04:
@jdeus 

This morning, I've just rebuilt it all from scratch and re-populated the SD-card from scratch and everything is fine and works as expected. So, I'm not sure how to help when things are working here on my side. What you can do is to make sure all dependencies are met (apt-get), start from a fully clean folder. Pay attention if there are error messages while building (due to missing packages etc).

I'm also building using a Debian machine:
```bash
$ uname -a
Linux debby 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
```

Alias used in "Build" and "Revisions" below.
``` bash
$ alias | grep rpo_clean_all
alias rpo_clean_all='repo forall -c '\''echo Cleaning ... $REPO_PATH && git clean -xdf && git checkout -f'\'''alias rpo_clean_all='repo forall -c '\''echo Cleaning ... $REPO_PATH && git clean -xdf && git checkout -f'\'''
alias rpo_rev='repo forall -c '\''echo $REPO_PATH -- `git log --oneline -1`'\'''
```

Build ...
```bash
  508  rpo_clean_all 
  509  repo sync -j3 -d
  510  time make -j8
  511  make img-help 
  512  history 
```

Revisions ...
```bash
$ rpo_rev 
arm-trusted-firmware -- 1da4e155 Resolve GCC6.x build errors
build -- 3b5d8f0 grub: ignore harmless warning
buildroot -- 62dca337fe package/samba4: bump version to 4.8.1
firmware -- 3fb63c413c firmware: FXL6408/GPIOman: Support config of termination via dt-blob firmware: GPIO expander: Add API to reconfigure pins firmware: GPIOMAN: Add API to reconfigure pins firmware: Mailbox service: Add command to reconfigure GPIO setup firmware: FXL6408: Return success code on reading status of an output firmware: GPIO expander: Add set/get_config functions to dummy driver
linux -- 7d976fc52ae3 kernel.h: add u64_to_user_ptr()
optee_benchmark -- 7437cb2 Initial CMake support
optee_client -- 2d542f2 Do not export CMakeLists.txt when running install
optee_examples -- 1c5d96f ta/Makefiles: Set LDFLAGS to null before building ta make scripts
optee_os -- af8149de core: make stack trace robust
optee_test -- c0e8678 xest: OpenSSL may be provided by the system
u-boot -- bc6d93cf5e rpi3: remove jtag enable code
```

Boot and xtest log, full log is [here](https://a.uguu.se/gyltcC2K5x4V_boot_and_xtest.log) (note, only available for 24-hours).
```bash
U-Boot 2016.03-00725-gbc6d93cf5e (Jul 04 2018 - 07:38:59 +0200)

DRAM:  944 MiB
RPI 3 Model B (0xa02082)
boot regs: 0x00000000 0x00000000 0x00000000 0x00000000
MMC:   bcm2835_sdhci: 0
reading uboot.env
In:    serial
Out:   lcd
Err:   lcd
Net:   Net Initialization Skipped
No ethernet found.
starting USB...
USB0:   Core Release: 2.80a
scanning bus 0 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
       scanning usb for ethernet devices... 1 Ethernet Device(s) found
Hit any key to stop autoboot:  0 
reading Image
10203648 bytes read in 886 ms (11 MiB/s)
reading bcm2710-rpi-3-b.dtb
13376 bytes read in 19 ms (687.5 KiB/s)
reading optee.bin
419024 bytes read in 53 ms (7.5 MiB/s)
## Flattened Device Tree blob at 01700000
   Booting using the fdt blob at 0x1700000
   reserving fdt memory region: addr=0 size=1000
   reserving fdt memory region: addr=8000000 size=2000000
   Loading Device Tree to 000000003ab2f000, end 000000003ab3543f ... OK

Starting kernel ...

## Transferring control to ARM TF (at address 8400000) (dtb at 3ab2f000)...
NOTICE:  BL3-1: v1.1(debug):1da4e155
NOTICE:  BL3-1: Built : 07:39:53, Jul  4 2018
INFO:    BL3-1: Initializing runtime services
INFO:    BL3-1: Initializing BL3-2
....
I/TC:  OP-TEE version: 3.2.0-rc1-10-gaf8149de #1 Wed Jul  4 05:39:20 UTC 2018 aarch64
....
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.6.3-01139-g7d976fc52ae3 (jbech@debby) (gcc version 6.2.1 20161016 (Linaro GCC 6.2-2016.11) ) #1 SMP Wed Jul 4 07:42:31 CEST 2018
....
Starting tee-supplicant...
Starting network: OK

Welcome to Buildroot, type root to login
buildroot login: root
# xtest
...
-----------------------
Result of testsuite regression:
regression_1001 OK
regression_1002 OK
regression_1003 OK
regression_1004 OK
regression_1005 OK
regression_1006 OK
regression_1007 OK
regression_1008 OK
regression_1009 OK
regression_1010 OK
regression_1011 OK
regression_1012 OK
regression_1013 OK
regression_1015 OK
regression_1016 OK
regression_1017 OK
regression_1018 OK
regression_1019 OK
regression_4001 OK
regression_4002 OK
regression_4003 OK
regression_4004 OK
regression_4005 OK
regression_4006 OK
regression_4007 OK
regression_4008 OK
regression_4009 OK
regression_4010 OK
regression_4011 OK
regression_4012 OK
regression_5006 OK
regression_6001 OK
regression_6002 OK
regression_6003 OK
regression_6004 OK
regression_6005 OK
regression_6006 OK
regression_6007 OK
regression_6008 OK
regression_6009 OK
regression_6010 OK
regression_6012 OK
regression_6013 OK
regression_6014 OK
regression_6015 OK
regression_6016 OK
regression_6017 OK
regression_6018 OK
regression_6019 OK
regression_6020 OK
regression_7001 OK
regression_7002 OK
regression_7003 OK
regression_7004 OK
regression_7005 OK
regression_7006 OK
regression_7007 OK
regression_7008 OK
regression_7009 OK
regression_7010 OK
regression_7013 OK
regression_7016 OK
regression_7017 OK
regression_7018 OK
regression_7019 OK
regression_8001 OK
regression_8002 OK
regression_8101 OK
regression_8102 OK
regression_8103 OK
regression_2001 OK
regression_2002 OK
regression_2003 OK
regression_2004 OK
+-----------------------------------------------------
16081 subtests of which 0 failed
74 test cases of which 0 failed
0 test case was skipped
TEE test application done!
# 
```

jbech-linaro, 2018-07-04:
FYI, since I was at it, I even took the time to update my "[rpi_nfs](https://github.com/jbech-linaro/build/commits/rpi_nfs)" branch which makes life easier for me when running RPi3 using NFS (this is no official stuff, it's just for my own sake). But, that also turned out to work fine. So both the official with SD card as well as when using NFS seem to be working fine here.

jdeus, 2018-07-05:
I've tried again from a clean machine and still the same result. Tried multiple SD cards, re-did each step multiple times, and same result. Tested the uart and the pi using raspbian and they both are OK.

jbech-linaro, 2018-07-05:
Sorry, but I have no clue why it isn't working. If you want to my binaries a try I've uploaded them [here](http://people.linaro.org/~joakim.bech/jdeus/). Replace your `rootfs.cpio.gz` with the one from my URL and you should be good to go. I've separately added kernel, U-Boot env, the dtb, TEE core also, but those are already in the `rootfs.cpio.gz`, so you shouldn't really need them.

jbech-linaro, 2018-07-05:
Stupid question, you are using RPi3 right and not one of the older versions? Also, we haven't tested this using the newer RPi3 B+, so I cannot tell if that could be a problem or not (and I don't have the RPi3 B+ myself, so I cannot test it).

jdeus, 2018-07-05:
Thanks for the link, I'll try that. I'm currently using the latest RPi3 to date, the 3 B+.

jbech-linaro, 2018-07-05:
> I'm currently using the latest RPi3 to date, the 3 B+.

I wonder if that could be the issue? I'm not sure if there is a new `dtb` file for the new RPI3 B+ or not. That could be worth checking.

You don't happen to have an old RPi3 laying around that you could give a try?

jdeus, 2018-07-05:
There is a new dtb called bcm2710-rpi-3-b-plus.dtb ([here](https://github.com/raspberrypi/firmware/tree/master/boot). I'll test with it.

jbech-linaro, 2018-07-05:
Right, having that said, be aware of the OP-TEE specifics, that are only on our fork (which is used by default in our setup):
https://github.com/linaro-swg/linux/commit/cc225a78910c37d78f8a00c80dcbf59ef7762884

Whatever dt/dtsi/dtb we end up with, the OP-TEE specifics must the there otherwise the probe in Linux kernel will fail.

jdeus, 2018-07-06:
Turns out it's u-boot related. If I do the following, then it boots:

`git clone --depth 1 git://git.denx.de/u-boot.git`
`cd u-boot`
`make rpi_3_defconfig`
`make CROSS_COMPILE=/root/devel/optee/toolchains/aarch64/bin/aarch64-linux-gnu-`

I also had to remove the following from config.txt, or it wouldn't boot for some reason:

`kernel_old=1`
`disable_commandline_tags=1`

Also, I copied bcm2710-rpi-3-b-plus.dtb over bcm2710-rpi-3-b.dtb (so I don't have to modify u-boot.env)

Now it boots but I get the following error message: **"tee-supplicant or tee device not found"** from [https://github.com/OP-TEE/build/blob/master/init.d.optee](init.d.optee)

In the rootfs tee-supplicant is however at /usr/sbin/tee-supplicant

Any idea?

jforissier, 2018-07-06:
@jdeus looks like the scripts looks for `/bin/tee-supplicant` but you're saying the correct path is `/usr/sbin/tee-supplicant`?

jbech-linaro, 2018-07-06:
Great progress!

> Now it boots but I get the following error message: "tee-supplicant or tee device not found" from init.d.optee

Weird, on the SD-card setup, that should autostart. On my NFS setup I need to manually start tee-supplicant. Can you simply try just that, i.,e.
``` bash
$ tee-supplicant &
# Then run xtest
$ xtest
```

If tee-supplicant still doesn't load, try
```bash
$ strace tee-supplicant
```
And that will probably give some hints.

jdeus, 2018-07-06:
@jforissier  That is correct.

@jbech-linaro Unfortunately it seems the `exit 1` from init.d.optee prevents the shell from starting and thus I get the blinking cursor usually immediately after the message "tee supplicant or TEE device not found"

jbech-linaro, 2018-08-20:
@jdeus did you manage to conclude anything here regarding 3B vs 3B+?

Hacklog24, 2019-01-20:
Hi,
I'm new to OP-TEE and wanted to learn more about it by running it on a Raspberry Pi 3B+. 
I think that I have the same problem, can someone help me please?
Here's the list of actions I just followed based on your build support page (https://github.com/OP-TEE/build):

    Installed the necessary dependencies;

    Installed repo;

3) Got OP-TEE's code through repo;
     repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml

4) Got the toolchains;

5) Ran the "make " command;

6) Flashed an SD card with the commands shown in "make img-help":

$ fdisk /dev/sdx   # where sdx is the name of your sd-card
   > p             # prints partition table
   > d             # repeat until all partitions are deleted
   > n             # create a new partition
   > p             # create primary
   > 1             # make it the first partition
   > <enter>       # use the default sector
   > +32M          # create a boot partition with 32MB of space
   > n             # create rootfs partition
   > p
   > 2
   > <enter>
   > <enter>       # fill the remaining disk, adjust size to fit your needs
   > t             # change partition type
   > 1             # select first partition
   > e             # use type 'e' (FAT16)
   > a             # make partition bootable
   > 1             # select first partition
   > p             # double check everything looks right
   > w             # write partition table to disk.

run the following as root
   $ mkfs.vfat -F16 -n BOOT /dev/sdx1
   $ mkdir -p /media/boot
   $ mount /dev/sdx1 /media/boot
   $ cd /media
   $ gunzip -cd /home/michele/devel/optee/build/../out-br/images/rootfs.cpio.gz | sudo cpio -idmv "boot/*"
   $ umount boot

run the following as root
   $ mkfs.ext4 -L rootfs /dev/sdx2
   $ mkdir -p /media/rootfs
   $ mount /dev/sdx2 /media/rootfs
   $ cd rootfs
   $ gunzip -cd /home/michele/devel/optee/build/../out-br/images/rootfs.cpio.gz | sudo cpio -idmv
   $ rm -rf /media/rootfs/boot/*
   $ cd .. && umount rootfs

But once the procedure is finished and the card inserted into the raspberry, the red LED flashes and      nothing happens on the screen. It's stuck on the rainbow screen.

Thank you very much
Best Regards,
Michelangelo

jbech-linaro, 2019-01-21:
@Hacklog24 , are using the [UART cable](https://github.com/OP-TEE/build/blob/master/docs/rpi3.md#61-debug-cable--uart-cable) or not? @igoropaniuk just [recently](https://github.com/OP-TEE/build/commit/42de80479fe7be3b28e71d2c45d7bc7aa5298366) added a patch that should make it work without the UART cable, but in case you're running an older commit than that, then it just works with the UART cable. Personally I'm only using the UART cable, I've never tried using HDMI myself.

igoropaniuk, 2019-01-21:
@Hacklog24
> I have an LCD display connected to the Raspberry,

It won't work, because it's not even configured in the default `config.txt` we're using.
Please consider HDMI connection or serial console cable (simple USB FTDI will work, here is setup details https://h3droid.com/blog/uart-cable)

For details check https://github.com/OP-TEE/build/blob/master/docs/rpi3.md#61-debug-cable--uart-cable

If you still plan to bring up DPI/LVDS display, check configuration flags (unfortunately I can't help you with any other suggestions regarding DPI display, as I don't have one) in https://www.raspberrypi.org/documentation/configuration/config-txt/video.md

And https://www.raspberrypi.org/forums/viewtopic.php?t=157109

Hacklog24, 2019-01-21:
Thank you very much for all the help and material.
So, I can try the connection with the HDMI and in case it does not work, try with the cable that you suggested? It’s right?

Thanks also for all the tips on the display.

jbech-linaro, 2019-01-31:
Issues like this has been sorted out and more up-to-date information can be found in: https://github.com/OP-TEE/optee_os/issues/2783#issuecomment-459359271, so I'm closing the ticket.

closed 2019-01-31 by jbech-linaro

johnphilby, 2019-03-19:
> Right, having that said, be aware of the OP-TEE specifics, that are only on our fork (which is used by default in our setup):
> [linaro-swg/linux@cc225a7](https://github.com/linaro-swg/linux/commit/cc225a78910c37d78f8a00c80dcbf59ef7762884)
> 
> Whatever dt/dtsi/dtb we end up with, the OP-TEE specifics must the there otherwise the probe in Linux kernel will fail.

@jbech-linaro How does one incorporate these changes when a new dtb such as bcm2710-rpi-3-b-plus.dtb [needs](https://github.com/raspberrypi/firmware/blob/master/boot/bcm2710-rpi-3-b-plus.dtb) to be used? Just cloning bcm2710-rpi-3-b.dtsi with a different name such as bcm2710-rpi-3-b-plus.dtsi in linux/arch/arm/boot/dts, as I have come to know, does not work. Any pointers on how this was done (git commit) would be helpful.

Would like to get optee on Rpi3 B + running with u-boot generated from the optee build system itself rather than hackish methods of copying from other sources.

jbech-linaro, 2019-03-20:
> Any pointers on how this was done (git commit) would be helpful.

Even though this doesn't really affect any platform code in OP-TEE I would have added another composite `PLATFORM` flag to the exiting RPI3 builds. That would made things look the same as for other platforms and platform flavors. So instead of using just `PLATFORM=rpi3` as today we would use `PLATFORM=rpi-3b` and `PLATFORM=rpi-3bplus`. By doing so we can "ifdef" in both build scripts as well as in OP-TEE platform code if needed.

Things touching this in our existing full developer setup for OP-TEE are:
https://github.com/OP-TEE/build/blob/master/rpi3.mk#L40
https://github.com/OP-TEE/build/blob/master/rpi3.mk#L166

> Would like to get optee on Rpi3 B + running with u-boot generated from the optee build system itself rather than hackish methods of copying from other sources.

Indeed.