OP-TEE/build #277: qemu build not working

simonjj22, 2018-07-04:
I am trying to build a fresh checkout for qemu but it fails with this error:

> /home/user/devel/optee/toolchains/aarch32/bin/../lib/gcc/arm-linux-gnueabihf/6.2.1/libgcc.a(_dvmd_lnx.o): In function `__aeabi_idiv0':
> 
> /home/tcwg-buildslave/workspace/tcwg-make-release/label/docker-trusty-amd64-tcwg-build/target/arm-linux-gnueabihf/snapshots/gcc-linaro-6.2-2016.11/libgcc/config/arm/lib1funcs.S:1359: undefined reference to `raise'
> 
> bios/link.mk:36: recipe for target '/home/user/devel/optee/build/../out/bios-qemu/bios.elf' failed

How do I fix this?

jbech-linaro, 2018-07-04:
I've been testing this from scratch earlier today and it is working fine. You have probably missed installing a dependency or diverged from our [official instructions of how to setup QEMU](https://github.com/OP-TEE/build#7-get-and-build-the-solution).

simonjj22, 2018-07-04:
@jbech-linaro Great, thanks for checking!
I created a new user and it did indeed work.
I had changed $HOME on my other account because I already had a version in ~/optee/devel could it be that the scripts determine the home directory differently?

jbech-linaro, 2018-07-04:
> could it be that the scripts determine the home directory differently?

No, I don't think so, everything we are doing is relative to the root of the repo forest. I've been using it from `$HOME`, `/mnt/...`, `/opt/...` etc. But anyhow, glad you got it to work!

closed 2018-07-04 by jbech-linaro

simonjj22, 2018-07-04:
It doesn't work if $HOME is different from /home/user!

simonjj22, 2018-07-04:
It also doesn't work in a different directory than /home/user/optee/devel if $HOME points to /home/user!

jbech-linaro, 2018-07-04:
I'm sure I follow you here, my QEMU setup is like this:
```bash
jbech@debby /media/jbech/SSHD_LINUX/devel/optee_projects/qemu/build [(no branch)]
$ pwd
/media/jbech/SSHD_LINUX/devel/optee_projects/qemu/build
jbech@debby /media/jbech/SSHD_LINUX/devel/optee_projects/qemu/build [(no branch)]
$ echo $HOME
/home/jbech
```
I.e., I'm having `$HOME` as `$HOME` is expected to be. But I still can have my QEMU setup in any other folder on any disk. I probably misunderstand something you are trying to tell me.

simonjj22, 2018-07-05:
Hm..no that's exactly what I am trying to tell you. Not sure what the problem is then.

simonjj22, 2018-07-05:
Damn. Seems I had some stale environment variables lying around. It works in a fresh shell.

jbech-linaro, 2018-07-05:
Alright! Problem solved! Great!

simonjj22, 2018-07-05:
@jbech-linaro Yeah, maybe add a comment to the readme where it tells you to use $HOME/optee/devel that any directory will work.

simonjj22, 2018-07-05:
Yes. Should I create a pull request? Changing this seems like a matter of seconds. :)

jbech-linaro, 2018-07-05:
> Yes. Should I create a pull request? Changing this seems like a matter of seconds. :)

Please go a head!

akbarsaleemt, 2018-08-22:
when I do make run only in qemu build after adding my own applicatio to optee_examples in qemu from optee_examples in optee I got following error
/******************************************************************************************/
CMake Error at basic/CMakeLists.txt:11 (target_link_libraries):
  Attempt to add link library "teec" to target "hello_world" which is not
  built in this directory.

CMake Error at basic/CMakeLists.txt:13 (install):
  install TARGETS given target "hello_world" which does not exist in this
  directory.

-- Configuring incomplete, errors occurred!
See also "/home/takkaakb/qemu/out-br/build/optee_examples-1.0/CMakeFiles/CMakeOutput.log".
package/pkg-generic.mk:211: recipe for target '/home/takkaakb/qemu/out-br/build/optee_examples-1.0/.stamp_configured' failed
make[2]: *** [/home/takkaakb/qemu/out-br/build/optee_examples-1.0/.stamp_configured] Error 1
Makefile:16: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/takkaakb/qemu/out-br'
common.mk:198: recipe for target 'buildroot' failed
make: *** [buildroot] Error 2

jbech-linaro, 2018-08-22:
> Attempt to add link library "teec" to target "hello_world" which is not
> built in this directory.

So, you need to remove references to `hello_world" in your "`basic`" TA's makefiles. I believe that is what the above message is saying, that you are trying to link to something that is not in this particular directory.