OP-TEE/build #283: no log about BL1 and BL2 on rpi3

tarichuo, 2018-07-18:
Hi all,

I built OP-TEE on rpi3 following the tutorial and xtest is successfully runned. But when I examine the booting log, I find that there is no log about BL1 and BL2 on arm-trusted-firmware. I think there should be something like this:
`NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.4(release):v1.4-329-g61e94684-dirty
NOTICE:  BL1: Built : 00:09:25, Nov  6 2017
NOTICE:  BL1: Booting BL2
NOTICE:  BL2: v1.4(release):v1.4-329-g61e94684-dirty
NOTICE:  BL2: Built : 00:09:25, Nov  6 2017
NOTICE:  BL1: Booting BL31
NOTICE:  BL31: v1.4(release):v1.4-329-g61e94684-dirty
NOTICE:  BL31: Built : 00:09:25, Nov  6 2017`

My booting log is below:
`
U-Boot 2016.03-gbc6d93c (Jul 17 2018 - 16:38:02 +0800)

DRAM:  944 MiB
RPI 3 Model B (0xa22082)
boot regs: 0x00000000 0x00000000 0x00000000 0x00000000
MMC:   bcm2835_sdhci: 0
reading uboot.env
In:    serial
Out:   lcd
Err:   lcd
Net:   Net Initialization Skipped
No ethernet found.
starting USB...
USB0:   Core Release: 2.80a
scanning bus 0 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
       scanning usb for ethernet devices... 1 Ethernet Device(s) found
Hit any key to stop autoboot:  0 
reading Image
10203648 bytes read in 870 ms (11.2 MiB/s)
reading bcm2710-rpi-3-b.dtb
13376 bytes read in 22 ms (593.8 KiB/s)
reading optee.bin
473088 bytes read in 57 ms (7.9 MiB/s)
## Flattened Device Tree blob at 01700000
   Booting using the fdt blob at 0x1700000
   reserving fdt memory region: addr=0 size=1000
   reserving fdt memory region: addr=8000000 size=2000000
   Loading Device Tree to 000000003ab2f000, end 000000003ab3543f ... OK

Starting kernel ...

## Transferring control to ARM TF (at address 8400000) (dtb at 3ab2f000)...
NOTICE:  BL3-1: v1.1(debug):1da4e15
NOTICE:  BL3-1: Built : 18:54:12, Jul 16 2018
INFO:    BL3-1: Initializing runtime services
INFO:    BL3-1: Initializing BL3-2
D/TC:0 add_phys_mem:539 TEE_SHMEM_START type NSEC_SHM 0x08000000 size 0x00400000
D/TC:0 add_phys_mem:539 TA_RAM_START type TA_RAM 0x08800000 size 0x01000000
D/TC:0 add_phys_mem:539 VCORE_UNPG_RW_PA type TEE_RAM_RW 0x08466000 size 0x0039a000
D/TC:0 add_phys_mem:539 VCORE_UNPG_RX_PA type TEE_RAM_RX 0x08420000 size 0x00046000
D/TC:0 add_phys_mem:539 TEE_RAM_START type TEE_RAM_RO 0x08400000 size 0x00020000
D/TC:0 add_phys_mem:539 CONSOLE_UART_BASE type IO_NSEC 0x3f200000 size 0x00200000
D/TC:0 verify_special_mem_areas:477 No NSEC DDR memory area defined
D/TC:0 add_va_space:578 type RES_VASPACE size 0x00a00000
D/TC:0 add_va_space:578 type SHM_VASPACE size 0x02000000
D/TC:0 dump_mmap_table:711 type TEE_RAM_RO   va 0x08400000..0x0841ffff pa 0x08400000..0x0841ffff size 0x00020000 (smallpg)
D/TC:0 dump_mmap_table:711 type TEE_RAM_RX   va 0x08420000..0x08465fff pa 0x08420000..0x08465fff size 0x00046000 (smallpg)
D/TC:0 dump_mmap_table:711 type TEE_RAM_RW   va 0x08466000..0x087fffff pa 0x08466000..0x087fffff size 0x0039a000 (smallpg)
D/TC:0 dump_mmap_table:711 type SHM_VASPACE  va 0x08800000..0x0a7fffff pa 0x00000000..0x01ffffff size 0x02000000 (pgdir)
D/TC:0 dump_mmap_table:711 type NSEC_SHM     va 0x0a800000..0x0abfffff pa 0x08000000..0x083fffff size 0x00400000 (pgdir)
D/TC:0 dump_mmap_table:711 type IO_NSEC      va 0x0ac00000..0x0adfffff pa 0x3f200000..0x3f3fffff size 0x00200000 (pgdir)
D/TC:0 dump_mmap_table:711 type RES_VASPACE  va 0x0ae00000..0x0b7fffff pa 0x00000000..0x009fffff size 0x00a00000 (pgdir)
D/TC:0 dump_mmap_table:711 type TA_RAM       va 0x0b800000..0x0c7fffff pa 0x08800000..0x097fffff size 0x01000000 (pgdir)
D/TC:0 core_mmu_entry_to_finer_grained:653 xlat tables used 1 / 5
D/TC:0 core_mmu_entry_to_finer_grained:653 xlat tables used 2 / 5
I/TC:  
D/TC:0 init_canaries:164 #Stack canaries for stack_tmp[0] with top at 0x8471f38
D/TC:0 init_canaries:164 watch *0x8471f3c
D/TC:0 init_canaries:164 #Stack canaries for stack_tmp[1] with top at 0x8472778
D/TC:0 init_canaries:164 watch *0x847277c
D/TC:0 init_canaries:164 #Stack canaries for stack_tmp[2] with top at 0x8472fb8
D/TC:0 init_canaries:164 watch *0x8472fbc
D/TC:0 init_canaries:164 #Stack canaries for stack_tmp[3] with top at 0x84737f8
D/TC:0 init_canaries:164 watch *0x84737fc
D/TC:0 init_canaries:165 #Stack canaries for stack_abt[0] with top at 0x8467138
D/TC:0 init_canaries:165 watch *0x846713c
D/TC:0 init_canaries:165 #Stack canaries for stack_abt[1] with top at 0x8467d78
D/TC:0 init_canaries:165 watch *0x8467d7c
D/TC:0 init_canaries:165 #Stack canaries for stack_abt[2] with top at 0x84689b8
D/TC:0 init_canaries:165 watch *0x84689bc
D/TC:0 init_canaries:165 #Stack canaries for stack_abt[3] with top at 0x84695f8
D/TC:0 init_canaries:165 watch *0x84695fc
D/TC:0 init_canaries:167 #Stack canaries for stack_thread[0] with top at 0x846b638
D/TC:0 init_canaries:167 watch *0x846b63c
D/TC:0 init_canaries:167 #Stack canaries for stack_thread[1] with top at 0x846d678
D/TC:0 init_canaries:167 watch *0x846d67c
D/TC:0 init_canaries:167 #Stack canaries for stack_thread[2] with top at 0x846f6b8
D/TC:0 init_canaries:167 watch *0x846f6bc
D/TC:0 init_canaries:167 #Stack canaries for stack_thread[3] with top at 0x84716f8
D/TC:0 init_canaries:167 watch *0x84716fc
I/TC:  OP-TEE version: 3.2.0-17-gb048d32 #1 Mon Jul 16 10:53:20 UTC 2018 aarch64
D/TC:0 tee_ta_register_ta_store:534 Registering TA store: 'REE' (priority 10)
D/TC:0 tee_ta_register_ta_store:534 Registering TA store: 'Secure Storage TA' (priority 9)
D/TC:0 mobj_mapped_shm_init:559 Shared memory address range: 8800000, a800000
I/TC:  Initialized
D/TC:0 init_primary_helper:917 Primary CPU switching to normal world boot
INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address = 0x80000
INFO:    BL3-1: Next image spsr = 0x3c9
D/TC:  generic_boot_cpu_on_handler:956 cpu 1: a0 0x0
D/TC:  init_secondary_helper:941 Secondary CPU Switching to normal world boot
D/TC:  generic_boot_cpu_on_handler:956 cpu 2: a0 0x0
D/TC:  init_secondary_helper:941 Secondary CPU Switching to normal world boot
D/TC:  generic_boot_cpu_on_handler:956 cpu 3: a0 0x0
D/TC:  init_secondary_helper:941 Secondary CPU Switching to normal world boot
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.6.3-17583-g7d976fc (hjy@ubuntu) (gcc version 6.2.1 20161016 (Linaro GCC 6.2-2016.11) ) #1 SMP Mon Jul 16 19:39:58 CST 2018
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] debug: ignoring loglevel setting.
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.`

Any help greatly appreciated!

vchong, 2018-07-18:
Maybe because rpi3 uses RESET_TO_BL31.

https://github.com/ARM-software/arm-trusted-firmware/blob/master/plat/rpi3/platform.mk#L73
https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/user-guide.rst
RESET_TO_BL31: Enable BL31 entrypoint as the CPU reset vector instead of the BL1 entrypoint.

tarichuo, 2018-07-19:
My repo is synchronized by `rpi3.xml` and I grep `RESET_TO_BL31` in root folder. There is a `RESET_TO_BL31		:= 0` in `./arm-trusted-firmware/Makefile`. So I think it is not the reason.

vchong, 2018-07-19:
Ok, `rpi3.xml` uses a fork, i.e. https://github.com/linaro-swg/arm-trusted-firmware/blob/1da4e15529a32fa244f5e3effc9a90549beb1a26. If you look at `plat/rpi3/platform.mk`, it doesn't have `RESET_TO_BL31`, but it only builds `BL31_SOURCES`. No BL1 and BL2 sources.

tarichuo, 2018-07-19:
Yes, just looked at `plat/rpi3/platform.mk`, it only builds `BL31_SOURCES`. Thanks for your help @vchong .

closed 2018-07-19 by tarichuo