OP-TEE/build #285: Make failed on hikey

jenkin233, 2018-07-20:
Hello,
I want to build OP-TEE on hikey 620 Board, and get the project as below.
 `repo init -u https://github.com/OP-TEE/manifest.git -m hikey.xml
repo sync
cd build
make toolchains
`
But when I input  "make" command in the terminal,an error occurs, here's the error message:
`ta/avb//entry.c:5:30: fatal error: tee_internal_api.h: No such file or directory
 #include <tee_internal_api.h>
                              ^
compilation terminated.
`
I guess there's something wrong with the Makefile, for example  the head files are not included when compiling,  right?
Thanks.

jforissier, 2018-07-20:
Hi @jenkin233,

The include paths (`-I`) are OK, but it looks like we have a dependency issue. The AVB TA might be compiled before `tee_internal_api.h` is copied to `out/arm/export-ta_arm32/include/tee_internal_api.h`.

Try `make -j2`, several times if needed.

jbech-linaro, 2018-08-06:
This hasn't been resolved? What puzzle me is that IBART have done almost 50 builds for HiKey 620 the last month and I haven't seen this issue there (I'm building with `-j8` there).

jforissier, 2018-08-06:
@jbech-linaro make (all) works OK it seems, but it is pure luck, because:

```
$ rm -rf out
$ make out/arm-plat-vexpress/ta/avb/entry.o
  CHK     out/arm-plat-vexpress/conf.mk
  UPD     out/arm-plat-vexpress/conf.mk
  CHK     out/arm-plat-vexpress/include/generated/conf.h
  UPD     out/arm-plat-vexpress/include/generated/conf.h
  CC      out/arm-plat-vexpress/ta/avb/entry.o
cc1: warning: out/arm-plat-vexpress/export-ta_arm32/include: No such file or directory [-Wmissing-include-dirs]
ta/avb/entry.c:5:30: fatal error: tee_internal_api.h: No such file or directory
compilation terminated.
mk/compile.mk:146: recipe for target 'out/arm-plat-vexpress/ta/avb/entry.o' failed
make: *** [out/arm-plat-vexpress/ta/avb/entry.o] Error 1
```

`make out/arm-plat-vexpress/ta/avb/user_ta_header.o` fails is a similar fashion.

jforissier, 2018-08-10:
Fixed in master.

closed 2018-08-10 by jforissier