OP-TEE/build #296: Same problem as in issue #274, make for rpi3 failed to create uboot.env

dmeignan, 2018-08-26:
I am opening another issue, but feel free to attach this to the previous closed issue https://github.com/OP-TEE/build/issues/274

In short, make fails to access `uboot.env` which is indeed nonexistent at this step.
I have the exact same output as the initial reported issue, and a similar configuration:
Freshly installed Ubuntu on 64 bits system, and I try to compile OP-TEE for rpy3b using `rpi3.xml` manifest. Note that I was unable to retrieve the `rpi3_stable.xml` manifest.

jforissier, 2018-08-26:
There is no `*_stable.xml` file any more. You clone a release branch instead (`-b 3.2.0`).

dmeignan, 2018-08-27:
> There is no *_stable.xml file any more. You clone a release branch instead (-b 3.2.0).

Thanks for the information. In this case, maybe the documentation in [OP-TEE build.git - 6. Manifests](https://www.op-tee.org/build/#6-manifests) should be adjusted and the column `stable` should provide the build version instead of the xml file. I am not sure if I have to open an issue in the [optee_website repository](https://github.com/OP-TEE/optee_website/issues) for this.

jbech-linaro, 2018-08-27:
@dmeignan , please see and review #297 that I just created.

jbech-linaro, 2018-08-27:
Regarding the issue itself. I just tried and I still cannot reproduce the error. It works just fine here on my side. I have the file there after building
```bash
$ ll ../out
total 96
drwxr-xr-x  2 jbech jbech  4096 Aug 27 16:14 .
drwxr-xr-x 17 jbech jbech  4096 Aug 27 16:19 ..
-rw-r--r--  1 jbech jbech 32768 Aug 27 16:14 head.bin
-rw-r--r--  1 jbech jbech 66480 Aug 27 16:14 head.o
-rw-r-----  1 jbech jbech 16384 Aug 27 16:14 uboot.env
```
Likewise it seems like there are dependencies all the way back to the `$(RPI3_UBOOT_ENV)` if you follow the dependency chain in `rpi3.mk`. I wonder if it could be that you're lacking some other tool on your system used when building, i.e., something that should be installed with apt-get. Another thing, would you mind paste the complete build log after doing (from the root of the repo forest and also please save/backup/commit changes before running this since this will delete everything not yet committed).
```bash
$ alias rpo_clean_all='repo forall -c '\''echo Cleaning ... $REPO_PATH && git clean -xdf && git checkout -f'\'''
$ rpo_clean_all
$ rm -rf out out-br
$ repo sync -j3 -d
$ cd build
$ make -j4 2>&1 | tee build.log
```

Then attach `build.log` or post it at some paste service.

dmeignan, 2018-08-28:
I was able to finalize the build by running `make u-boot-rpi-bin` separately.
I will try to take the time to clean and rebuild (with log), so I can post the log..

jbech-linaro, 2018-08-28:
Just an hour ago there were a couple of new patches merged for RPi3 which contains quite a few updates. I haven't tested this myself, but it might behave a bit different compared to yesterday. But I plan to give it a try myself soon.

dmeignan, 2018-08-28:
Just another small point, I used the steps described on https://www.op-tee.org/build/ website and not the README.md of the build repository (I now use to this last updated documentation).
In the prerequisites, there is now `expect` installed, which was not installed in the prerequisites step of https://www.op-tee.org/build/
I don't know if it makes a difference.

jbech-linaro, 2018-08-28:
If everything works, then that should just be a mirror. I.e., our web admins are copying information from various `github.com/OP-TEE/*`gits over to optee.org. In the past this was done once a day, but I think it's no longer done automatically. @kylekirkby, correct? For most up-to-date information it's best to use the source directly instead of optee.org. Having that said, we're having discussions in general about our documentation, since we've identified some problems right now and what you brought up is one of them.

> there is now expect installed, which was not installed in the prerequisites step of https://www.op-tee.org/build/
> I don't know if it makes a difference.

I don't think so, `expect` is only used when testing QEMU using Travis and/or Shippable.

dmeignan, 2018-10-30:
With the last update https://github.com/OP-TEE/build/commit/2e947f2ee5ae8d2fa3624f90c893f901187dd0b0 on Aug 28, it seems that the build procedure has no problem. I tried after cleaning all repo and build caches and I had no problem.
However, a colleague of mine had the same initial problem when he tried to build the stable version 3.2.0, but I am not sure if version 3.2.0 is still considered a stable version.

jbech-linaro, 2018-10-30:
> @dmeignan : but I am not sure if version 3.2.0 is still considered a stable version.

Of course it is still considered as stable. Everything tagged with a release tag was stable at the time of testing and what is happening in the future doesn't change a particular release. Regarding 3.2.0, it was tested successfully by both Igor ([comment](https://github.com/OP-TEE/optee_os/pull/2404#issuecomment-399413577)) and me ([comment](https://github.com/OP-TEE/optee_os/pull/2404#issuecomment-399892169)). So two developers was able to run 3.2.0 successfully on RPi3. To rule out eventual problems I'd recommend you colleague to just wipe the entire forest and re-start from scratch. That is typically what I'm doing when testing a release tag.

jedichen121, 2018-11-05:
I tried to build the 3.2.0 yesterday and the make failed at uboot.env, reporting "No such file or directory". However, I was able to build 3.3.0 previously. Though that build hangs at crng init done. I'm using Ubuntu 18.04 by the way.

jbech-linaro, 2018-11-06:
> @jedichen121 Not sure if this is because I didn't clean up the previous build properly.

Probably yes.

@dmeignan @jedichen121 @Rusack @gkarop ... so a general update here on my side. To see whether I can see the issues on latest using RPi3 (model B, not B+), I've today rebuilt it all from scratch, deployed and then running xtest and as usual everything works as expected here on my side. So unfortunately I cannot reproduce the error(s) you guys are seeing. This is what I've done:

```bash
$ rm -rf rpi3
$ mkdir rpi3
$ cd rpi3/
$ repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml --reference /media/jbech/SSHD_LINUX/devel/optee_projects/reference
$ repo sync -j3
$ # I'm using a local mirror of the toolchain for all my OP-TEE gits
$ ln -s ../reference/toolchains .
$ cd build
$ time make -j8 
$ make img-help 
...
$ sudo mount /dev/sdi1 /media/boot
$ sudo rm -rf /media/boot/*
$ ll /media/boot/*
$ gunzip -cd /media/jbech/SSHD_LINUX/devel/optee_projects/rpi3/build/../out-br/images/rootfs.cpio.gz | sudo cpio -idmv "boot/*"
$ sudo umount boot
$ sudo mount /dev/sdi2 /media/rootfs
$ sudo rm -rf /media/rootfs/*
$ ll /media/rootfs/
$ cd rootfs/
$ gunzip -cd /media/jbech/SSHD_LINUX/devel/optee_projects/rpi3/build/../out-br/images/rootfs.cpio.gz | sudo cpio -idmv
$ sudo rm -rf /media/rootfs/boot/*
$ cd .. && sudo umount rootfs
```

Then it boots up nicely, full boot log can be seen here: http://ix.io/1r76
```bash
...
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.5(debug):v1.5-620-g8ae41aec
NOTICE:  BL1: Built : 03:50:44, Nov  6 2018
...
NOTICE:  BL31: v1.5(debug):v1.5-620-g8ae41aec
NOTICE:  BL31: Built : 03:50:44, Nov  6 2018
...
I/TC: OP-TEE version: 3.3.0-13-g7edfb8f5 #1 Tue Nov  6 02:50:08 UTC 2018 aarch64
...
U-Boot 2018.09-rc1-00230-ga032e0a6ae (Nov 06 2018 - 03:49:32 +0100)
...
Starting kernel ...
...
[    0.000000] Linux version 4.14.56-v8 (jbech@debby) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018-08 (arm-rel-8.23))) #1 SMP PREEMPT Tue Nov 6 03:52:31 CET 2018
...
Welcome to Buildroot, type root or test to login
buildroot login: [    7.533697] random: crng init done
root
# 
```

And then xtest ....
```bash

f testsuite regression:
regression_1001 OK
regression_1002 OK
regression_1003 OK
regression_1004 OK
regression_1005 OK
regression_1006 OK
regression_1007 OK
regression_1008 OK
regression_1009 OK
regression_1010 OK
regression_1011 OK
regression_1012 OK
regression_1013 OK
regression_1015 OK
regression_1016 OK
regression_1017 OK
regression_1018 OK
regression_1019 OK
regression_4001 OK
regression_4002 OK
regression_4003 OK
regression_4004 OK
regression_4005 OK
regression_4006 OK
regression_4007 OK
regression_4008 OK
regression_4009 OK
regression_4010 OK
regression_4011 OK
regression_4012 OK
regression_5006 OK
regression_6001 OK
regression_6002 OK
regression_6003 OK
regression_6004 OK
regression_6005 OK
regression_6006 OK
regression_6007 OK
regression_6008 OK
regression_6009 OK
regression_6010 OK
regression_6012 OK
regression_6013 OK
regression_6014 OK
regression_6015 OK
regression_6016 OK
regression_6017 OK
regression_6018 OK
regression_6019 OK
regression_6020 OK
regression_7001 OK
regression_7002 OK
regression_7003 OK
regression_7004 OK
regression_7005 OK
regression_7006 OK
regression_7007 OK
regression_7008 OK
regression_7009 OK
regression_7010 OK
regression_7013 OK
regression_7016 OK
regression_7017 OK
regression_7018 OK
regression_7019 OK
regression_8001 OK
regression_8002 OK
regression_8101 OK
regression_8102 OK
regression_8103 OK
regression_2001 OK
regression_2002 OK
regression_2003 OK
regression_2004 OK
+-----------------------------------------------------
16081 subtests of which 0 failed
74 test cases of which 0 failed
0 test case was skipped
TEE test application done!
#
```

It's all running latest available as of 2 hours ago .... which can be seen on my clone command. But for reference, these are the commits in use.
```bash
$ rpo_rev 
arm-trusted-firmware -- 8ae41aec Merge pull request #1505 from Yann-lms/mmc_delays
build -- 6e720ec qemu, qemu_v8: set QEMU_SMP ?= 2
buildroot -- 339d550e92 Update for 2018.08
firmware -- 3fb63c413c firmware: FXL6408/GPIOman: Support config of termination via dt-blob firmware: GPIO expander: Add API to reconfigure pins firmware: GPIOMAN: Add API to reconfigure pins firmware: Mailbox service: Add command to reconfigure GPIO setup firmware: FXL6408: Return success code on reading status of an output firmware: GPIO expander: Add set/get_config functions to dummy driver
linux -- d639e3a10d8a arm64: dt: RPi3: Add optee node
optee_benchmark -- 845a783 benchmark: misc. cosmetic fixes
optee_client -- c48bc3b tee-supplicant: cmake: enable RPMB emulation by default
optee_examples -- 552c9d0 secure_storage: Add CMakeLists.txt
optee_os -- 7edfb8f5 core: mobj.c: sort #includes
optee_test -- 0142c7d Android.mk: add required header files to LOCAL_ADDITIONAL_DEPENDENCIES
u-boot -- a032e0a6ae travis: give every job a name
```

jedichen121, 2018-11-07:
@jbech-linaro Thank you very much for the detailed info. 
After some debugging I was able to see the login prompt. There are two problems related: 

1. The SD card I used had a problem. During formatting it has some linux signature remained. Switching to another card or format it again fixed this issue. 
2. The login prompt just doesn't show up on monitor when using HDMI cable. In one test I used serial-usb cable and HDMI cable at the same time. Even I can see the login prompt from picocom, it doesn't show up on monitor. Not sure why. 

But still, I met new problems. I tried to type "root" to login but failed. I connect keyboard to rpi3 directly and it can detects the keyboard. But I couldn't see any commands I typed show up. I tried three different keyboards I have and tried typing commands in picocom but none worked. I'm not very familiar with serial debugging. Have you seen this problem before?

jedichen121, 2018-11-07:
I've got another question about using repo reference. In the document it's saying 
`Start by setup a clean forest that you will not touch, in this example, let us call that optee-ref and put that under for $HOME/devel/optee-ref. This step will take roughly an hour.`
I'm not sure if I understand this correct, does this step corresponds to running the following command?
```
repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml
repo sync -j3
```
In my case this doesn't need an hour, but about 10mins. It's the `make` part that needs an hour. 

So after this initial step, I should never touch this folder again, but only use this as a reference in creating other optee working trees? Then I'm not seeing why this can speed things up, as it's the make part that takes the longest time. Or should I run `make` after `repo sync`? Thank you for any clarification.

jbech-linaro, 2018-11-08:
> @jedichen121  I have and tried typing commands in picocom but none worked. I'm not very familiar with serial debugging. Have you seen this problem before?

I'm the opposite, I never ever used the HDMI and keyboards directly connected to the RPi3. I'm always just using the UART in a terminal window. The only other cable used (except power :smile:) is the Ethernet cable, but that is only when I'm using NFS.

I think it is like this, traditionally when using Raspberry Pi, people are [downloading Raspian](https://www.raspberrypi.org/downloads/) or something similar that is officially supported by the RPi organization. That are images that has been tweaked and configured to behave nicely with RPi devices. I.e., there is a special kernel, with all the correct kconfig flags etc enabled, all drivers installed etc.

Compare this to our setup. We are using Buildroot to put together a root fs that contains **enough** to
  a) Boot up the device
  b) Run xtest for OP-TEE without regressions
  c) Have some extra developer tools available, strace, gdbserver etc.

When it comes to the kernel we haven't spent any time on tweaking the config so "everything else" works. Likewise we haven't looked into adding/enabling drivers, libraries etc for other things. Simply put, the RPi3 setup for OP-TEE that we are providing is a barebone setup aimed for OP-TEE development over a serial connection. Having that said, Philip Attfield from Sequitur Labs (who did the initial RPi3 port for OP-TEE) gave a couple of workshops in the past where he had OP-TEE running with Raspbian. So it is possible to make it work, but it's not on our radar to do that (but we would happily accept patches in case someone has been able to do that on their own).

> @jedichen121 **repo ...**  In my case this doesn't need an hour, but about 10mins. It's the make part that needs an hour.

It totally depends on the speed of your internet connection and the speed at the other end (the ones hosting the projects). One hour is probably a quite high "guesstimate". But it's not uncommon that it takes 15-30 minutes to clone everything from scratch (without any caches). 10 minutes sounds good, you probably have a good connection!

> @jedichen121  So after this initial step, I should never touch this folder again, but only use this as a reference in creating other optee working trees?

Exactly, that is what I've tried to explain [here](https://github.com/OP-TEE/build#7101-reference-existing-project-to-speed-up-repo-sync). I have such a "reference" tree containing lots of gits that are usually used in my projects.

>  @jedichen121 Then I'm not seeing why this can speed things up, as it's the make part that takes the longest time. Or should I run make after repo sync?

If you have a mirror like the one above, let's call that tree "my-mirror", then when you create a brand new forest, then you reference this mirror. i.e., 

```bash
repo init -u https://github.com/OP-TEE/manifest.git --reference <path-to>/my-mirror
```
By doing so the next step:
```bash
repo sync -j3
```
Will take ~10-20 seconds instead of 10-30 minutes. That is where the major speed up is when it comes to cloning gits. Then you should also [install ccache](https://github.com/OP-TEE/build#7102-use-ccache), so you can used cached objects files when compiling, which save even more time after doing an initial build. With repo mirror + ccache, new clones + builds usually takes around ~10 minutes in **total** (as an example, see full clone + flash + build + run xtest times at [IBART](http://optee.mooo.com:5000/)).

jbech-linaro, 2019-01-31:
Issues like this has been sorted out and more up-to-date information can be found in: https://github.com/OP-TEE/optee_os/issues/2783#issuecomment-459359271, so I'm closing the ticket.

closed 2019-01-31 by jbech-linaro