OP-TEE/build #301: Build error after adding gst1-interpipe package

ArunSahu42, 2018-09-04:
I got the following error after adding gst1-interpipe and while compiling. Please anyone suggest me how I will solve this issue.

  GEN     /home/sahuuaru/Q8TEE/out-br/Makefile
>>> gst1-interpipe 10dba89eecc200e49f0b432e299f1c8a4312b657 Downloading
  CHK     include/generated/utsrelease.h
Reinitialized existing Git repository in /home/sahuuaru/Q8TEE/buildroot/dl/gst1-interpipe/git/.git/
  CHK     scripts/mod/devicetable-offsets.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
Fetching all references
Could not fetch special ref '10dba89eecc200e49f0b432e299f1c8a4312b657'; assuming it is not special.
warning: refname '10dba89eecc200e49f0b432e299f1c8a4312b657' is ambiguous.
Git normally never creates a ref that ends with 40 hex characters
because it will be ignored when you just specify 40-hex. These refs
may be created by mistake. For example,

  git checkout -b $br $(git rev-parse ...)

where "$br" is somehow empty and a 40-hex ref is created. Please
examine these refs and maybe delete them. Turn this message off by
running "git config advice.objectNameWarning false"
ERROR: gst1-interpipe-10dba89eecc200e49f0b432e299f1c8a4312b657.tar.gz has wrong sha256 hash:
ERROR: expected: 3f8cc5cf4861e5836938c66415b47555550c93b06b1bc9faed83e89bacf705cf
ERROR: got     : 5c1ed271062458b01b696ed543f4a6fe3ca1f04301ee2300bf9d48c2d26b0303
ERROR: Incomplete download, or man-in-the-middle (MITM) attack
--2018-09-04 10:45:43--  http://sources.buildroot.net/gst1-interpipe/gst1-interpipe-10dba89eecc200e49f0b432e299f1c8a4312b657.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.25.211.19, 104.25.210.19, 2400:cb00:2048:1::6819:d213, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.25.211.19|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-09-04 10:45:44 ERROR 404: Not Found.

--2018-09-04 10:45:44--  http://sources.buildroot.net/gst1-interpipe-10dba89eecc200e49f0b432e299f1c8a4312b657.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.25.210.19, 104.25.211.19, 2400:cb00:2048:1::6819:d313, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.25.210.19|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-09-04 10:45:44 ERROR 404: Not Found.

package/pkg-generic.mk:133: recipe for target '/home/sahuuaru/Q8TEE/out-br/build/gst1-interpipe-10dba89eecc200e49f0b432e299f1c8a4312b657/.stamp_downloaded' failed
make[2]: *** [/home/sahuuaru/Q8TEE/out-br/build/gst1-interpipe-10dba89eecc200e49f0b432e299f1c8a4312b657/.stamp_downloaded] Error 1
Makefile:16: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/home/sahuuaru/Q8TEE/out-br'
common.mk:198: recipe for target 'buildroot' failed
make: *** [buildroot] Error 2
make: *** Waiting for unfinished jobs....
  Building modules, stage 2.
  MODPOST 205 modules
make[1]: Leaving directory '/home/sahuuaru/Q8TEE/linux'

jforissier, 2018-09-04:
Hi @ArunSahu42,

You probably faced a temporary issue on the buildroot server. It worked for me with the following messages (note that I do have the warning about the refname but the HTTP download then proceeds fine):
```
>>> gst1-interpipe 10dba89eecc200e49f0b432e299f1c8a4312b657 Downloading
Initialized empty Git repository in /home/jerome.forissier/.cache/buildroot-dl/gst1-interpipe/git/.git/
Fetching all references
remote: Counting objects: 342, done.
remote: Compressing objects: 100% (117/117), done.
remote: Total 342 (delta 94), reused 150 (delta 67), pack-reused 158
Receiving objects: 100% (342/342), 124.99 KiB | 825.00 KiB/s, done.
Resolving deltas: 100% (167/167), done.
From https://github.com/RidgeRun/gst-interpipe
 * [new branch]      develop                                     -> origin/develop
 * [new branch]      feature/cleanup-build-warnings              -> origin/feature/cleanup-build-warnings
 * [new branch]      feature/fix-segment-before-caps             -> origin/feature/fix-segment-before-caps
 * [new branch]      feature/fix-tests                           -> origin/feature/fix-tests
 * [new branch]      feature/start-stream-with-valid-srcbasetime -> origin/feature/start-stream-with-valid-srcbasetime
 * [new branch]      gh-pages                                    -> origin/gh-pages
 * [new branch]      master                                      -> origin/master
warning: refname '10dba89eecc200e49f0b432e299f1c8a4312b657' is ambiguous.
Git normally never creates a ref that ends with 40 hex characters
because it will be ignored when you just specify 40-hex. These refs
may be created by mistake. For example,

  git checkout -b $br $(git rev-parse ...)

where "$br" is somehow empty and a 40-hex ref is created. Please
examine these refs and maybe delete them. Turn this message off by
running "git config advice.objectNameWarning false"
Submodule 'common' (git://anongit.freedesktop.org/gstreamer/common) registered for path 'common'
Cloning into '/home/jerome.forissier/.cache/buildroot-dl/gst1-interpipe/git/common'...
Submodule path 'common': checked out 'ac2f647695e7bd4b433ea108ee1d0e23901797d4'
gst1-interpipe-10dba89eecc200e49f0b432e299f1c8a4312b657.tar.gz: OK (sha256: 3f8cc5cf4861e5836938c66415b47555550c93b06b1bc9faed83e89bacf705cf)
>>> gst1-interpipe 10dba89eecc200e49f0b432e299f1c8a4312b657 Extracting
...
```

You should try again.

ArunSahu42, 2018-09-04:
Ok. I will try it again. Thanks

jbech-linaro, 2019-01-31:
I'm closing the ticket, either because it has already been answered or that it is no longer relevant or it could be lack of response from the author. Having that said, feel free to re-open the ticket if you have more to add to the ticket.

closed 2019-01-31 by jbech-linaro