OP-TEE/build #303: Arrow keys not working in menu

r4rounak, 2018-09-11:
I followed the instructions for Hikey 960. Everything got flashed without any errors. After reboot I pressed Esc in the console. The menu was displayed but the arrow keys were not working. Return key worked, but nothing else did. What am I doing wrong ?

jforissier, 2018-09-11:
Try `^` (caret) and `v`. Apparently, Grub is not smart enough to understand the real arrow keypresses :-/

r4rounak, 2018-09-12:
@jforissier Your solution worked. The arrow keys do work in a similar interface that is shown after the recovery binaries are loaded using hikey_idt and I press RETURN in the console. But they don't work in the menu that is shown after reboot. I wonder why the implementations are different.
I am closing the issue because my problem is solved.

closed 2018-09-12 by r4rounak