OP-TEE/build #304: Run error

Sun2018421, 2018-09-12:
I found this error when I first ran it.
The device is a virtual machine ubuntu 16.04 120g hard disk 4g memory.
I feel that it is a memory problem.
[image]
[image]
[image]

Sun2018421, 2018-09-12:
i am poor ai English

jforissier, 2018-09-12:
The root fs (`./out-br/images/rootfs.cpio.gz`) failed to build for some reason. Try `make buildroot` and look for errors.

Sun2018421, 2018-09-12:
Thanks you ,i will try it

Sun2018421, 2018-11-18:
I want to install OP-TEE on rpi3 , so i download rp3.xml . After making toolchains ,  i make all run it on ubuntu 16.04LTS . 
It tells me 
make[1]:leaving directory '/home/sun/devel/op-tee/out-br'
make:*** NO rule to make target 'run'. stop
why?

jforissier, 2018-11-18:
Target "run" is for emulated targets only (QEMU, FVP). You probably need to "make flash"

Sun2018421, 2018-11-18:
it tells me 
make:*** No rule to make target 'flash' . stop
I run it on my ubuntu , but i want to install OP-TEE on my rp3 . should i  run it on rp3 or ubuntu.

Sun2018421, 2018-11-19:
Thanks you very much , i will read it carefully again . :)

closed 2018-11-25 by Sun2018421