OP-TEE/build #305: What should I flash in the system partition for Hikey 960?

r4rounak, 2018-09-12:
The **flash** target of make flashes ptable, xloader, fastboot, fip, nvme and boot partitions. Nothing is flashed to the **system** partition. Is this supposed to work ? I am asking because I tried and it didn't work. I flashed the partitions and rebooted into normal mode (switches ON-OFF-OFF). I got the menu, but when I chose grub, it just returned me to the menu. What am I missing ?

r4rounak, 2018-10-01:
No response ? Do I need to add more information to get a response ?

jbech-linaro, 2018-10-01:
It's simply not needed, the root fs (Buildroot) is included in boot image:
https://github.com/OP-TEE/build/blob/master/hikey960.mk#L253
https://github.com/OP-TEE/build/blob/master/hikey960.mk#L347

r4rounak, 2018-10-01:
So it should just boot to a shell prompt, right ? Or am I missing something ?

jbech-linaro, 2018-10-01:
If it should work, boot to a shell? Yes, it should and I'm pretty confident it does, @jforissier did some extensive testing with HiKey960 for the 3.3.0 release later this week, see his comment [here](https://github.com/OP-TEE/optee_os/pull/2552#issuecomment-425834277) posted just two hours ago.

r4rounak, 2018-10-01:
In my case it waits for a while for me to press ESCAPE for boot options, then starts Fastboot mode. Is this the expected behaviour ? After it started Fastboot mode, I pressed return to get the menu, from which I selected Boot Manager followed by Grub. On choosing Grub it just returned me back to the Boot Manager menu.

jbech-linaro, 2018-10-01:
I don't own a 960 myself, but I'd presume it behaves similar to the old HiKey, i.e. with the jumpers correctly set, it should go right into Grub and load the kernel etc. On the old HiKey you'd get different behavior during boot depending on the jumpers. Either it went into fastboot for a short while before continuing or it went directly into grub. @jforissier how to you enabled the switches/jumpers on your 960 for a normal boot?

jforissier, 2018-10-01:
- If switches are set to On/Off/On, then the boot is forced into fastboot. That's what is described when you run "make flash".
- If the switches are set to On/Off/Off, normal boot from eMMC is selected. After pressing the reset button, you should see (1) some debug text, (2) "Press escape for boot options" (lasts 1 second only), (3) The Grub menu with 3 entries: 1/ "GNU/Linux (eMMC) UART6 (v2 or newer board)", 2/ "GNU/Linux (eMMC) UART5 (v1 board", and 3/ "Fastboot". If you wait for 10 seconds the first entry is automatically selected.

Personally I never change the switches and select 3/ instead to go into fastboot mode and reflash the board (perhaps we should update the "make flash" help text?)

r4rounak, 2018-10-01:
@jforissier 
My switches were set to On/Off/Off
"Press escape for boot options" seems to last longer than a second.
After the delay, it does not show the three options you mentioned. Instead, it starts Fastboot mode.
I have built from the 3.2.0 branch. Does the behaviour you have described only work in master branch ?

jforissier, 2018-10-01:
@r4rounak I'm using the master branch, although 3.2.0 should be similar...

rounaka, 2018-10-05:
Everything worked as you described when I built from master branch.

closed 2018-10-05 by rounaka