OP-TEE/build #306: Linux kernel hang at crng init done

Rusack, 2018-09-13:
I'm trying to install OP-TEE and TF-A on the Raspberry Pi 3 (model B), therefore I used the manifest file for the rpi3 and the  `repo` command (Didn't specify the release or anything, just took the last one). The build went fine, I then installed the result on the micro SD following instructions of the `make img-help` output. My problem is appearing at the linux kernel boot, it hangs forever after random : `crng init done , I'm connected by UART. 
I'm joining the build log as well as the boot log.
[build.log](https://github.com/OP-TEE/build/files/2379198/build.log)
[boot.log](https://github.com/OP-TEE/build/files/2379202/boot.log)

gkarop, 2018-09-13:
I have a similar issue. I'm using a Raspberry Pi 3 (model B, V 1.2, 2015) and release branch 3.2 (`-b 3.2.0`). I also had no problems with build, installed OP-TEE following the instructions on `make img-help` and I get the output to the screen through HDMI. The first time I boot Pi it is stuck in `crng init done` as @Rusack mentioned previously; see the screenshot below:
[image]

If I power off and then power on again the Pi it is stuck in the following screen (every time apart from the first boot it is stuck in the following screen):
[image]

I also tried `rootfs.cpio.gz` that @jbech-linaro provided here: 
[https://github.com/OP-TEE/build/issues/276#issuecomment-402683393](https://github.com/OP-TEE/build/issues/276#issuecomment-402683393)
but the result was exactly the same as above.

jbech-linaro, 2018-09-13:
And from the comments in #276 we haven't done anything with the RPi3 B+ model (I still only have the old RPi3). So, I wouldn't be surprised if it's (still) an RPi3 **B+** issue.

igoropaniuk, 2018-09-13:
@gkarop Seems the dhcp-client is the reason of this hang. Try to connect RPi3 to ethernet and see what happens next (I did have a similar issue)

Rusack, 2018-09-13:
I'm using a Rpi 3 model B v1.5 2015, that's what is written on it.

gkarop, 2018-09-14:
> And from the comments in #276 we haven't done anything with the RPi3 B+ model (I still only have the old RPi3). So, I wouldn't be surprised if it's (still) an RPi3 **B+** issue.

@jbech-linaro In the past I tried in a RPi3 **B+** where indeed it does not boot. Currently I'm using an RPi3 **B** model (the screenshots above are from the **B** one). Could it be something with the different versions? Mine is model B v1.2.

> @gkarop Seems the dhcp-client is the reason of this hang. Try to connect RPi3 to ethernet and see what happens next (I did have a similar issue)

@igoropaniuk I tried with ethernet with no success; compared to the screenshots above I only get one more message saying "Link up" and then it is stuck there.

Is there a way to disable dhcp?

jforissier, 2018-09-14:
@gkarop 

> Is there a way to disable dhcp?

Quick and dirty way: delete this line in `common.mk`:
```
        @echo "slink /etc/rc.d/S02_udhcp_networking /etc/init.d/udhcpc 755 0 0" \
```

gkarop, 2018-09-14:
Nothing changed. In my `common.mk` file it is already conditional so probably it was never called anyway:
	
```
@if [ "$(QEMU_USERNET_ENABLE)" = "y" ]; then \
		echo "slink /etc/rc.d/S02_udhcp_networking /etc/init.d/udhcpc 755 0 0" \
		>> $(fl); \
	fi
```

jforissier, 2018-09-14:
@gkarop sorry I failed to notice you're on 3.2.0. Current master does not have `QEMU_USERNET_ENABLE` anymore.

gkarop, 2018-09-17:
I have tried many different things in order to disable `udhcpc` but it always starts during boot. Any ideas? In which configuration files I should search in particular?

dmeignan, 2018-10-30:
Any news with this issue, I have the same problem with a Rpi 3 model B v1.2 2015.

jforissier, 2018-10-30:
@dmeignan should be fixed by https://github.com/OP-TEE/build/pull/316

jedichen121, 2018-10-31:
I just tried the build yesterday and I'm still hanging at the `crng init done`. I checked my local code and found that the changes in the newest commit were there. I'm also using Rpi3 and I'm not using UART but connected directly with HDMI. The following code in the #316 is not in my version of common.mk. Should I add it and try make again?
```
	@echo "BR2_ROOTFS_OVERLAY=\"$(ROOT)/build/br-ext/board/qemu/overlay\"" >> \
		../out-br/extra.conf
```

jbech-linaro, 2018-11-01:
So to summarize things that needs to be addressed and looked at into in the future.
 1. The core team doesn't have any RPi3 model B+, we only have the first RPi3 revision (B).
  1. People are seeing various errors using RPi3 model B. Most of them seems to be that it hangs during boot and `crng init done` is the final message seen.
 2. Using the first RPi3 revision, the core team have no issues building, booting nor running test.

To address "1", we (core team) need to buy a RPi3 model B and start debugging. We can do that, but we have the roadmap planned for several months a head of us with other things and features, which means that it could take a while until we have time to do proper debugging and testing with the newer RPi3 revision. So a faster way forward would be to help users having the issue to narrow down the problem (basically what we've tried in this thread).

Related, if and when I find time, I've planned to add the RPi3 to the automatic build, boot, test cycle initiated by GitHub commits (basically add a config for [IBART](http://optee.mooo.com:5000/)). But with recent changes to the RPi3 boot (doing a more proper boot), it has become harder to deploy the files automatically. I think it still id doable, but just a bit more tricky. If I ever get to work on this, we should hopefully be able to rule out problems like this.

Regarding "2", we can re-test again to ensure that the first RPi3 revision still works. As mentioned yesterday in https://github.com/OP-TEE/build/issues/296#issuecomment-434236933 3.2.0 was tested OK by two independent developers and 3.3.0 released ~2 weeks ago was also tested OK also, see https://github.com/OP-TEE/optee_os/pull/2552#issuecomment-426284630).

So, it feels like there is something else going on here. I suspect it has something to do with the build system itself, that updates or don't update certain files when doing things in a certain order (i.e., maybe flawed dependencies in the makefiles).

jbech-linaro, 2019-01-31:
Issues like this has been sorted out (and yet another release has been made) and more up-to-date information can be found in [this comment](https://github.com/OP-TEE/optee_os/issues/2783#issuecomment-459359271). I'm closing the ticket.

closed 2019-01-31 by jbech-linaro

david-foo, 2019-06-05:
I met the same issue.
I did something on Debain image on my SD card, which is download for raspberrypi.org. For example 2019-04-08-raspbian-stretch-lite.img.
Because the SD card is too bigger I read out the content of boot and root partition then put them onto a small image, so that I can share works to small SD card.
Here are steps I did:
1, rsync -a boot and rootfs from SD card to PC
2,create a small(but enough) blank image with dd, and then partion and format it as fat and ext4
3,rsync –a boot and rootfs to the responding part of the blank image file.
4,flash the new iamge back to small SD card then boot raspiberry, but it stoped here:

[    3.022831] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    3.031781] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.052569] smsc95xx v1.0.4
[    3.116771] smsc95xx 1-1.1:1.0: eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:f7:xx:xx

It seems that the kernel does not recognize the rootfs so that it stoped on mounting the rootfs. But why?
I used to this for other linux board and it works. Can someone give some advises?