OP-TEE/build #307: Poplar board does not boots, stops before U-Boot

gsora, 2018-09-15:
I'm having some troubles getting my Poplar board to boot anything.

I've followed the standard build guide, everything went smoothly, but when I reset the board to make it load from the eMMC, the boot process stops with this error:

```
Bootrom start
Boot Media: eMMC
Decrypt auxiliary code ...OK

lsadc voltage min: 000000FE, max: 000000FF, aver: 000000FE, index: 00000000

Entry boot auxiliary code

Auxiliary code - v1.00
DDR code - V1.1.2 20160205
Build: Mar 24 2016 - 17:09:44

Reg Version:  v134
Reg Time:     2016/03/18 09:44:55
Reg Name:     hi3798cv2dmb_hi3798cv200_ddr3_2gbyte_8bitx4_4layers.reg

Boot auxiliary code success
Bootrom success

LOADER:  Built 2018/09/15 13:07:10+0200 Commit-id df5706a
LOADER:  Switched to aarch64 mode
LOADER:  Entering ARM TRUSTED FIRMWARE
LOADER:  CPU0 executes at 0x0200e000

NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.5(release):v1.6-rc0-4-ge52ed092
NOTICE:  BL1: Built : 13:07:06, Sep 15 2018
ERROR:   error initializing fip
ERROR:   Failed to load BL2 firmware.
```

Is there anything I could try to make it boot correctly?

Thanks.

gsora, 2018-09-15:
After some tinkering, I got a fastboot.bin image that correctly boots OP-TEE, but only if I manually build the ATF and l-loader with recovery switches active.

My guess is that building this way, I'm embedding all the needed binaries right in the fastboot image, while with the standard method I'm reading off the eMMC. Something is missing from the partition images maybe?

gsora, 2018-09-17:
I didn't issue any particular command, I just initialized my work folder with the `poplar.xml` manifest file, and then issued `make`. The fastboot.bin produced by this process leaves the board in the state described in my first message. 

To produce a USB-bootable fastboot image I followed these instructions: https://github.com/96boards-poplar/Documentation/blob/master/OP-TEE/OP-TEE.md

I passed the `POPLAR_RECOVERY=1` to make when building ATF, and `RECOVERY=1` when building l-loader. 

The image produced includes OP-TEE, and passes the test suite.

vchong, 2018-09-17:
It's been a while so I don't quite remember all the details, but if I'm reading everything correctly, `fastboot.bin` shouldn't be generated by this process?

https://github.com/OP-TEE/build/blob/master/poplar.mk#L150
https://github.com/96boards-poplar/l-loader/blob/master/Makefile#L54

gsora, 2018-09-17:
Yes, exactly, I am able to generate a recovery fastboot.img that boots via USB, but somehow it doesn't boot when flashed to the board. 

I'm pretty sure the eMMC is not faulty either because with the USB supplied fastboot image, Debian boots. 

I am using the 1GB version of the Poplar board, and I've supplied the relevant flag to the OP-TEE build phase, but at this point I'm wondering if there's something else I should tweak to make the boot image load on my board revision.

vchong, 2018-09-17:
I just tried and verified that `fastboot.bin` is not generated.

```
repo init -u https://github.com/OP-TEE/manifest.git -m poplar.xml
repo sync
make toolchains
make l-loader
```

If you're having trouble, please follow recovery procedure [here](https://github.com/96boards-poplar/Documentation/blob/master/debian/debian_recovery.md).

gsora, 2018-09-17:
The recovery procedure does produce a valid `fastboot.bin` file (which is the `l-loader.bin` file renamed), the problem I'm having is that I'm not able to produce a working fastboot image, at least something that does load BL2 and beyond. I've opened an issue on ARM Trusted Firmware repository [here](https://github.com/ARM-software/tf-issues/issues/627), because it might be related.

jbech-linaro, 2019-01-31:
@gsora did you get anywhere with this? Still an issue? If not, then I'd like to close the ticket.

gsora, 2019-02-12:
Hi, sorry for the late reply. 

As you can see from [here](https://github.com/ARM-software/tf-issues/issues/627) the issue has been fixed in ARM TF directly. 

I'm closing this issue, thanks everyone!

closed 2019-02-12 by gsora