OP-TEE/build #308: Normal world doesnt recognize gdbserver

oferch, 2018-09-17:
According to the docs I need to start gdbserver on the normal window terminal in order to debug the program. But it doesnt have the gdbserver command (even when compiling with GDBSERVER=y)

Ofer

jbech-linaro, 2018-09-17:
Strange, I spent quite a bit of time with it last week and I had no issues (I was running the QEMU v7 build). [This](http://people.linaro.org/~joakim.bech/optee-gdb-plugin/) is my personal stuff, but this (readme.txt) is basically what I'm doing. But as you are saying to debug on the device, it is only `GDBSERVER=y` that is necessary. Try to clean out the build and build again.

oferch, 2018-09-17:
@jbech-linaro  Thanks for the quick reply.
I'm also running the QEMU7.
Any hint on what I should look/test?
What is the GDBSERVER=y affecting? (so I might check there, that indeed its affecting)

jbech-linaro, 2018-09-17:
The `GDBSERVER=y` basically sets a flag for Buildroot, such that Buildroot will download and include it on the device. So, there is not really any other magic to it.

closed 2019-01-31 by jbech-linaro