OP-TEE/build #310: Make xtest fails

jdeus, 2018-09-26:
Hi,

Currently experiencing problems when building xtest (platform=rpi3):

cd ~/devel/optee/build
make xtest-common
**make: *** No rule to make target 'optee-client', needed by 'xtest-common'.  Stop.**

jbech-linaro, 2018-09-26:
@jdeus we have another OP-TEE release (v3.3.0) targeted for the 5th of October, so probably me or someone else is going to try the RPi3 a try any day, so stay tuned.

jbech-linaro, 2018-09-26:
But hey! `make xtest-common` is not what you are supposed to run. That the old legacy way of building. You should build xtest using buildroot. I.e., simply run `make run` instead.

jdeus, 2018-09-26:
Turns out I also have the same issue when running a simple make build (not run, as I'm on the rpi3): https://github.com/OP-TEE/build/issues/282

I think it might be the cause of my issue there, as benchmark.ta is missing: https://github.com/OP-TEE/optee_os/issues/2553

jbech-linaro, 2019-01-31:
RPi3 issues has been sorted out, please see for example: https://github.com/OP-TEE/optee_os/issues/2783 https://github.com/OP-TEE/optee_os/issues/2770. Closing the ticket.

closed 2019-01-31 by jbech-linaro