OP-TEE/build #312: Implementing own system call in optee

Anushalagadapati, 2018-10-03:
I tried to implement my own system call "hello" in optee, for that i wrote the prototype for sys_hello function in qemu/linux/arch/sh/include/asm/syscalls_64.h and function definition in qemu/linux/arch/arm/kernel/module.c and i assigned system call number in qemu/linux/arch/x86/entry/syscalls/syscall_64.tbl.
When i am calling this function from TA it shows implicit declaration of function 'sys_hello' .Can any one help me how to overcome this error.

jbech-linaro, 2018-10-03:
You also need to add the same in libutee. Search for UTEE/utee and you will see how it is used.

jforissier, 2018-10-03:
@Anushalagadapati the files you are mentioning are Linux kernel, not OP-TEE...

closed 2019-01-31 by jbech-linaro