OP-TEE/build #320: Booting stuck of Raspberry pi: nonblocking pool is initialized

mofanv, 2018-11-13:
Hi Team,

I booted an SD card to Raspberry Pi and powered it on. The Pi got stuck at:

>random: nonblocking pool is initialized

If there were USB things plugged in the board, the stuck information will show the disconnection and connection again of the USB keyboard or mouse.

I used Virtualbox on a Mac, and Debian 9.5 (amd 64) was installed as the build environment. I followed **build/README.md** and **build/docs/rpi3.md**, used **make img-help** to boot the SD card. I have tried two SD cards, but the problem still there.

Could you please give me some suggestions? and how do I fix this?

Thanks a lot!
Vincent

jbech-linaro, 2018-11-13:
Hi, marking this as duplicate, since this is discussed in #296, #306 (and ev. #274).

closed 2018-11-13 by jbech-linaro

mofanv, 2018-11-16:
Thanks!
Finally,  I solved this weird problem through using the **UART** instead of the HDMI with a big screen.