OP-TEE/build #325: Building for rpi3 failed

orif1, 2018-12-10:
Hi,
I followed the steps described in README.md and ran:

```
> repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml -b 3.2.0
> repo sync
> cd build
> make toolchains
> make
```

After running "make" (and waiting a very long time for compilation to complete), I received the following error:

```
make[1]: Leaving directory '/work/thesis/work/Thesis/devel/optee/u-boot'
'/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../linux/arch/arm64/boot/dts/broadcom/bcm2710-rpi-3-b.dtb' -> '/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../out-br/target/boot/bcm2710-rpi-3-b.dtb'
'/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../build/rpi3/firmware/config.txt' -> '/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../out-br/target/boot/config.txt'
'/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../linux/arch/arm64/boot/Image' -> '/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../out-br/target/boot/Image'
'/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../arm-trusted-firmware/build/rpi3/debug/optee.bin' -> '/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../out-br/target/boot/optee.bin'
install: cannot stat '/home/orifai01/work/thesis/work/Thesis/devel/optee/build/../out/uboot.env': No such file or directory
Makefile:169: recipe for target 'update_rootfs' failed
make: *** [update_rootfs] Error 1
```

Does anyone happen to know what is missing and how to fix it?

Thanks!

jbech-linaro, 2018-12-10:
I've replied and re-tested this in many threads, so at least I cannot reproduce the problem:
https://github.com/OP-TEE/build/issues/296#issuecomment-434236933
https://github.com/OP-TEE/optee_os/pull/2404#issuecomment-399413577
https://github.com/OP-TEE/optee_os/pull/2404#issuecomment-399892169
(running latest as of November https://github.com/OP-TEE/build/issues/296#issuecomment-436125208).

jbech-linaro, 2018-12-10:
I suspect there is some issue with the dependency chain in the Makefile that some people are able to trigger, but as I said, I've tried this numerous of time and I simply cannot reproduce it and therefore it's hard to give any advice.

orif1, 2018-12-10:
Thanks @jbech-linaro . I'm not sure if it is related, but I also receive the following error during "repo init":
curl: (22) The requested URL returned error: 404 Not Found
As I understand, this is expected correct?

jbech-linaro, 2018-12-10:
> @orif1  As I understand, this is expected correct?

No, that's not a problem, that is a server side thing that is always seen. If the message confuses you, then you can get rid of it by also add `--no-clone-bundle` to the repo init command:
```bash
$ repo init --help
    --no-clone-bundle   disable use of /clone.bundle on HTTP/HTTPS
```

> @orif1 What else can I try?

Can you give it a try using latest instead, i.e, paste more or less what I have written in [this](https://github.com/OP-TEE/build/issues/296#issuecomment-436125208) comment? However, take note that I'm using a local reference and toolchain, so for you that would instead be:
```bash
$ rm -rf rpi3
$ mkdir rpi3
$ cd rpi3/
$ repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml
$ repo sync -j3
$ make toolchains -j2
$ cd build
$ time make -j8 
$ make img-help 
...
```

orif1, 2018-12-10:
Hi @jbech-linaro, I tried what you suggested. It looks better but I get a different error at the end:

```
Python build finished, but the necessary bits to build these modules were not found:
_bsddb             _sqlite3           _tkinter        
bsddb185           bz2                dbm             
dl                 gdbm               imageop         
readline           sunaudiodev                        
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
_hashlib                                              

Makefile:544: recipe for target 'sharedmods' failed
make[3]: *** [sharedmods] Error 1
package/pkg-generic.mk:229: recipe for target '/work/Thesis/devel/optee/out-br/build/host-python-2.7.15/.stamp_built' failed
make[2]: *** [/work/Thesis/devel/optee/out-br/build/host-python-2.7.15/.stamp_built] Error 2
Makefile:16: recipe for target '_all' failed
make[1]: *** [_all] Error 2
make[1]: Leaving directory '/work/Thesis/devel/optee/out-br'
common.mk:201: recipe for target 'buildroot' failed
make: *** [buildroot] Error 2

real	95m59.885s
user	134m42.936s
sys	22m30.861s
```
I would be very glad for your help.

jbech-linaro, 2018-12-10:
I've never seen that error before, but a quick [Google search](https://www.google.se/search?q=failed+to+build+_hashlib) gives many references to issues like this. Probably a missing package on your distro. Just to cross check, you have installed the https://github.com/OP-TEE/build#71-prerequisites?

orif1, 2018-12-10:
@jbech-linaro Yes.

pannzh, 2018-12-12:
I was trying to build for rpi3 and failed when compiling `net/ipv4/netfilter/ipt_ECN.c`.  Here is the error message:
```
net/ipv4/netfilter/ipt_ECN.c:29:46: warning: 'struct ipt_ECN_info' declared inside parameter list will not be visible outside of this definition or declaration
 set_ect_ip(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
                                              ^~~~~~~~~~~~
net/ipv4/netfilter/ipt_ECN.c: In function 'set_ect_ip':
net/ipv4/netfilter/ipt_ECN.c:33:44: error: dereferencing pointer to incomplete type 'const struct ipt_ECN_info'
  if ((iph->tos & IPT_ECN_IP_MASK) != (einfo->ip_ect & IPT_ECN_IP_MASK)) {
                                            ^~
net/ipv4/netfilter/ipt_ECN.c: At top level:
net/ipv4/netfilter/ipt_ECN.c:48:47: warning: 'struct ipt_ECN_info' declared inside parameter list will not be visible outside of this definition or declaration
 set_ect_tcp(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
                                               ^~~~~~~~~~~~
net/ipv4/netfilter/ipt_ECN.c: In function 'set_ect_tcp':
net/ipv4/netfilter/ipt_ECN.c:58:14: error: dereferencing pointer to incomplete type 'const struct ipt_ECN_info'
  if ((!(einfo->operation & IPT_ECN_OP_SET_ECE) ||
              ^~
net/ipv4/netfilter/ipt_ECN.c:58:28: error: 'IPT_ECN_OP_SET_ECE' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_ECE'?
  if ((!(einfo->operation & IPT_ECN_OP_SET_ECE) ||
                            ^~~~~~~~~~~~~~~~~~
                            IPT_ECN_OP_MATCH_ECE
net/ipv4/netfilter/ipt_ECN.c:58:28: note: each undeclared identifier is reported only once for each function it appears in
net/ipv4/netfilter/ipt_ECN.c:60:28: error: 'IPT_ECN_OP_SET_CWR' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_CWR'?
      (!(einfo->operation & IPT_ECN_OP_SET_CWR) ||
                            ^~~~~~~~~~~~~~~~~~
                            IPT_ECN_OP_MATCH_CWR
net/ipv4/netfilter/ipt_ECN.c: In function 'ecn_tg':
net/ipv4/netfilter/ipt_ECN.c:84:11: error: dereferencing pointer to incomplete type 'const struct ipt_ECN_info'
  if (einfo->operation & IPT_ECN_OP_SET_IP)
           ^~
net/ipv4/netfilter/ipt_ECN.c:84:25: error: 'IPT_ECN_OP_SET_IP' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_IP'?
  if (einfo->operation & IPT_ECN_OP_SET_IP)
                         ^~~~~~~~~~~~~~~~~
                         IPT_ECN_OP_MATCH_IP
net/ipv4/netfilter/ipt_ECN.c:85:24: error: passing argument 2 of 'set_ect_ip' from incompatible pointer type [-Werror=incompatible-pointer-types]
   if (!set_ect_ip(skb, einfo))
                        ^~~~~
net/ipv4/netfilter/ipt_ECN.c:29:60: note: expected 'const struct ipt_ECN_info *' but argument is of type 'const struct ipt_ECN_info *'
 set_ect_ip(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
net/ipv4/netfilter/ipt_ECN.c:88:26: error: 'IPT_ECN_OP_SET_ECE' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_ECE'?
  if (einfo->operation & (IPT_ECN_OP_SET_ECE | IPT_ECN_OP_SET_CWR) &&
                          ^~~~~~~~~~~~~~~~~~
                          IPT_ECN_OP_MATCH_ECE
net/ipv4/netfilter/ipt_ECN.c:88:47: error: 'IPT_ECN_OP_SET_CWR' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_CWR'?
  if (einfo->operation & (IPT_ECN_OP_SET_ECE | IPT_ECN_OP_SET_CWR) &&
                                               ^~~~~~~~~~~~~~~~~~
                                               IPT_ECN_OP_MATCH_CWR
net/ipv4/netfilter/ipt_ECN.c:90:25: error: passing argument 2 of 'set_ect_tcp' from incompatible pointer type [-Werror=incompatible-pointer-types]
   if (!set_ect_tcp(skb, einfo))
                         ^~~~~
net/ipv4/netfilter/ipt_ECN.c:48:61: note: expected 'const struct ipt_ECN_info *' but argument is of type 'const struct ipt_ECN_info *'
 set_ect_tcp(struct sk_buff *skb, const struct ipt_ECN_info *einfo)
                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
net/ipv4/netfilter/ipt_ECN.c: In function 'ecn_tg_check':
net/ipv4/netfilter/ipt_ECN.c:101:11: error: dereferencing pointer to incomplete type 'const struct ipt_ECN_info'
  if (einfo->operation & IPT_ECN_OP_MASK) {
           ^~
net/ipv4/netfilter/ipt_ECN.c:101:25: error: 'IPT_ECN_OP_MASK' undeclared (first use in this function); did you mean 'IPT_ECN_IP_MASK'?
  if (einfo->operation & IPT_ECN_OP_MASK) {
                         ^~~~~~~~~~~~~~~
                         IPT_ECN_IP_MASK
net/ipv4/netfilter/ipt_ECN.c:109:27: error: 'IPT_ECN_OP_SET_ECE' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_ECE'?
  if ((einfo->operation & (IPT_ECN_OP_SET_ECE|IPT_ECN_OP_SET_CWR)) &&
                           ^~~~~~~~~~~~~~~~~~
                           IPT_ECN_OP_MATCH_ECE
net/ipv4/netfilter/ipt_ECN.c:109:46: error: 'IPT_ECN_OP_SET_CWR' undeclared (first use in this function); did you mean 'IPT_ECN_OP_MATCH_CWR'?
  if ((einfo->operation & (IPT_ECN_OP_SET_ECE|IPT_ECN_OP_SET_CWR)) &&
                                              ^~~~~~~~~~~~~~~~~~
                                              IPT_ECN_OP_MATCH_CWR
[... 23 lines cut ...]
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

[... cut ...]
 automake bc bison build-essential cscope curl device-tree-compiler \
 expect flex ftp-upload gdisk iasl libattr1-dev libc6:i386 libcap-dev \
 libfdt-dev libftdi-dev libglib2.0-dev libhidapi-dev libncurses5-dev \
 libpixman-1-dev libssl-dev libstdc++6:i386 libtool libz1:i386 make \
 mtools netcat python-crypto python-serial python-wand unzip uuid-dev \
 xdg-utils xterm xz-utils zlib1g-dev

mkdir optee && cd optee
repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml -b 3.2.0
repo sync
cd build
make toolchains
make
```

msa2, 2019-01-15:
Just downloaded a clean rpi3 version ubuntu 18.04 ...
```
repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml -b 3.2.0
repo sync
cd build
make toolchains
make

```
...and in the end, I get

```
make[1]: Leaving directory 'xxx/optee/u-boot'
'xxx/optee/build/../linux/arch/arm64/boot/dts/broadcom/bcm2710-rpi-3-b.dtb' -> 'xxx/optee/build/../out-br/target/boot/bcm2710-rpi-3-b.dtb'
'xxx/optee/build/../build/rpi3/firmware/config.txt' -> 'xxx/optee/build/../out-br/target/boot/config.txt'
'xxx/optee/build/../linux/arch/arm64/boot/Image' -> 'xxx/optee/build/../out-br/target/boot/Image'
'xxx/optee/build/../arm-trusted-firmware/build/rpi3/debug/optee.bin' -> 'xxx/optee/build/../out-br/target/boot/optee.bin'
install: cannot stat 'xxx/optee/build/../out/uboot.env': No such file or directory
Makefile:169: recipe for target 'update_rootfs' failed

```

jbech-linaro, 2019-01-15:
@msa2 , we're about to make the next OP-TEE release within 2 weeks, which means that we will do regression testing. I.e., part of that is to smoke out issues. So, let's keep an eye at https://github.com/OP-TEE/optee_os/pull/2735 where we report successfully tested devices (which includes RPi3).

jbech-linaro, 2019-01-15:
Oh, I noticed that you've tried 3.2.0. That was tested successfully by @igoropaniuk (https://github.com/OP-TEE/optee_os/pull/2404#issuecomment-399465962) and me (https://github.com/OP-TEE/optee_os/pull/2404#issuecomment-399892169). So, I suspect that it might be some missing dependency or some stray files from previous builds. Try to clean the entire tree and `rm -rf <rpi3>/out && rm -rf <rpi3/out-br` and then rebuild.

msa2, 2019-01-15:
This is probably another problem: does not boot...
```

picocom -b 115200 /dev/ttyUSB0 
picocom v2.2

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Type [C-a] [C-h] to see available commands

Terminal ready
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v2.0(debug):v2.0-507-g34efb683
NOTICE:  BL1: Built : 13:53:01, Jan 15 2019
INFO:    BL1: RAM 0x100ee000 - 0x100f7000
INFO:    BL1: cortex_a53: CPU workaround for 843419 was applied
INFO:    BL1: cortex_a53: CPU workaround for 855873 was applied
NOTICE:  rpi3: Detected: Raspberry Pi 3 Model B (1GB, Sony, UK) [0x00a02082]
INFO:    BL1: Loading BL2
INFO:    Loading image id=1 at address 0x100b4000
INFO:    Image id=1 loaded: 0x100b4000 - 0x100b9410
NOTICE:  BL1: Booting BL2
INFO:    Entry point address = 0x100b4000
INFO:    SPSR = 0x3c5
NOTICE:  BL2: v2.0(debug):v2.0-507-g34efb683
NOTICE:  BL2: Built : 13:53:02, Jan 15 2019
INFO:    BL2: Doing platform setup
INFO:    BL2: Loading image id 3
INFO:    Loading image id=3 at address 0x100e0000
INFO:    Image id=3 loaded: 0x100e0000 - 0x100e9078
INFO:    BL2: Loading image id 4
INFO:    Loading image id=4 at address 0x10100000
INFO:    Image id=4 loaded: 0x10100000 - 0x1010001c
INFO:    OPTEE ep=0x10100000
INFO:    OPTEE header info:
INFO:          magic=0x4554504f
INFO:          version=0x2
INFO:          arch=0x1
INFO:          flags=0x0
INFO:          nb_images=0x1
INFO:    BL2: Loading image id 21
INFO:    Loading image id=21 at address 0x10100000
INFO:    Image id=21 loaded: 0x10100000 - 0x101453e0
INFO:    BL2: Skip loading image id 22
INFO:    BL2: Loading image id 5
INFO:    Loading image id=5 at address 0x11000000
INFO:    Image id=5 loaded: 0x11000000 - 0x11070150
NOTICE:  BL1: Booting BL31
INFO:    Entry point address = 0x100e0000
INFO:    SPSR = 0x3cd
ERROR:   mmap_add_region_check() failed. error -22
ASSERT: lib/xlat_tables_v2/xlat_tables_core.c:743
BACKTRACE: START: assert
0: EL3: 0x100e4270
1: EL3: 0x100e6210
2: EL3: 0x100e4e6c
3: EL3: 0x100e4f4c
4: EL3: 0x100e449c
5: EL3: 0x100e0354
6: EL3: 0x100e00d8
7: EL3: 0x100b4090
BACKTRACE: END: assert

```

jbech-linaro, 2019-01-15:
@msa2 since earlier today, I've tried and I was actually able to reproduce the 3.2.0 uboot.env error when I was building without `-jX` flag and I think the issue seems to be a missing dependency to the `u-boot-rpi-bin` target in the Makefile. I almost never build without j flag and that is probably why haven't been able to reproduce this in the past. I'm gonna dig a bit more into and send a PR for it.

Note that we won't update the 3.2.0 tag. But in case you want to get past the problem you can:
```bash
$ make && make u-boot-rpi-bin && make
```
I know, not a very nice workaround, but it's **a** workaround.

jbech-linaro, 2019-01-15:
Update, when looking into this I realize that the rpi3.mk has changed after we introduced TF-A in a proper way, i.e., it doesn't make sense to file a PR for the dependency error seen at 3.2.0. Having that said, I've created a fix that you either can check out directly (and **before** typing any make commands). I.e.
```bash
$ repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml -b 3.2.0
$ repo sync -j3
$ cd build
$ git remote add jbech https://github.com/jbech-linaro/build.git
$ git remote update
$ git checkout jbech/3.2.0-dependency-fix
$ make toolchains
$ make
```

Or you can just apply this patch in `build.git` **before** starting any build.
```patch
From bf0eb48f572ad8ccd804d48f5b0875bd9f239f0b Mon Sep 17 00:00:00 2001
From: Joakim Bech <joakim.bech@linaro.org>
Date: Tue, 15 Jan 2019 14:45:54 +0100
Subject: [PATCH] rpi3: Fix u-boot target dependencies for 3.2.0 stable release

---
 rpi3.mk | 35 +++++++++++++++++++++++------------
 1 file changed, 23 insertions(+), 12 deletions(-)

diff --git a/rpi3.mk b/rpi3.mk
index c336393..f92566a 100644
--- a/rpi3.mk
+++ b/rpi3.mk
@@ -43,9 +43,9 @@ MODULE_OUTPUT		?= $(ROOT)/module_output
 ################################################################################
 # Targets
 ################################################################################
-all: arm-tf buildroot optee-os u-boot u-boot-rpi-bin \
+all: arm-tf buildroot optee-os u-boot \
 	linux update_rootfs
-clean: arm-tf-clean buildroot-clean u-boot-clean u-boot-rpi-bin-clean \
+clean: arm-tf-clean buildroot-clean u-boot-clean \
 	optee-os-clean head-bin-clean
 
 include toolchain.mk
@@ -84,33 +84,44 @@ arm-tf-clean:
 U-BOOT_EXPORTS ?= CROSS_COMPILE=$(AARCH64_CROSS_COMPILE) ARCH=arm64
 
 .PHONY: u-boot
-u-boot: $(RPI3_HEAD_BIN)
+u-boot: rpi3-u-boot-bin rpi3-u-boot-env
+
+.PHONY: u-boot-clean
+u-boot-clean: u-boot-bin-clean rpi3-u-boot-bin-clean rpi3-head-bin-clean \
+	rpi3-u-boot-env-clean
+
+u-boot-bin: 
 	$(U-BOOT_EXPORTS) $(MAKE) -C $(U-BOOT_PATH) rpi_3_defconfig
 	$(U-BOOT_EXPORTS) $(MAKE) -C $(U-BOOT_PATH) all
 	$(U-BOOT_EXPORTS) $(MAKE) -C $(U-BOOT_PATH) tools
 
-u-boot-clean:
+.PHONY: u-boot-bin-clean
+u-boot-bin-clean:
 	$(U-BOOT_EXPORTS) $(MAKE) -C $(U-BOOT_PATH) clean
 
-u-boot-rpi-bin: $(RPI3_UBOOT_ENV) u-boot
+rpi3-u-boot-bin: rpi3-head-bin u-boot-bin
 	cd $(U-BOOT_PATH) && cat $(RPI3_HEAD_BIN) $(U-BOOT_BIN) > $(U-BOOT_RPI_BIN)
 
-u-boot-rpi-bin-clean:
+.PHONY: rpi3-u-boot-bin-clean
+rpi3-u-boot-bin-clean:
 	rm -f $(U-BOOT_RPI_BIN)
 
-$(RPI3_HEAD_BIN): $(RPI3_FIRMWARE_PATH)/head.S
+rpi3-head-bin: $(RPI3_FIRMWARE_PATH)/head.S
 	mkdir -p $(ROOT)/out/
 	$(AARCH64_CROSS_COMPILE)as $< -o $(ROOT)/out/head.o
-	$(AARCH64_CROSS_COMPILE)objcopy -O binary $(ROOT)/out/head.o $@
+	$(AARCH64_CROSS_COMPILE)objcopy -O binary $(ROOT)/out/head.o $(RPI3_HEAD_BIN)
 
-head-bin-clean:
+.PHONY: rpi3-head-bin-clean
+rpi3-head-bin-clean:
 	rm -f $(RPI3_HEAD_BIN) $(ROOT)/out/head.o
 
-$(RPI3_UBOOT_ENV): $(RPI3_UBOOT_ENV_TXT) u-boot
+.PHONY: rpi3-u-boot-env
+rpi3-u-boot-env: $(RPI3_UBOOT_ENV_TXT) u-boot-bin
 	mkdir -p $(ROOT)/out
-	$(U-BOOT_PATH)/tools/mkenvimage -s 0x4000 -o $(ROOT)/out/uboot.env $(RPI3_UBOOT_ENV_TXT)
+	$(U-BOOT_PATH)/tools/mkenvimage -s 0x4000 -o $(RPI3_UBOOT_ENV) $(RPI3_UBOOT_ENV_TXT)
 
-u-boot-env-clean:
+.PHONY: rpi3-u-boot-env-clean
+rpi3-u-boot-env-clean:
 	rm -f $(RPI3_UBOOT_ENV)
 
 ################################################################################
-- 
2.17.1

```

The second issue isn't something I'm aware of and I haven't looked into it either.

msa2, 2019-01-16:
An update...
`repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml -b 3.2.0
`
- builds ok with -j8, without additional patches 
- result image boots ok

`repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml
`
- builds ok
- but, does not boot (xlat_tables_core.c assert error)

Not sure what my problem with the latest version is, but I can use the 3.2.0 version for now.

jbech-linaro, 2019-01-16:
@msa2 , thanks for confirming. I'd expect that we will see the same error when doing the regression testing (and have to figure out how to solve it also).

jbech-linaro, 2019-01-16:
> `repo init -u https://github.com/OP-TEE/manifest.git -m rpi3.xml `
> 
> * builds ok
> * but, does not boot (xlat_tables_core.c assert error)

Indeed, I could reproduce the error. It seems to have something to do with TF-A. Since going back to the TF-A 2.0 release tag makes it work again. I.e., with that as the only change and the rest on latest xtest passes:
```bash
23304 subtests of which 0 failed
93 test cases of which 0 failed
0 test cases were skipped
TEE test application done!
```
@jforissier , if we cannot figure out the root cause right away, then I suggest that we step back TF-A in the manifest for RPi3 and file a bug at our manifest.git or build.git and deal with this after the release instead (which means that the stable release would work at least). Meanwhile, I'll give it a git bisect try to see if I can find the commit introducing the regression.

igoropaniuk, 2019-01-16:
@jbech-linaro 
btw, you mean TF-A revision change in https://github.com/OP-TEE/manifest/pull/128 or earlier ?

igoropaniuk, 2019-01-16:
because the last time I tried RPi3 setup was Dec 22 (https://github.com/OP-TEE/build/pull/328)

jbech-linaro, 2019-01-16:
> @jbech-linaro
> btw, you mean TF-A revision change in [OP-TEE/manifest#128](https://github.com/OP-TEE/manifest/pull/128) or earlier ?

Yes, I only tested QEMU-v8 and HiKey, not RPi3 (or any other device).