OP-TEE/build #326: Run failure on hikey960

huashizhoujielun, 2018-12-14:
Hi,
I am trying to build OP-TEE on Hikey960.
After following the steps as [https://github.com/OP-TEE/build](url) 
 and press sudo make flash:

> fastboot flash ptable /home/tgj/optee/hikey/build/../l-loader/prm_ptable.img
target reported max download size of 471859200 bytes
sending 'ptable' (24 KB)...
OKAY [  0.012s]
writing 'ptable'...
OKAY [  0.012s]
finished. total time: 0.024s
fastboot flash xloader /home/tgj/optee/hikey/build/../tools-images-hikey960/hisi-sec_xloader.img
target reported max download size of 471859200 bytes
sending 'xloader' (151 KB)...
OKAY [  0.027s]
writing 'xloader'...
OKAY [  0.339s]
finished. total time: 0.366s
fastboot flash fastboot /home/tgj/optee/hikey/build/../l-loader/l-loader.bin
target reported max download size of 471859200 bytes
sending 'fastboot' (25 KB)...
OKAY [  0.011s]
writing 'fastboot'...
OKAY [  0.011s]
finished. total time: 0.023s
fastboot flash fip /home/tgj/optee/hikey/build/../arm-trusted-firmware/build/hikey960/release/fip.bin
target reported max download size of 471859200 bytes
sending 'fip' (1519 KB)...
OKAY [  0.196s]
writing 'fip'...
OKAY [  0.020s]
finished. total time: 0.216s
fastboot flash nvme /home/tgj/optee/hikey/build/../tools-images-hikey960/hisi-nvme.img
target reported max download size of 471859200 bytes
sending 'nvme' (128 KB)...
OKAY [  0.023s]
writing 'nvme'...
OKAY [  0.020s]
finished. total time: 0.043s
fastboot flash boot /home/tgj/optee/hikey/build/../out/boot-fat.uefi.img
target reported max download size of 471859200 bytes
sending 'boot' (65536 KB)...
OKAY [  8.102s]
writing 'boot'...
OKAY [  0.497s]
finished. total time: 8.599s

But when i turn the switch into 1-0-0(boot in normal mode),it showed nothing!

vchong, 2018-12-14:
Try `make recovery` instead of `make flash`. See if that helps.

jforissier, 2018-12-14:
Also, where are you expecting output?
Console is on UART6 on the LS (low-speed) [expansion connector](https://www.seeedstudio.com/upload/image/20180111/1515660535980341.png).
It is the default port for use with the 96boards UART/USB [adapter](https://www.96boards.org/product/uartserial/).

huashizhoujielun, 2018-12-14:
I just bought that adapter online few hours ago.
Maybe i should retry it after few days with this adapter .
Thanks!

huashizhoujielun, 2018-12-14:
I run the recovery script in https://github.com/96boards-hikey/tools-images-hikey960  for many times.
None of any error while recovery and flash.

closed 2020-01-19 by github-actions[bot]