OP-TEE/build #327: Think I have a problem with booting

gperezalbaUGR, 2018-12-19:
Hi, 
I've followed the instructions to build in RPi3. Everything is ok but when I put the SD in the RPi, and connect the RPi3 to a monitor through HDMI and turn it on I get the result you can see in the attached images.
At the beginning it is similar to the output you've pasted in other issues, but when it says:
"Starting kernel ...
## Transferring control to ARM TF (at address8400000) (dtb at 3ab2f000)..."

[image]

then it switches to another screen and I don't know if it's right (think no I've seen the result must be different).

[image]

In the final state you can see in the Final image I can`t write anything.

[image]

Any help is gratefull

Guillermo

jbech-linaro, 2018-12-19:
Unfortunately our OP-TEE setup for RPi3 doesn't work with HDMI, please us a UART cable instead (see the section about that in the instructions). Also more information related to the HDMI boot issue can be seen in https://github.com/OP-TEE/build/issues/306#issuecomment-434966674, https://github.com/OP-TEE/build/issues/320.

igoropaniuk, 2018-12-19:
Here is the fix https://github.com/OP-TEE/build/pull/328

jbech-linaro, 2018-12-19:
@gperezalbaUGR , can you give @igoropaniuk fix a try?

gperezalbaUGR, 2018-12-19:
I'm trying the @igoropaniuk fix which takes me here: https://patchwork.ozlabs.org/patch/602668/
Please confirm me that the files should be:

post-build.sh
`#!/bin/sh

set -u
set -e

# Add a console on tty1
if [ -e ${TARGET_DIR}/etc/inittab ]; then
    grep -qE '^tty1::' ${TARGET_DIR}/etc/inittab || \
	sed -i '/GENERIC_SERIAL/a\
ttyAMA0::respawn:/sbin/getty -L  ttyAMA0 0 vt100 # GENERIC_SERIAL
tty1::respawn:/sbin/getty -L  tty1 0 vt100 # HDMI console' ${TARGET_DIR}/etc/inittab
fi`

cmdline.txt
`root=/dev/mmcblk0p2 rootwait console=tty1 console=ttyAMA0,115200`

raspberrypi3_64_defconfig (added following line because my original file doesn't have BR2_TARGET_GENERIC_GETTY_PORT="tty1" line)

`BR2_TARGET_GENERIC_GETTY_PORT="ttyAMA0"`

Once I've made those changes I have to run `make` again, isn't it?

Apart from the fix... @jbech-linaro  the UART cable is one like this (https://www.kubii.es/componentes-raspberry-pi/1761-cable-usb-a-ttl-4-pin-kubii-3272496006263.html), ins't it?

Thanks both of you.

igoropaniuk, 2018-12-21:
@gperezalbaUGR any updates?

gperezalbaUGR, 2018-12-21:
@igoropaniuk I've done just those two changes (https://github.com/OP-TEE/build/commit/eafaa00fbff6930ab272afc3d11ed9026d4f3882) and what I get from the HDMI is exactly the same as before...
I've made the changes and did `make` again, is there anything else to do?

igoropaniuk, 2018-12-21:
1. BTW, which tag did you fetch initially? based on your photos, seems that it's not the master branch (I guess you're using `3.2.0`).
2. Could you please show the contents of `inittab` file. If you're in the build directory, just run and copy output here:
```
cat out-br/target/etc/inittab 
```

igoropaniuk, 2018-12-21:
ah, got it

In this particular commit you pointed I did a mistake while renaming the variable (which @jforissier noticed).

Please use the last pushed commit (https://github.com/OP-TEE/build/pull/328/commits/daa45dc1d26f556d95f8167709d7efa9c8caf249)

igoropaniuk, 2018-12-21:
> Could you please show the contents of inittab file. If you're in the build directory, just run and copy output here:

After applying the patch from https://github.com/OP-TEE/build/commit/daa45dc1d26f556d95f8167709d7efa9c8caf249, and running `make all`, it should be like this:

```
# /etc/inittab
#
# Copyright (C) 2001 Erik Andersen <andersen@codepoet.org>
#
# Note: BusyBox init doesn't support runlevels.  The runlevels field is
# completely ignored by BusyBox init. If you want runlevels, use
# sysvinit.
#
# Format for each entry: <id>:<runlevels>:<action>:<process>
#
# id        == tty to run on, or empty for /dev/console
# runlevels == ignored
# action    == one of sysinit, respawn, askfirst, wait, and once
# process   == program to run

# Startup the system
::sysinit:/bin/mount -t proc proc /proc
::sysinit:/bin/mount -o remount,rw /
::sysinit:/bin/mkdir -p /dev/pts /dev/shm
::sysinit:/bin/mount -a
::sysinit:/sbin/swapon -a
null::sysinit:/bin/ln -sf /proc/self/fd /dev/fd
null::sysinit:/bin/ln -sf /proc/self/fd/0 /dev/stdin
null::sysinit:/bin/ln -sf /proc/self/fd/1 /dev/stdout
null::sysinit:/bin/ln -sf /proc/self/fd/2 /dev/stderr
::sysinit:/bin/hostname -F /etc/hostname
# now run any rc scripts
::sysinit:/etc/init.d/rcS

# Put a getty on the serial port
ttyS0::respawn:/sbin/getty -L  ttyS0 0 vt100 # GENERIC_SERIAL
tty1::respawn:/sbin/getty -L  tty1 0 vt100 # HDMI console

# Stuff to do for the 3-finger salute
#::ctrlaltdel:/sbin/reboot

# Stuff to do before rebooting
::shutdown:/etc/init.d/rcK
::shutdown:/sbin/swapoff -a
::shutdown:/bin/umount -a -r

```

The significant lines are these:
```
...
ttyS0::respawn:/sbin/getty -L  ttyS0 0 vt100 # GENERIC_SERIAL
tty1::respawn:/sbin/getty -L  tty1 0 vt100 # HDMI console
...
```
which tells to spawn getty/login to both serial and HDMI consoles

gperezalbaUGR, 2018-12-21:
Yes @igoropaniuk it's working now!
I'm really grateful

closed 2018-12-21 by gperezalbaUGR