OP-TEE/build #329: Copy files to SD card

orif1, 2018-12-20:
Hi,

Which files exactly do I need to copy to RPI3 SD card?
* There is one file in the out folder:
uboot.env
* And the following in the out-br folder:
folders: build, host, images, target
files: ..config.tmp, .br-external, xml without a name, defconfig, extra.conf, Makefile, staging

Thanks,
Orit

jbech-linaro, 2018-12-20:
Hi @orif1 ,

Running `make img-help` will give an answer to that question. It's lots of files, so please follow the what `make img-help` says instead of trying to copy manually.

orif1, 2018-12-23:
Hi jbech,

Sorry for the basic question. I think I'm doing something wrong. 
I run "make img-help" from the build directory and get the following instructions:

`orifai01@ssg-dev-vb ~/work/optee-forsd/optee/build
$ make img-help
$ fdisk /dev/sdx   # where sdx is the name of your sd-card
   > p             # prints partition table
   > d             # repeat until all partitions are deleted
   > n             # create a new partition
   > p             # create primary
   > 1             # make it the first partition
   > <enter>       # use the default sector
   > +32M          # create a boot partition with 32MB of space
   > n             # create rootfs partition
   > p
   > 2
   > <enter>
   > <enter>       # fill the remaining disk, adjust size to fit your needs
   > t             # change partition type
   > 1             # select first partition
   > e             # use type 'e' (FAT16)
   > a             # make partition bootable
   > 1             # select first partition
   > p             # double check everything looks right
   > w             # write partition table to disk.

run the following as root
   $ mkfs.vfat -F16 -n BOOT /dev/sdx1
   $ mkdir -p /media/boot
   $ mount /dev/sdx1 /media/boot
   $ cd /media
   $ gunzip -cd /home/orifai01/work/optee-forsd/optee/build/../out-br/images/rootfs.cpio.gz | sudo cpio -idmv "boot/*"
   $ umount boot

run the following as root
   $ mkfs.ext4 -L rootfs /dev/sdx2
   $ mkdir -p /media/rootfs
   $ mount /dev/sdx2 /media/rootfs
   $ cd rootfs
   $ gunzip -cd /home/orifai01/work/optee-forsd/optee/build/../out-br/images/rootfs.cpio.gz | sudo cpio -idmv
   $ rm -rf /media/rootfs/boot/*
   $ cd .. && umount rootfs
`
Are these the right instructions?
If so,  I have not found /dev/sdx1 or /dev/sdx2.

Note that I received the op-tee project already compiled from a friend because I wasn't able to compile it on my machine.

Thank you

jbech-linaro, 2018-12-27:
> Are these the right instructions?

Yes!

> If so, I have not found /dev/sdx1 or /dev/sdx2.

That's just pseudonym for the true drive and the reason for the 'x' is simply to reduce the likelyhood of wiping the wrong disk (as stated in the docs [1]). I.e., you as a user are supposed to find out what `/dev/xxxyyyzzz` your memory card gets when plugging it into your computer. 

```
[1] The last step is to partition and format the memory card and to put the files onto the same.
That is something we don't want to automate, since if anything goes wrong, in worst case it 
might wipe one of your regular hard disks. Instead what we have done, is that we have 
created another makefile target that will tell you exactly what to do. Run that command 
and follow the instructions there.
```

> 
> Note that I received the op-tee project already compiled from a friend because I wasn't able to compile it on my machine.

Then you will have a hard time to do updates and work with this and the paths etc from `make img-help` will of course point to weird things, since the output from that make target uses absolute paths to **your** repo forest. Having that said, you can still put Trusted Applications and any other file in the root fs if you want too. I.e., mount the system as described in the last part of `make img-help` and then copy the files to the mounted folder (TA's should be located under `/lib/optee_armtz`).

My recommendation is to save you both time and headache and to learn a bit more is to build it all from scratch on your own.

closed 2019-01-31 by jbech-linaro